lardemua/atom #458: New branch for frustums

danifpdra, 2022-05-16:
Hi @miguelriemoliveira,

I created a new branch for RGB frustums, the branch is called **frustum**...

You can see the frustum with

`roslaunch larcc_calibration collect_data.launch output_folder:=~/datasets/larcc_sim/tests_for_paper overwrite:=true bag_rate:=0.1 bag_start:=0
`

The idea is to get some images like this 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

danifpdra, 2022-05-19:
Check this https://github.com/DavidB-CMU/rviz_tools_py

miguelriemoliveira, 2022-05-20:
We have our first filled frustum 

[image]

miguelriemoliveira, 2022-05-20:
and for the entire larcc

[image]

danifpdra, 2022-05-20:
Looks great! Is it possible to see them in real time running the bag? Or just offline?

miguelriemoliveira, 2022-05-20:
Sure it is. I changed the colors from random to fixed. We are now defining a colormap Pastel2 https://matplotlib.org/3.5.0/gallery/color/colormap_reference.html to define the sensor's color based on its index.

Looks even better:

[image]

I will commit in a moment.

danifpdra, 2022-05-20:
Great! You should have been a designer

miguelriemoliveira, 2022-05-20:
> Great! You should have been a designer
Wasn't it a doctor?

Ok, this is operational and in the master branch. Deleted the frustum branch.

Have fun with the images for the paper.

closed 2022-05-20 by miguelriemoliveira

danifpdra, 2022-05-20:
You are multi-talented!

Thanks, will do.

PR #472 (2022-05-20): Frustum with filled areas
Fixes #458.