lardemua/atom #461: Think about a better way to evaluate lidar-lidar evaluation (for example point to line distance)

danifpdra, 2022-05-18:


miguelriemoliveira, 2022-05-19:
Well, we could use that point to line idea. Do you think we should work on it for this paper?

danifpdra, 2022-05-19:
I don't know, I explained what was happening in that section but I think the reviewers can devalue our results, don't you think?

miguelriemoliveira, 2022-05-19:
> I don't know, I explained what was happening in that section but I think the reviewers can devalue our results, don't you think?

Let me read your discussion and then see if it leaves that impression. 
Also, @manuelgitgomes working on icp calibration #463 so we can later compare and see of the calibration using icp is also "bad".

manuelgitgomes, 2022-05-19:
> That means that when projecting transforming the point clouds from one coordinate system to the other we could actually end up with a displacement up to 50mm...

I'm not 100% sure I understood, but we can argue that if the results are under 50mm, the calibration was successful in achieving the best it could regarding the sensor's resolution. If they are over 50mm, some improvements need to be taken. So we can argue that the calibration inside the simulated environment reached its peak while the one in the real world can be improved.

danifpdra, 2022-05-19:
> I'm not 100% sure I understood, but we can argue that if the results are under 50mm, the calibration was successful in achieving the best it could regarding the sensor's resolution. If they are over [...]

I don't think we can argue that... Because, as I said, it can just be a case of luck. For example in lidar2 to lidar3 we have 13mm, that means the we could have had the luck of the lidar scans aligning. And when in lidar 1 to lidar2 we have higher results that means that we could just have had bad luck and the lidar scans ended up with a 50mm alignment. Which really means that those results do not indicate that the calibration was accurate or not.

manuelgitgomes, 2022-05-19:
> Which really means that those results do not indicate that the calibration was accurate or not.

Ok, that makes sense. And an error of 50 can mean two different things. It can mean a correct calibration with the point in between two points or it can mean a displacement in the vertical or horizontal direction of 50 mm, which is not a correct calibration, right?

danifpdra, 2022-05-19:
> Ok, that makes sense. And an error of 50 can mean two different things. It can mean a correct calibration with the point in between two points or it can mean a displacement in the vertical or [...]

Yes, exactly that!

miguelriemoliveira, 2022-05-19:
> I don't think we can argue that... Because, as I said, it can just be a case of luck. For example in lidar2 to lidar3 we have 13mm, that means the we could have had the luck of the lidar scans [...]

the way I think about it is that we cannot expect a calibration to be more accurate than the sensor's resolution. So if the resolution is about 50mm and we calibrate with an error of 10, we were probably lucky. But if, for example we get an error or 80 mm then it means the calibration could be better and is therefore not accurate. So I think I agree with @manuelgitgomes when he says this part:

> we can argue that if the results are under 50mm, the calibration was successful in achieving the best it could regarding the sensor's resolution. If they are over 50mm, some improvements need to be taken. 

@danifpdra  replied:

> I don't think we can argue that... Because, as I said, it can just be a case of luck. For example in lidar2 to lidar3 we have 13mm, that means the we could have had the luck of the lidar scans [...]

and yes, luck (randomness) is always involved. But that's why we compute this for multiple points and multiple collections, to extract the tendency. So if for multiple points and multiple collections we are having accuracy values close to the sensors resolution I would say our calibration is the "most accurate it can be", whereas if our accuracy is considerably larger than the sensor resolution it is fair to say that the calibration can be improved.

Does it make sense? I did not understand this part

> Ok, that makes sense. And an error of 50 can mean two different things. It can mean a correct calibration with the point in between two points or it can mean a displacement in the vertical or horizontal direction of 50 mm, which is not a correct calibration, right?

manuelgitgomes, 2022-05-19:
> Does it make sense? I did not understand this part

An error of 50mm can mean the projection in red, which is correct, just unlucky, but it can also mean the projection in blue, which is incorrect.
[image]
[image]

miguelriemoliveira, 2022-05-19:
I think I still do not understand. Let's talk, perhaps its better.

miguelriemoliveira, 2023-03-31:
Nobody is going to work on this in the near future.

closed 2023-03-31 by miguelriemoliveira