lardemua/atom #473: New evaluation script - sensor to frame evaluation

manuelgitgomes, 2022-05-21:
Create a new evaluation script, evaluating differences in transforms of the sensor w.r.t. a specific frame.

miguelriemoliveira, 2022-05-21:
[image]

manuelgitgomes, 2022-05-22:
Hello! A first iteration of this evaluation was developed. You can try it by running the command:
```bash
rosrun atom_evaluation sensor_to_frame_evaluation -train_json $ATOM_DATASETS/atlascar2/gazebo_move/atom_calibration.json -test_json $ATOM_DATASETS/atlascar2/gazebo_move/dataset.json -ss top_left_camera -tf base_link
```
It currently retrieves the transform between a sensor and a frame. The first results are:
```bash
+--------------+------------+-----------+
| Collection # | Trans (mm) | Rot (deg) |
+--------------+------------+-----------+
|     000      |  11.1533   |   3.7041  |
|     001      |  11.1533   |   3.7041  |
|     002      |  11.1533   |   3.7041  |
|     003      |  11.1533   |   3.7041  |
|     004      |  11.1533   |   3.7041  |
|     005      |  11.1533   |   3.7041  |
|     006      |  11.1533   |   3.7041  |
|     007      |  11.1533   |   3.7041  |
|     008      |  11.1533   |   3.7041  |
|     009      |  11.1533   |   3.7041  |
|     010      |  11.1533   |   3.7041  |
|     011      |  11.1533   |   3.7041  |
|     012      |  11.1533   |   3.7041  |
|     013      |  11.1533   |   3.7041  |
|     014      |  11.1533   |   3.7041  |
|     015      |  11.1533   |   3.7041  |
|     016      |  11.1533   |   3.7041  |
|     017      |  11.1533   |   3.7041  |
|     018      |  11.1533   |   3.7041  |
|     019      |  11.1533   |   3.7041  |
|     020      |  11.1533   |   3.7041  |
|     021      |  11.1533   |   3.7041  |
|     022      |  11.1533   |   3.7041  |
|     023      |  11.1533   |   3.7041  |
|     024      |  11.1533   |   3.7041  |
|     025      |  11.1533   |   3.7041  |
|     026      |  11.1533   |   3.7041  |
|     027      |  11.1533   |   3.7041  |
|     028      |  11.1533   |   3.7041  |
|     029      |  11.1533   |   3.7041  |
|     030      |  11.1533   |   3.7041  |
|     031      |  11.1533   |   3.7041  |
|     032      |  11.1533   |   3.7041  |
|     033      |  11.1533   |   3.7041  |
|     034      |  11.1533   |   3.7041  |
|     035      |  11.1533   |   3.7041  |
|     036      |  11.1533   |   3.7041  |
|     037      |  11.1533   |   3.7041  |
|     038      |  11.1533   |   3.7041  |
|     039      |  11.1533   |   3.7041  |
|     040      |  11.1533   |   3.7041  |
|     041      |  11.1533   |   3.7041  |
|     042      |  11.1533   |   3.7041  |
|     043      |  11.1533   |   3.7041  |
|     044      |  11.1533   |   3.7041  |
|     045      |  11.1533   |   3.7041  |
|     046      |  11.1533   |   3.7041  |
|     047      |  11.1533   |   3.7041  |
|     048      |  11.1533   |   3.7041  |
|     049      |  11.1533   |   3.7041  |
|     050      |  11.1533   |   3.7041  |
|     051      |  11.1533   |   3.7041  |
|     052      |  11.1533   |   3.7041  |
|     053      |  11.1533   |   3.7041  |
|   Averages   |  11.1533   |   3.7041  |
+--------------+------------+-----------+

```

Does this make sense? The ground truth transform does not change between collections, so does it make sense to calculate in every collection?
And how can this translate to the real world?
Is there a way to calculate an RMS with only translation and rotation errors?
Thanks!

miguelriemoliveira, 2022-05-23:
Thanks Manuel.  I did not antecipate this but its true: we will have the
same resilt for all collections, it makes sense.

One test l use for assessing if the script is functioning properly is: if
you run using the same datasets both for train and test, is the error zero?