lardemua/atom #478: ATOM inspect dataset changed?

danifpdra, 2022-05-26:
Hi @miguelriemoliveira ,

This file changed https://github.com/lardemua/atom/blob/de8e171ec7ffb3083e0cabe74b0700978f3a28b6/atom_evaluation/scripts/inspect_atom_dataset

it doesn't show the green and red table anymore and it's just a list of all collections... why? it's must harder to read

danifpdra, 2022-05-26:
What does the two last lines mean?

```Collection 0:
Sensor 3dlidar
Complete (non partial) detection
Sensor hand_camera
Partial detection
Sensor world_camera
Partial detection
[False, True, True]
[False, True, True]
```

miguelriemoliveira, 2022-05-26:
If it changed it was not intentional. I will add it to my todo list.

miguelriemoliveira, 2022-05-28:
Mistery solved. There were two files:

1. atom_calibration/inspect_dataset (which was doing what we expected)

and 

2. atom_evaluation/inspect_atom_dataset (which was printing old stuff)

I am deleting 2, and renaming 1 to have the name of 2.

closed 2022-05-28 by miguelriemoliveira

danifpdra, 2022-05-28:
ok, thanks

commit 73646d00 (2022-05-28): Fix #478