lardemua/atom #479: Error launch collector

danifpdra, 2022-05-31:
Hi @miguelriemoliveira ,

When running

`roslaunch larcc_calibration collect_data.launch output_folder:=~/datasets/larcc_sim/tests_for_paper overwrite:=true bag_rate:=0.1 bag_start:=0`

I am getting this error

```
Traceback (most recent call last):
  File "/home/daniela/catkin_ws/devel/lib/atom_calibration/collect_data", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/scripts/collect_data", line 145, in <module>
    data_collector = DataCollectorAndLabeler(args, server, menu_handler)
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 175, in __init__
    self.abstract_transforms = self.getAllAbstractTransforms()
  File "/home/daniela/catkin_ws/src/calibration/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 381, in getAllAbstractTransforms
    all_frames = self.listener.getFrameStrings()
  File "/opt/ros/noetic/lib/python3/dist-packages/tf/listener.py", line 97, in getFrameStrings
    data = yaml.load(self._buffer.all_frames_as_yaml()) or {}
TypeError: load() missing 1 required positional argument: 'Loader'
```

I have a feeling we've had this error before... Did you change anything? Does it also happen to you? Do you know how to solve it?

danifpdra, 2022-05-31:
Fixed it with 

`!pip install pyyaml==5.4.1`

Closing.

closed 2022-05-31 by danifpdra

miguelriemoliveira, 2022-05-31:
Great!

manuelgitgomes, 2022-10-11:
I had a similar error on a different script. This is due to the new version of pyyaml. Try installing the 5.4.1 with
```bash
pip3 install pyyaml=5.4.1
```

Obter o Outlook para Android<https://aka.ms/AAb9ysg>