lardemua/atom #483: Standarizing the scripts in ATOM

miguelriemoliveira, 2022-06-04:
Hi @danifpdra and @manuelgitgomes ,

I am writing the documentation for ATOM and I noticed that there is no standard for launching the scripts in atom.

The original idea was to call everything with ros launch files, but in fact we are calling python scripts in many places.

I would say that the best would be to use the launch files for setting up the ros stuff such as rviz, but then always call the python script.

Right now we have:

create_calibration_pkg        , a python script
configure_calibration_pkg   , a python script
set_initial_estimate              , a roslaunch
collect_data                         , a roslaunch
dataset_playback                , a roslaunch
calibrate                              , a roslaunch and a python script

perhaps we should have:

create_calibration_pkg        , a python script
configure_calibration_pkg   , a python script
set_initial_estimate              , a roslaunch + a python script
collect_data                         , a roslaunch + a python script
dataset_playback                , a roslaunch + a python script
calibrate                              , a roslaunch + a python script

What do you think?

manuelgitgomes, 2022-06-05:
Hello! Sure, it makes sense!
However, will the script be the one on atom_calibration or will a new one be created for each specific calibration package?

danifpdra, 2022-06-05:
> Hi @danifpdra and @manuelgitgomes ,
>
> I am writing the documentation for ATOM and I noticed that there is no standard for launching the scripts in atom.
>
> The original idea was to call everything with [...]

Why not everything just with one command? It gives the user more trouble to use two

miguelriemoliveira, 2022-06-05:
Hi @danifpdra and @manuelgitgomes ,

thanks for your replies. 

> Why not everything just with one command? It gives the user more trouble to use two

Right. that would be better. The problems with that are:

   1. All prints are flushed to the same terminal (including the one from the rosbag play), which makes it difficult to see anything.
   2. it is not consistent throughout atom, sometimes we have launch files, sometimes scripts ... sometimes launch files and scripts
   3. The configuration the argparse allows for make us have a lot of command line arguments in the python scripts. Passing all these to the launch files is very difficult and makes using atom more complex

> However, will the script be the one on atom_calibration or will a new one be created for each specific calibration package?

That's a good question. In fact what happens under the hood is that we are always calling atom_calibration/script python codes like this:

    rosrun atom_calibration script_do_something --name my_robot_calibration ...

but it looks much nicer when we do

    rosrun my_robot_calibration do_something 

or 

    roslaunch my_robot_calibration do_something.launch 

Perhaps we can have a talk. Do you have any ideas on how to simplify?

Perhaps everything should be launch files ... not sure ...

danifpdra, 2022-06-06:
Hi,

I think that some things that are standard for every script should be customized using templates... For example when launching the dataset reviewer we need to launch the visualization plus a python script but we need to give that script all the information, why not also configure that script with the configure_calibration_pkg and the roslaunch launches the visualization plus the script and it doesn't require us to give any argument unless we want to change something...

manuelgitgomes, 2022-06-06:
Hello! Sorry, today is impossible as I will be on atlascar.
Tommorrow morning maybe?

miguelriemoliveira, 2022-06-06:
Tomorrow is not good for me. 

Let's do it Wednesday 9h?

manuelgitgomes, 2022-06-06:
Sure, wednesday at 14?

miguelriemoliveira, 2022-06-06:
This is difficult :-). Wednesday afternoon I am busy.

Lets do it Thursday 14? In the meantime I will work on the documentation.

danifpdra, 2022-06-06:
That works for me.

miguelriemoliveira, 2022-06-07:
@manuelgitgomes , can you?

manuelgitgomes, 2022-06-07:
@miguelriemoliveira yes! I reacted to your comment with a thumbs up. These reactions are not sent through email, next time I will answer the "normal" way!

miguelriemoliveira, 2022-06-07:
Ah, ok, my bad.

You can do the emojis, no problem.

See you Thursday 9h then.

danifpdra, 2022-06-07:
Thursday at 9 or 14? You said 14 and now you said 9

miguelriemoliveira, 2022-06-07:
Sorry 14h

closed 2022-08-15 by miguelriemoliveira