lardemua/atom #485: Depth camera labeling using image clicks

miguelriemoliveira, 2022-06-05:
Hi @danifpdra ,

I am thinking about changing the depth camera automatic labeling seed point selection to an image click mechanism.
Something like "just click inside the pattern".

It would be better that the current approach, right?

danifpdra, 2022-06-05:
> Hi @danifpdra ,
> 
> I am thinking about changing the depth camera automatic labeling seed point selection to an image click mechanism. Something like "just click inside the pattern".
> 
> It would be better that the current approach, right?

Yes, sure

miguelriemoliveira, 2022-06-06:
Got it

https://youtube.com/shorts/OuLHvp95UBI?feature=share

Will commit in a sec

miguelriemoliveira, 2022-06-06:
Done in d3faa4576fd0db611f5ee107fe8b0346a73aa736

closed 2022-06-06 by miguelriemoliveira