lardemua/atom #491: IMU pose calibration?

peci1, 2022-06-14:
Hi, this project looks very interesting. I'm wondering if you have plans to also add calibration of IMU pose with respect to the defined base_link frame. I know this is a very different problem than the other calibrations you offer in Atom, but maybe you have a clue how to do it right.

miguelriemoliveira, 2022-06-14:
Hi @peci1 ,

Thanks for the nice words. BTW we are preparing a documentation for atom. If you use ATOM this should be helpful.
We are releasing it soon but I will share it here.

https://lardemua.github.io/atom_documentation/

We are not considering expanding ATOM to the calibration of IMUs in the near future. Perhaps if I get a M.Sc. Thesis student in september ...

In any case I know that Kalibr was made considering the calibration of cameras and IMUs .. perhaps it helps ...

https://github.com/ethz-asl/kalibr

peci1, 2022-06-14:
Thank you for the quick reply. We're not using Atom (yet), but I'm considering it as it seems really straightforward.

If we get to using it and find some issues, we'll let you know.

One thing that I was missing when reading the Readme was the requirement for how many poses and what calibration boards are needed. Can that information be somehow generalized and mentioned in the tutorial?

Feel free to close this issue as wontfix.

miguelriemoliveira, 2022-06-14:
Hi @peci1 ,

> If we get to using it and find some issues, we'll let you know.

Great!

> One thing that I was missing when reading the Readme was the requirement for how many poses and what calibration boards are needed. Can that information be somehow generalized and mentioned in the tutorial?

We use about 20 to 30 collections for training and its generally enough. That's about the same you need when calibrating a stereo system with opencv for example.  In any case thanks for the tip, I will include this information.

closed 2022-06-14 by miguelriemoliveira

peci1, 2022-06-14:
Thanks for possible updates of the documentation. It would probably be best to estimate the numbers in per-pair or per-sensor or per-fov-degree units. For example, our robots have 4-6 cameras pointing to all directions and an omnidirectional lidar, so it would be useful to know that it is enough to e.g. have 10 samples per a camera-lidar pair.

miguelriemoliveira, 2022-06-14:
> Thanks for possible updates of the documentation. It would probably be best to estimate the numbers in per-pair or per-sensor or per-fov-degree units. For example, our robots have 4-6 cameras [...]

Well, that's a tricky one because ATOM is able to calibrate everything altogether, so it does not work in pairs. Of course ATOM need several collections were several sensors see the pattern, but I cannot have a good estimate for how many. Nonetheless, ATOM provides several evaluation scripts to be run post calibration which will tell you the reprojection errors between sensors.

So you could try with a dataset and if the calibration is not accurate enough add a few more images and calibrate again.

miguelriemoliveira, 2022-06-15:
Thanks @peci1 ,

In ATOM we consider an RGB-D camera as two separate sensors, which are to be calibrated just like any other two sensors in the system. In fact we are now running experiments where we compare the extrinsic calibration between the depth and rgb sensor of an RGDD camera produced by ATOM with the factory calibration of the device.

ATOM also supports intrinsic camera calibration (for now just RGB), but requires a first guess for these parameters. In practice, what we do is use the ROS monocular camera calibration 

http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration

and the depth camera intrinsic calibration 

http://wiki.ros.org/openni_launch/Tutorials/IntrinsicCalibration

to produce camera_info messages which are then use as first guess for ATOM's intrinsic parameter estimation which is simultaneous with the extrinsic calibration.

I will also try to clarify this in the documentation.