lardemua/atom #492: Mininum number of detected point in evaluation

manuelgitgomes, 2022-06-16:
Hello @miguelriemoliveira  and @danifpdra!
When taking results for my thesis, a particular situation caught my attention. When evaluating RGB to RGB, one collection appeared like this:
[image]
Only one correspondent point between the source and the target image. This collection skews the results, as this collection with one detection is worth as much as other collections with many more detections.
I can think of two different solutions:
- The average of the results at the end is not given based on collections and is given instead based on detections
- If you want to keep each collection with the same value, a minimum number of detections should be placed in order to avoid this kind of situation

What do you prefer?

miguelriemoliveira, 2022-06-16:
I vote for the first. In fact, when we say that for this dataset we have 0.8 pixels of average error, we are doing the what?

If we are doing the average of each collection, and then the average of the averages I think its wrong.

It should be the average of the weighted averages, were the weight corresponds to the number of points divided by the total number of points for all collections.

So I vote for the first option.

miguelriemoliveira, 2022-06-20:
@danifpdra I think that's for the calibration, this is in the evaluation...