lardemua/atom #495: Cross collection evaluation

manuelgitgomes, 2022-06-23:
Create an evaluation script that evaluates cross collection in reference to odom.

manuelgitgomes, 2022-06-23:
Hello @miguelriemoliveira! First results!!!
Without a for cycle, so every sensor_source is compared to the sensor_target in collection 002.
```bash
+--------------+-----------+-------------+-------------+------------+-----------+
| Collection # | RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
+--------------+-----------+-------------+-------------+------------+-----------+
|     000      |   0.3042  |    0.1600   |    0.1830   |  473.0665  |  38.0853  |
|     002      |   0.3957  |    0.2494   |    0.1765   |   3.1672   |   0.0400  |
|     003      |   0.5373  |    0.3972   |    0.2069   |  494.9906  |   9.7523  |
|     005      |   0.7615  |    0.4091   |    0.3663   |  890.0401  |  30.7030  |
|     006      |   1.2945  |    1.2231   |    0.1972   | 1176.9260  |  39.8715  |
|     014      |   1.8724  |    1.8288   |    0.2376   |  995.8144  |  81.0185  |
|     015      |   1.2191  |    1.1461   |    0.2101   |  944.0224  |  58.6476  |
|     016      |   0.9118  |    0.8445   |    0.1837   |  485.3810  |  58.4013  |
|     017      |   0.4155  |    0.2874   |    0.1877   |  693.8194  |  43.4426  |
|     018      |   0.2886  |    0.1808   |    0.1336   |  788.9106  |  43.4873  |
|   Averages   |   0.8001  |    0.6726   |    0.2083   |  694.6138  |  40.3449  |
+--------------+-----------+-------------+-------------+------------+-----------+

```

To compare, the regular rgb_to_rgb_evaluation.

```bash
+--------------+-----------+-------------+-------------+------------+-----------+
| Collection # | RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
+--------------+-----------+-------------+-------------+------------+-----------+
|     000      |   0.2556  |    0.1555   |    0.1182   |   1.9794   |   0.4632  |
|     002      |   0.3957  |    0.2494   |    0.1765   |   3.1672   |   0.0400  |
|     003      |   0.5176  |    0.3861   |    0.1302   |   2.5821   |   0.1747  |
|     005      |   0.9208  |    0.4721   |    0.4350   |   1.5703   |   0.1420  |
|     006      |   0.8880  |    0.8272   |    0.1871   |   2.5806   |   0.1421  |
|     014      |   0.5643  |    0.3938   |    0.2263   |   1.0497   |   0.0839  |
|     015      |   0.5477  |    0.2461   |    0.2552   |   1.4621   |   0.1226  |
|     016      |   0.3644  |    0.2964   |    0.0953   |   1.4866   |   0.0844  |
|     017      |   0.3077  |    0.1965   |    0.1110   |   1.4183   |   0.1789  |
|     018      |   0.3184  |    0.2345   |    0.0852   |   1.2571   |   0.2727  |
|   Averages   |   0.5080  |    0.3458   |    0.1820   |   1.8553   |   0.1705  |
+--------------+-----------+-------------+-------------+------------+-----------+
```

As it can be seen, collection 002 presents the same results, as expected.
However, both translation and rotation are considerably higher in the cross collection. Is this large of a difference expected? It seems like a bug in the translation and rotation calculation, as I did not change it and more than likely it needed change to function correctly. Are these values interesting? Should I try to fix them?
Nevertheless, the RMS results are really good!
I will continue work on this tomorrow

manuelgitgomes, 2022-06-24:
I believe the RGB to RGB cross collection is done! From the 10 used collections, 45 pairs can be formed, and they are presented here:

```bash
+-----------------+-----------+-------------+-------------+------------+-----------+
| Collection Pair | RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
+-----------------+-----------+-------------+-------------+------------+-----------+
|     000-002     |   0.3042  |    0.1600   |    0.1830   |   1.5116   |   0.2103  |
|     000-003     |   0.3838  |    0.2406   |    0.1815   |   4.1979   |   0.2644  |
|     000-005     |   0.6321  |    0.4290   |    0.2526   |   3.3498   |   0.4647  |
|     000-006     |   0.5249  |    0.3388   |    0.2581   |   5.8086   |   0.3474  |
|     000-014     |   0.6440  |    0.4820   |    0.2624   |   4.6392   |   0.4433  |
|     000-015     |   0.6130  |    0.3767   |    0.2990   |   3.4351   |   0.3694  |
|     000-016     |   0.3734  |    0.2591   |    0.1458   |   2.5066   |   0.5439  |
|     000-017     |   0.4135  |    0.2771   |    0.1828   |   2.2199   |   0.4765  |
|     000-018     |   0.5001  |    0.3579   |    0.2145   |   1.6849   |   0.2866  |
|     002-003     |   0.4981  |    0.2861   |    0.2749   |   0.9229   |   0.0874  |
|     002-005     |   0.6568  |    0.4446   |    0.3053   |   1.3953   |   0.2871  |
|     002-006     |   0.4875  |    0.2430   |    0.2864   |   1.2257   |   0.1697  |
|     002-014     |   1.0808  |    0.9593   |    0.3654   |   6.2306   |   0.2662  |
|     002-015     |   1.2378  |    1.0909   |    0.4107   |   3.6374   |   0.2381  |
|     002-016     |   1.1494  |    1.0770   |    0.2283   |   4.9903   |   0.3778  |
|     002-017     |   1.3070  |    1.2080   |    0.3094   |   2.9241   |   0.3133  |
|     002-018     |   1.7980  |    1.7099   |    0.2929   |   3.3141   |   0.1429  |
|     003-005     |   0.7665  |    0.5386   |    0.3367   |   3.3329   |   0.3540  |
|     003-006     |   0.5511  |    0.3741   |    0.2507   |   0.7832   |   0.2129  |
|     003-014     |   1.5591  |    1.4894   |    0.1892   |   7.3969   |   0.3358  |
|     003-015     |   1.6889  |    1.5852   |    0.3818   |   5.3273   |   0.3431  |
|     003-016     |   1.7116  |    1.6697   |    0.1964   |   6.5460   |   0.3811  |
|     003-017     |   1.8491  |    1.7939   |    0.2115   |   4.7951   |   0.4027  |
|     003-018     |   2.6749  |    2.6242   |    0.2254   |   5.2328   |   0.2468  |
|     005-006     |   0.8971  |    0.5513   |    0.4482   |   3.8640   |   0.0905  |
|     005-014     |   0.6763  |    0.4174   |    0.3664   |   4.6431   |   0.1228  |
|     005-015     |   0.5715  |    0.3789   |    0.2767   |   3.0291   |   0.1906  |
|     005-016     |   0.6315  |    0.3356   |    0.3259   |   2.8971   |   0.2811  |
|     005-017     |   0.8074  |    0.4395   |    0.4487   |   1.6252   |   0.1933  |
|     005-018     |   1.0056  |    0.5869   |    0.4807   |   1.4734   |   0.1464  |
|     006-014     |   1.9139  |    1.8227   |    0.3347   |   6.3850   |   0.1546  |
|     006-015     |   2.0977  |    2.0015   |    0.3987   |   6.7674   |   0.1660  |
|     006-016     |   2.0538  |    1.9890   |    0.3385   |   6.1895   |   0.3352  |
|     006-017     |   2.2839  |    2.2043   |    0.4079   |   5.7890   |   0.1975  |
|     006-018     |   3.2556  |    3.1736   |    0.4174   |   6.0868   |   0.1328  |
|     014-015     |   0.6532  |    0.4432   |    0.2873   |   6.6693   |   0.2745  |
|     014-016     |   0.4565  |    0.3364   |    0.1481   |   1.2220   |   0.1611  |
|     014-017     |   0.4822  |    0.3202   |    0.2060   |   5.0176   |   0.1874  |
|     014-018     |   0.7336  |    0.5121   |    0.2807   |   4.7227   |   0.2558  |
|     015-016     |   0.3946  |    0.2174   |    0.2272   |   6.6745   |   0.3379  |
|     015-017     |   0.4187  |    0.2771   |    0.1867   |   2.9058   |   0.0515  |
|     015-018     |   0.6925  |    0.4477   |    0.3590   |   3.2500   |   0.1681  |
|     016-017     |   0.4613  |    0.3276   |    0.2033   |   2.6257   |   0.2598  |
|     016-018     |   0.6500  |    0.4887   |    0.2183   |   2.2078   |   0.3070  |
|     017-018     |   0.4626  |    0.2935   |    0.2233   |   1.9078   |   0.2943  |
|     Averages    |   1.0001  |    0.8351   |    0.2851   |   3.8525   |   0.2639  |
+-----------------+-----------+-------------+-------------+------------+-----------+
```

miguelriemoliveira, 2022-06-24:
Why @manuelgitgomes ,

why only 45 pairwise combinations? Because you need a detection on both sensors, right?

So we get 0.5 in the intra collection evaluation, and 1 pixel error in the cross collection evaluation?

These results are very plausible I think.

manuelgitgomes, 2022-06-24:
> why only 45 pairwise combinations? Because you need a detection on both sensors, right?

From the 18 collections, only 10 collections have a detection in both sensors.
I only considered one-way pairings, that is, if collection 000 is already paired with 002, then collection 002 does not need to be paired with collection 000.
To calculate the number of pairs in 10 objects I used [this website](https://www.calculatorsoup.com/calculators/discretemathematics/combinations.php), which gave me 45 pairs, the same number of rows in the results table.

> So we get 0.5 in the intra collection evaluation, and 1 pixel error in the cross collection evaluation?

Exactly!

miguelriemoliveira, 2022-06-24:
Forgot to say before: congratulations. The results look great.

BTW, the odom frame should be a command line argument, since its name may change from system to system. Called the fixed frame. In our case It is fixed by the odometry, but it could be fixed using slam, gps, or some other form of localization.

I can't resist to say now you should try cross collection evaluation also with cross sensor. 
It should be easily done, right? The table would be something like this:

```bash
+-----------------+-----------+-------------+-------------+------------+-----------+
| Collection Pair (sensor a - sensor b)|  RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
+-----------------+-----------+-------------+-------------+------------+-----------+
|     000-002     |   0.3042  |    0.1600   |    0.1830   |   1.5116   |   0.2103  |
|     002-000     |   0.3042  |    0.1600   |    0.1830   |   1.5116   |   0.2103  |
|     000-003     |   0.3838  |    0.2406   |    0.1815   |   4.1979   |   0.2644  |
|     000-005     |   0.6321  |    0.4290   |    0.2526   |   3.3498   |   0.4647  |
|     017-018     |   0.4626  |    0.2935   |    0.2233   |   1.9078   |   0.2943  |
|     Averages    |   1.0001  |    0.8351   |    0.2851   |   3.8525   |   0.2639  |
+-----------------+-----------+-------------+-------------+------------+-----------+
```

miguelriemoliveira, 2022-06-24:
> From the 18 collections, only 10 collections have a detection in both sensors.
> I only considered one-way pairings, that is, if collection 000 is already paired with 002, then collection 002 does not need to be paired with collection 000.

Yes, it makes sense since its the same sensor. But for the cross sensor results you will have to take into account the two-way pairing.

> To calculate the number of pairs in 10 objects I used [this website](https://www.calculatorsoup.com/calculators/discretemathematics/combinations.php), which gave me 45 pairs, the same number of rows in the results table.

This is done directly in python:

https://stackoverflow.com/questions/36429507/python-combinations-without-repetitions

we use it here:

https://github.com/lardemua/atom/blob/671e301310a0543ab91a99ba7c7ef9a4dbd33551/atom_core/src/atom_core/ros_utils.py#L45-L56

manuelgitgomes, 2022-06-24:
> BTW, the odom frame should be a command line argument, since its name may change from system to system. Called the fixed frame. In our case It is fixed by the odometry, but it could be fixed using slam, gps, or some other form of localization.

I used the world_link from the dataset, which should change accordingly to the fixation method. Should I change that to an argument?

> I can't resist to say now you should try cross collection evaluation also with cross sensor.

So, two different tables, one from sensor a - sensor b and one from sensor b - sensor a? Or the same table with the two-way pairings?

> This is done directly in python:
> 
> https://stackoverflow.com/questions/36429507/python-combinations-without-repetitions
> 
> we use it here:

Thanks!

manuelgitgomes, 2022-06-24:
Because I did not understood fully the assignment, I moved on to rgb_to_lidar_cross_collection_evaluation. I did a two way pairing and the results are as follows.

Top left camera to top lidar:

```bash
+-----------------+-----------+-------------+-------------+
| Collection Pair | RMS (pix) | X err (pix) | Y err (pix) |
+-----------------+-----------+-------------+-------------+
|     000-000     |   2.5749  |    2.1152   |    0.4102   |
|     000-001     |   3.3854  |    3.0071   |    0.3970   |
|     000-002     |   2.0879  |    1.5415   |    0.5651   |
|     000-003     |   3.0161  |    2.0960   |    0.5775   |
|     000-004     |   2.6198  |    1.9918   |    0.6493   |
|     000-005     |   3.5530  |    2.6806   |    0.3832   |
|     000-006     |   2.8362  |    2.0894   |    0.2417   |
|     000-009     |   4.8964  |    3.3294   |    0.1968   |
|     000-010     |   3.1457  |    2.7075   |    0.3203   |
|     000-011     |   4.1472  |    2.8363   |    0.7290   |
|     000-012     |   3.1471  |    2.4350   |    0.3004   |
|     000-014     |   2.8031  |    1.7659   |    0.5620   |
|     000-015     |   2.1316  |    1.4719   |    0.3491   |
|     000-016     |   2.7527  |    2.1783   |    0.2596   |
|     000-017     |   2.3809  |    2.1547   |    0.2966   |
|     000-018     |   3.1643  |    2.5129   |    0.4965   |
|     001-000     |   1.4295  |    1.1111   |    0.4329   |
|     001-001     |   1.8872  |    1.1928   |    0.6328   |
|     001-002     |   2.3088  |    1.6349   |    1.1105   |
|     001-003     |   3.0751  |    2.1652   |    0.6983   |
|     001-004     |   1.9452  |    1.5475   |    0.6509   |
|     001-005     |   3.4501  |    2.9224   |    0.2378   |
|     001-006     |   2.6451  |    2.0372   |    0.5265   |
|     001-009     |   3.1369  |    2.9523   |    0.4943   |
|     001-010     |   3.9072  |    3.1688   |    0.9022   |
|     001-011     |   2.2824  |    1.8470   |    0.3174   |
|     001-012     |   3.3300  |    2.6229   |    0.6827   |
|     001-014     |   1.0561  |    0.6030   |    0.5294   |
|     001-015     |   1.8981  |    1.4545   |    0.4855   |
|     001-016     |   2.0818  |    1.7142   |    0.6036   |
|     001-017     |   1.5199  |    1.0686   |    0.2471   |
|     001-018     |   1.1051  |    0.8827   |    0.3764   |
|     002-000     |   3.1723  |    2.3053   |    0.2787   |
|     002-001     |   4.2396  |    3.6785   |    0.2653   |
|     002-002     |   2.6904  |    1.9781   |    0.5289   |
|     002-003     |   2.0362  |    1.6377   |    0.2701   |
|     002-004     |   2.2900  |    1.9639   |    0.2639   |
|     002-005     |   2.4706  |    1.9345   |    0.2667   |
|     002-006     |   1.8468  |    1.3538   |    0.2913   |
|     002-009     |   6.4802  |    4.5633   |    0.5672   |
|     002-010     |   4.0134  |    3.3025   |    0.5768   |
|     002-011     |   4.9871  |    3.3139   |    0.7470   |
|     002-012     |   3.9102  |    2.8931   |    0.5157   |
|     002-014     |   3.1560  |    1.6708   |    0.6397   |
|     002-015     |   2.9165  |    1.7645   |    0.6270   |
|     002-016     |   3.7274  |    3.1753   |    0.3021   |
|     002-017     |   3.6366  |    2.7508   |    0.4531   |
|     002-018     |   4.7154  |    3.9333   |    0.3693   |
|     003-000     |   2.9560  |    2.1194   |    0.3000   |
|     003-001     |   3.9681  |    3.2934   |    0.2366   |
|     003-002     |   2.2711  |    1.6646   |    0.3965   |
|     003-003     |   1.7211  |    1.2775   |    0.2802   |
|     003-004     |   1.9389  |    1.5563   |    0.2856   |
|     003-005     |   2.3035  |    1.6921   |    0.3459   |
|     003-006     |   1.9732  |    1.4386   |    0.4600   |
|     003-009     |   6.2493  |    4.2092   |    0.5598   |
|     003-010     |   3.8711  |    3.1537   |    0.6965   |
|     003-011     |   4.8555  |    3.1717   |    1.0822   |
|     003-012     |   3.7179  |    2.8644   |    0.5853   |
|     003-014     |   3.1349  |    1.8166   |    0.6143   |
|     003-015     |   2.5401  |    1.4357   |    0.7698   |
|     003-016     |   3.3063  |    2.8042   |    0.2349   |
[... 286 lines cut ...]
|     017-000     |   2.1947  |    1.5168   |    0.7426   |
|     017-002     |   2.1857  |    1.4265   |    0.5222   |
|     017-003     |   2.9447  |    2.1761   |    0.6527   |
|     017-005     |   3.2984  |    2.4972   |    0.4780   |
|     017-006     |   2.1353  |    1.5502   |    0.2699   |
|     017-014     |   2.5259  |    1.1018   |    0.6523   |
|     017-015     |   1.7889  |    1.4954   |    0.3900   |
|     017-016     |   1.8060  |    1.0751   |    0.2820   |
|     017-017     |   1.3900  |    1.2137   |    0.2609   |
|     017-018     |   2.3416  |    1.9884   |    0.5841   |
|     018-000     |   3.1689  |    2.3827   |    0.5360   |
|     018-002     |   3.5577  |    1.8805   |    0.9645   |
|     018-003     |   4.6248  |    3.2088   |    1.0104   |
|     018-005     |   4.5790  |    3.5575   |    0.3247   |
|     018-006     |   3.0346  |    2.3410   |    0.2991   |
|     018-014     |   3.9387  |    2.2407   |    1.1026   |
|     018-015     |   2.8802  |    2.4673   |    0.8390   |
|     018-016     |   2.2911  |    1.4353   |    0.5309   |
|     018-017     |   1.8874  |    1.6433   |    0.3924   |
|     018-018     |   2.9215  |    2.4190   |    0.6632   |
|     Averages    |   2.5997  |    1.8783   |    0.5340   |
+-----------------+-----------+-------------+-------------+
```

This RMS of 2.60 is slightly high than the intra-collection one, which is 2.03.

manuelgitgomes, 2022-06-24:
I also changed the rgb/rgb to two-way pairing, with results:

```bash
+-----------------+-----------+-------------+-------------+------------+-----------+
| Collection Pair | RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
+-----------------+-----------+-------------+-------------+------------+-----------+
|     000-000     |   0.2556  |    0.1555   |    0.1182   |   1.9794   |   0.4632  |
|     000-002     |   0.3042  |    0.1600   |    0.1830   |   1.5116   |   0.2103  |
|     000-003     |   0.3838  |    0.2406   |    0.1815   |   4.1979   |   0.2644  |
|     000-005     |   0.6321  |    0.4290   |    0.2526   |   3.3498   |   0.4647  |
|     000-006     |   0.5249  |    0.3388   |    0.2581   |   5.8086   |   0.3474  |
|     000-014     |   0.6440  |    0.4820   |    0.2624   |   4.6392   |   0.4433  |
|     000-015     |   0.6130  |    0.3767   |    0.2990   |   3.4351   |   0.3694  |
|     000-016     |   0.3734  |    0.2591   |    0.1458   |   2.5066   |   0.5439  |
|     000-017     |   0.4135  |    0.2771   |    0.1828   |   2.2199   |   0.4765  |
|     000-018     |   0.5001  |    0.3579   |    0.2145   |   1.6849   |   0.2866  |
|     002-000     |   0.9228  |    0.8438   |    0.2388   |   2.8810   |   0.5527  |
|     002-002     |   0.3957  |    0.2494   |    0.1765   |   3.1672   |   0.0400  |
|     002-003     |   0.4981  |    0.2861   |    0.2749   |   0.9229   |   0.0874  |
|     002-005     |   0.6568  |    0.4446   |    0.3053   |   1.3953   |   0.2871  |
|     002-006     |   0.4875  |    0.2430   |    0.2864   |   1.2257   |   0.1697  |
|     002-014     |   1.0808  |    0.9593   |    0.3654   |   6.2306   |   0.2662  |
|     002-015     |   1.2378  |    1.0909   |    0.4107   |   3.6374   |   0.2381  |
|     002-016     |   1.1494  |    1.0770   |    0.2283   |   4.9903   |   0.3778  |
|     002-017     |   1.3070  |    1.2080   |    0.3094   |   2.9241   |   0.3133  |
|     002-018     |   1.7980  |    1.7099   |    0.2929   |   3.3141   |   0.1429  |
|     003-000     |   1.3486  |    1.3062   |    0.1673   |   4.8308   |   0.4520  |
|     003-002     |   0.5373  |    0.3972   |    0.2069   |   5.0997   |   0.1362  |
|     003-003     |   0.5176  |    0.3861   |    0.1302   |   2.5821   |   0.1747  |
|     003-005     |   0.7665  |    0.5386   |    0.3367   |   3.3329   |   0.3540  |
|     003-006     |   0.5511  |    0.3741   |    0.2507   |   0.7832   |   0.2129  |
|     003-014     |   1.5591  |    1.4894   |    0.1892   |   7.3969   |   0.3358  |
|     003-015     |   1.6889  |    1.5852   |    0.3818   |   5.3273   |   0.3431  |
|     003-016     |   1.7116  |    1.6697   |    0.1964   |   6.5460   |   0.3811  |
|     003-017     |   1.8491  |    1.7939   |    0.2115   |   4.7951   |   0.4027  |
|     003-018     |   2.6749  |    2.6242   |    0.2254   |   5.2328   |   0.2468  |
|     005-000     |   0.6853  |    0.4216   |    0.3516   |   1.0596   |   0.6718  |
|     005-002     |   0.7615  |    0.4091   |    0.3663   |   0.9425   |   0.1254  |
|     005-003     |   0.8055  |    0.4125   |    0.4352   |   2.4781   |   0.0972  |
|     005-005     |   0.9208  |    0.4721   |    0.4350   |   1.5703   |   0.1420  |
|     005-006     |   0.8971  |    0.5513   |    0.4482   |   3.8640   |   0.0905  |
|     005-014     |   0.6763  |    0.4174   |    0.3664   |   4.6431   |   0.1228  |
|     005-015     |   0.5715  |    0.3789   |    0.2767   |   3.0291   |   0.1906  |
|     005-016     |   0.6315  |    0.3356   |    0.3259   |   2.8971   |   0.2811  |
|     005-017     |   0.8074  |    0.4395   |    0.4487   |   1.6252   |   0.1933  |
|     005-018     |   1.0056  |    0.5869   |    0.4807   |   1.4734   |   0.1464  |
|     006-000     |   1.9375  |    1.8779   |    0.2795   |   5.6590   |   0.6818  |
|     006-002     |   1.2945  |    1.2231   |    0.1972   |   5.7536   |   0.1184  |
|     006-003     |   1.3686  |    1.2907   |    0.2555   |   3.9457   |   0.1088  |
|     006-005     |   1.0687  |    0.9001   |    0.2317   |   4.3639   |   0.1746  |
|     006-006     |   0.8880  |    0.8272   |    0.1871   |   2.5806   |   0.1421  |
|     006-014     |   1.9139  |    1.8227   |    0.3347   |   6.3850   |   0.1546  |
|     006-015     |   2.0977  |    2.0015   |    0.3987   |   6.7674   |   0.1660  |
|     006-016     |   2.0538  |    1.9890   |    0.3385   |   6.1895   |   0.3352  |
|     006-017     |   2.2839  |    2.2043   |    0.4079   |   5.7890   |   0.1975  |
|     006-018     |   3.2556  |    3.1736   |    0.4174   |   6.0868   |   0.1328  |
|     014-000     |   1.1236  |    1.0665   |    0.1343   |   4.7300   |   0.7543  |
|     014-002     |   1.8724  |    1.8288   |    0.2376   |   4.2195   |   0.2644  |
|     014-003     |   1.9085  |    1.8500   |    0.2237   |   5.8839   |   0.2156  |
|     014-005     |   2.7034  |    2.6434   |    0.4209   |   5.2658   |   0.0755  |
|     014-006     |   3.0255  |    2.9728   |    0.3850   |   6.7666   |   0.1285  |
|     014-014     |   0.5643  |    0.3938   |    0.2263   |   1.0497   |   0.0839  |
|     014-015     |   0.6532  |    0.4432   |    0.2873   |   6.6693   |   0.2745  |
|     014-016     |   0.4565  |    0.3364   |    0.1481   |   1.2220   |   0.1611  |
|     014-017     |   0.4822  |    0.3202   |    0.2060   |   5.0176   |   0.1874  |
|     014-018     |   0.7336  |    0.5121   |    0.2807   |   4.7227   |   0.2558  |
[... cut ...]
|     018-015     |   0.4785  |    0.2037   |    0.2866   |   1.5376   |   0.2659  |
|     018-016     |   0.2580  |    0.1452   |    0.1421   |   4.3560   |   0.2104  |
|     018-017     |   0.3253  |    0.1824   |    0.1833   |   0.8003   |   0.1471  |
|     018-018     |   0.3184  |    0.2345   |    0.0852   |   1.2571   |   0.2727  |
|     Averages    |   0.9306  |    0.7792   |    0.2535   |   3.5163   |   0.2653  |
+-----------------+-----------+-------------+-------------+------------+-----------+

```

miguelriemoliveira, 2022-06-24:
> Another question. Should the PnP be calculated only for the source sensor or for both? If both, should a table be double the size of the number of permutations (2x10x10=200), half of it calculated with one PnP and the other with another?

We have the same dilema in the standard rgb to rgb evaluation, and have decided to choose only one sensor.

manuelgitgomes, 2022-06-24:
> We have the same dilema in the standard rgb to rgb evaluation, and have decided to choose only one sensor.

Ok, so I do not understand what is asked when the professor asks for a cross-sensor evaluation!

miguelriemoliveira, 2022-06-24:
> Because I did not understood fully the assignment, I moved on to rgb_to_lidar_cross_collection_evaluation. I did a two way pairing and the results are as follows.
> 
> Top left camera to top lidar:

Hi @manuelgitgomes ,

results also look very good. Congrats.

miguelriemoliveira, 2022-06-24:
> > We have the same dilema in the standard rgb to rgb evaluation, and have decided to choose only one sensor.
> 
> Ok, so I do not understand what is asked when the professor asks for a cross-sensor evaluation!

Hi @manuelgitgomes ,

I mean we select one sensor to use the PNP.

manuelgitgomes, 2022-06-24:
> Hi @manuelgitgomes ,
> 
> I mean we select one sensor to use the PNP.

Sorry, can I call you?

miguelriemoliveira, 2022-06-24:
I will call you in a moment ok?
*  Miguel Riem de Oliveira                                             *
Diretor de curso Mestrado Eng. Automação Industrial
email:    ***@***.***    |    ***@***.***
zoom: https://videoconf-colibri.zoom.us/j/7155402102
office: 22.2.18                 |    ext: 23887
Dep. de Engenharia Mecânica, Universidade de Aveiro
*Campus Universitário de Santiago, 3810-193  Portugal*

manuelgitgomes, 2022-06-27:
Created lidar to lidar cross collection, but something is weird with the results.
Here are the results for the simulation dataset without anchor:

```bash
+-----------------+---------+-----------+-----------+-------------+-------------+
| Collection Pair | RMS (m) | X err (m) | Y err (m) | X StDev (m) | Y StDev (m) |
+-----------------+---------+-----------+-----------+-------------+-------------+
|     000-001     |  0.0093 |   0.0049  |   0.0053  |    0.0040   |    0.0039   |
|     000-002     |  0.0341 |   0.0056  |   0.0055  |    0.0041   |    0.0043   |
|     000-003     |  0.0355 |   0.0056  |   0.0054  |    0.0041   |    0.0041   |
|     000-004     |  0.0351 |   0.0058  |   0.0055  |    0.0044   |    0.0042   |
|     000-005     |  0.0455 |   0.0052  |   0.0053  |    0.0041   |    0.0041   |
|     000-006     |  0.0447 |   0.0050  |   0.0050  |    0.0040   |    0.0037   |
|     000-009     |  0.0426 |   0.0053  |   0.0049  |    0.0038   |    0.0035   |
|     000-010     |  0.0421 |   0.0052  |   0.0049  |    0.0041   |    0.0036   |
|     000-011     |  0.0416 |   0.0050  |   0.0054  |    0.0037   |    0.0039   |
|     000-012     |  0.0425 |   0.0047  |   0.0053  |    0.0036   |    0.0041   |
|     000-014     |  0.0418 |   0.0050  |   0.0059  |    0.0038   |    0.0040   |
|     000-015     |  0.0425 |   0.0053  |   0.0052  |    0.0042   |    0.0040   |
|     000-016     |  0.0195 |   0.0056  |   0.0052  |    0.0042   |    0.0040   |
|     000-017     |  0.0197 |   0.0050  |   0.0056  |    0.0040   |    0.0039   |
|     000-018     |  0.0523 |   0.0051  |   0.0053  |    0.0038   |    0.0039   |
|     001-002     |  0.0340 |   0.0057  |   0.0060  |    0.0047   |    0.0055   |
|     001-003     |  0.0355 |   0.0058  |   0.0059  |    0.0043   |    0.0053   |
|     001-004     |  0.0349 |   0.0057  |   0.0058  |    0.0050   |    0.0044   |
|     001-005     |  0.0454 |   0.0063  |   0.0055  |    0.0053   |    0.0043   |
|     001-006     |  0.0447 |   0.0063  |   0.0055  |    0.0055   |    0.0042   |
|     001-009     |  0.0428 |   0.0059  |   0.0051  |    0.0053   |    0.0038   |
|     001-010     |  0.0423 |   0.0060  |   0.0050  |    0.0054   |    0.0040   |
|     001-011     |  0.0418 |   0.0057  |   0.0055  |    0.0048   |    0.0044   |
|     001-012     |  0.0427 |   0.0052  |   0.0058  |    0.0040   |    0.0050   |
|     001-014     |  0.0423 |   0.0061  |   0.0060  |    0.0055   |    0.0046   |
|     001-015     |  0.0428 |   0.0055  |   0.0061  |    0.0045   |    0.0056   |
|     001-016     |  0.0192 |   0.0054  |   0.0051  |    0.0042   |    0.0044   |
|     001-017     |  0.0199 |   0.0050  |   0.0061  |    0.0040   |    0.0054   |
|     001-018     |  0.0522 |   0.0057  |   0.0064  |    0.0043   |    0.0060   |
|     002-003     |  0.0100 |   0.0062  |   0.0049  |    0.0048   |    0.0038   |
|     002-004     |  0.0095 |   0.0059  |   0.0046  |    0.0046   |    0.0035   |
|     002-005     |  0.0376 |   0.0059  |   0.0046  |    0.0043   |    0.0033   |
|     002-006     |  0.0375 |   0.0063  |   0.0048  |    0.0047   |    0.0036   |
|     002-009     |  0.0388 |   0.0061  |   0.0052  |    0.0057   |    0.0036   |
|     002-010     |  0.0387 |   0.0057  |   0.0056  |    0.0052   |    0.0041   |
|     002-011     |  0.0382 |   0.0046  |   0.0059  |    0.0043   |    0.0045   |
|     002-012     |  0.0392 |   0.0039  |   0.0061  |    0.0036   |    0.0047   |
|     002-014     |  0.0374 |   0.0045  |   0.0060  |    0.0032   |    0.0043   |
|     002-015     |  0.0375 |   0.0052  |   0.0062  |    0.0039   |    0.0047   |
|     002-016     |  0.0320 |   0.0056  |   0.0064  |    0.0049   |    0.0065   |
|     002-017     |  0.0330 |   0.0055  |   0.0069  |    0.0045   |    0.0070   |
|     002-018     |  0.0449 |   0.0050  |   0.0058  |    0.0037   |    0.0043   |
|     003-004     |  0.0093 |   0.0054  |   0.0049  |    0.0046   |    0.0032   |
|     003-005     |  0.0377 |   0.0064  |   0.0045  |    0.0048   |    0.0034   |
|     003-006     |  0.0376 |   0.0059  |   0.0048  |    0.0047   |    0.0037   |
|     003-009     |  0.0387 |   0.0061  |   0.0053  |    0.0049   |    0.0038   |
|     003-010     |  0.0387 |   0.0057  |   0.0056  |    0.0051   |    0.0041   |
|     003-011     |  0.0382 |   0.0047  |   0.0061  |    0.0038   |    0.0049   |
|     003-012     |  0.0393 |   0.0044  |   0.0066  |    0.0036   |    0.0055   |
|     003-014     |  0.0376 |   0.0052  |   0.0061  |    0.0039   |    0.0046   |
|     003-015     |  0.0375 |   0.0047  |   0.0065  |    0.0035   |    0.0048   |
|     003-016     |  0.0319 |   0.0048  |   0.0073  |    0.0040   |    0.0074   |
|     003-017     |  0.0330 |   0.0050  |   0.0067  |    0.0037   |    0.0078   |
|     003-018     |  0.0449 |   0.0049  |   0.0062  |    0.0036   |    0.0050   |
|     004-005     |  0.0377 |   0.0066  |   0.0049  |    0.0052   |    0.0036   |
|     004-006     |  0.0376 |   0.0066  |   0.0049  |    0.0053   |    0.0038   |
|     004-009     |  0.0386 |   0.0062  |   0.0052  |    0.0050   |    0.0040   |
|     004-010     |  0.0386 |   0.0056  |   0.0054  |    0.0049   |    0.0039   |
|     004-011     |  0.0382 |   0.0052  |   0.0062  |    0.0040   |    0.0047   |
|     004-012     |  0.0393 |   0.0044  |   0.0066  |    0.0034   |    0.0058   |
|     004-014     |  0.0373 |   0.0051  |   0.0062  |    0.0037   |    0.0050   |
|     004-015     |  0.0373 |   0.0049  |   0.0065  |    0.0037   |    0.0047   |
[... cut ...]
|     017-018     |  0.0484 |   0.0061  |   0.0064  |    0.0043   |    0.0058   |
|     Averages    |  0.0364 |   0.0054  |   0.0061  |    0.0043   |    0.0051   |
+-----------------+---------+-----------+-----------+-------------+-------------+
```

The results are very similar, although not equal. On the anchored sensor dataset, should the values be all 0? Because the sensor is placed in the correct spot. Or does this have to do with #461? The error can be inherent to the poses and offsets of the beams, leading to innacurate results.

manuelgitgomes, 2022-06-27:
Hello @miguelriemoliveira! This is same sensor cross collection evaluation!

miguelriemoliveira, 2022-06-27:
> Hello @miguelriemoliveira! This is same sensor cross collection evaluation!

and was this sensor the anchored sensor or was it another one?

manuelgitgomes, 2022-06-27:
On the first results, no, but on the second ones, yes

miguelriemoliveira, 2022-06-27:
Can we talk 5 mins?

manuelgitgomes, 2022-07-04:
Closed by #502

closed 2022-07-04 by manuelgitgomes