lardemua/atom #496: Evaluation scripts retrieving the intrinsics of test dataset

manuelgitgomes, 2022-06-23:
In evaluation. the intrinsic parameters used are the ones from the test dataset and not the ones optimized from the train dataset.
Should this be like this?
https://github.com/lardemua/atom/blob/60f89351f7d518f107c35036eb29a45551621681/atom_evaluation/scripts/rgb_to_rgb_evaluation#L208-L211

miguelriemoliveira, 2022-06-23:
Hi @manuelgitgomes ,

actually they are copied from the train to the test dataset some lines above:

https://github.com/lardemua/atom/blob/60f89351f7d518f107c35036eb29a45551621681/atom_evaluation/scripts/rgb_to_rgb_evaluation#L188-L194

Actually I don't like this way of doing it, because its confusing.

We should have a train dataset, a test dataset, and a mixed dataset.
This way its more explicit.

I did a function to produce the mixed dataset from the train and test datasets.
It is defined here:

https://github.com/lardemua/atom/blob/671e301310a0543ab91a99ba7c7ef9a4dbd33551/atom_core/src/atom_core/dataset_io.py#L668-L706

and used here

https://github.com/lardemua/atom/blob/671e301310a0543ab91a99ba7c7ef9a4dbd33551/atom_evaluation/scripts/depth_to_rgb_evaluation#L64-L73

and here

https://github.com/lardemua/atom/blob/671e301310a0543ab91a99ba7c7ef9a4dbd33551/atom_evaluation/scripts/lidar_to_rgb_evaluation#L91-L100

Once I have time I will put all evaluation scripts using this method (#497).

If you can you should also use this method.

manuelgitgomes, 2022-06-23:
Ok, thank you!

miguelriemoliveira, 2022-06-24:
Not sure what you mean. I will call you later.

manuelgitgomes, 2022-06-24:
Remove train_dataset and test_dataset after mixed is created

closed 2023-03-31 by miguelriemoliveira