lardemua/atom #498: Too many measurements at collect data step

sensorCalib, 2022-06-24:
Hi,

at step 4. when trying to collect data, we seem to get a dimension mismatch. This is the error occuring:

`[ERROR] [1656064461.734798, 1656052507.736720]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7f3929d15d60>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/user/ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 295, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/user/ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 464, in labelData
    self.labels, seed_point, inliers = labelPointCloud2Msg(self.msg, x_marker, y_marker, z_marker,
  File "/home/user/ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/label_messages.py", line 45, in labelPointCloud2Msg
    points[:, 0] = pc['x'].flatten()  # flatten because some pcs are of shape (npoints,1) rather than (npoints,)
ValueError: could not broadcast input array from shape (65536) into shape (64)`

We have 64 lines with 1024 measurement-values each, which results in the 65536 measurements.
However, only one measurment per coordinate is expected.

Any idea how to fix this? Thank you very much.

miguelriemoliveira, 2022-06-24:
Hello,

Before we help, could you please provide more information? What robotic system are you calibrating? How many sensors? Of which modality?

It seems you have just created this account in github. Also, you do not identify yourself. If you could provide some more information that would be very helpful.

sensorCalib, 2022-06-27:
Hello,

thanks for the feedback, I'll gladly provide you with more information.

Robotic system: ROS Noetic, working with noetic-devel branch
How many sensors: two (ouster os1, zed 2i) The error happens with the ouster.
Which modality: ouster os1 has modality lidar3d, zed 2i has modality rgb

I am a commercial user and working on a research project about environment detection.

miguelriemoliveira, 2022-06-27:
Hi @sensorCalib ,

I have never worked with this sensor before, but it should be very similar to velodyne.

To investigate I would need your calibration.yml file (or a link to your calibration repository) and also a bag file with the sensor data. IT can be just a small 1 min bag or something.

Could you provide this please?

miguelriemoliveira, 2022-07-01:
Hi @sensorCalib ,

I have downloaded the bag and yml. Thanks. 

SI created a calibration package of my own since you did not share yours, but now to configure the calibration package I need this xacro file:

"/home/user/ws/src/base/description/urdf/system.urdf.xacro"

Can you provide it?

Thanks

miguelriemoliveira, 2022-07-01:
I have compared the ouster lidar ros messages that you have 

```
header: 
  seq: 1284
  stamp: 
    secs: 1656052510
    nsecs: 181796381
  frame_id: "ouster_os1_0_base_link"
height: 64
width: 1024
fields: "<array type: sensor_msgs/PointField, length: 9>"
is_bigendian: False
point_step: 48
row_step: 49152
data: "<array type: uint8, length: 3145728>"
is_dense: True
```

with the ones we use from velodynes:

```
header: 
  seq: 6319
  stamp: 
    secs: 1649240013
    nsecs: 829686165
  frame_id: "lidar_1"
height: 1
width: 26808
fields: "<array type: sensor_msgs/PointField, length: 6>"
is_bigendian: False
point_step: 22
row_step: 589776
data: "<array type: uint8, length: 589776>"
is_dense: True
```

I think the problem is, as you discussed above, the fact that the ouster point cloud is not flattened (i.e. height=1) as is the velodyne point cloud.

Let me investigate further.

miguelriemoliveira, 2022-07-01:
Hi @sensorCalib ,

I changed the label messages code to be able to handle both velodynes and the ouster lidars.

In the standalone test I did it works: 

https://github.com/lardemua/atom/blob/noetic-devel/atom_calibration/scripts/tests/test_ouster_lidar

Please test and let me know if it runs. 

I suspect we will have additional problems with the ouster lidar in some other places, but without the xacro file I cannot test the complete pipeline.

sensorCalib, 2022-07-04:
Hi @miguelriemoliveira ,

thank you very much for your help so far. I've retried the collect data step with the fix you added and I am getting past that bug mentioned above. As you suspected, there is an error further down the line:

`[ERROR] [1656944209.650737, 1656052507.832743]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7f7fb9fcfdf0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/martelv/agri-gaia-catkin-ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 295, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/martelv/agri-gaia-catkin-ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 464, in labelData
    self.labels, seed_point, inliers = labelPointCloud2Msg(self.msg, x_marker, y_marker, z_marker,
  File "/home/martelv/agri-gaia-catkin-ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/label_messages.py", line 156, in labelPointCloud2Msg
    x, y, z = pc['x'][idx], pc['y'][idx], pc['z'][idx]
IndexError: index 64 is out of bounds for axis 0 with size 64`

I have included a urdf file with the data from my xacro files, hope this helps:
https://www.dropbox.com/scl/fo/o40e4s4vu32bpn99p1j2e/h?dl=0&rlkey=0ugua76pvhx1jsnxwbvanyz6p

sensorCalib, 2022-07-12:
To get around that crash in line 156, I flattened the pc variable. So instead of
`x, y, z = pc['x'][idx], pc['y'][idx], pc['z'][idx]`

I am using

`x, y, z = pc['x'].flatten()[idx], pc['y'].flatten()[idx], pc['z'].flatten()[idx]`

now. It's running without a crash for a while, but the outout dir defined in the command-line parameter only is an empty dir.
Eventually, the data collection will fail again
`
[ERROR] [1657628706.062701, 1656052506.920389]: bad callback: <bound method InteractiveDataLabeler.sensorDataReceivedCallback of <atom_calibration.collect.interactive_data_labeler.InteractiveDataLabeler object at 0x7fcd9e6ef6a0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/user/ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 295, in sensorDataReceivedCallback
    self.labelData()  # label the data
  File "/home/user/ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/interactive_data_labeler.py", line 464, in labelData
    self.labels, seed_point, inliers = labelPointCloud2Msg(self.msg, x_marker, y_marker, z_marker,
  File "/home/user/ws/src/atom-calibration/atom_calibration/src/atom_calibration/collect/label_messages.py", line 133, in labelPointCloud2Msg
    distances = abs((A * pts[:, 0] + B * pts[:, 1] + C * pts[:, 2] + D)) / \
UnboundLocalError: local variable 'A' referenced before assignment
`

A will be used for RANSAC iterations, but it will only be initialized if "the number of inliers is larger than the previous max".

miguelriemoliveira, 2022-07-13:
Hi @sensorCalib ,

thank you for the urdf. However, the urdf calls for several stl files that you did not share.

Let me say that this is not my preferred  way of working. You clearly created a github account just for creating this issue, which appears to have the clear intent of hiding your identity or that of your company, and now you are providing information drop by drop, which makes progress very difficult.

Why don't you share the repository? If its private, create a temporary access for me since you are asking for my help.

As you've seen before, I am willing to help, although I am taking a bit of time to respond. But if you want my help to make ATOM work in your system then we must have mutual trust.

sensorCalib, 2022-07-15:
Hi @miguelriemoliveira ,
thank you very much again for you help and please excuse that way of communication. I indeed tried to not reveal more information than necessary, but you are absolutely right: if I am asking for your help, there should be some mutual trust. I have talked to my superior about this and got permission to share more information.

So, about my identity and the project I am working on: I am a developer from KRONE, an agricultural company from Germany. As a part of the Agri-Gaia project, we are working on several projects in the AI field, environmental detection being one of them:
https://www.agri-gaia.de/

Please feel free to ask for more information if you like.

About the repository: our repositories can only be accessed from our companies network, so granting you access isn't done easily. Would it be alright to provide you our complete catkin-ws instead? This should contain all necessary files.

https://www.dropbox.com/s/bnqf0ldd2fzloxs/agri-gaia-catkin-ws.tar.gz?dl=0

./src/atom-result/calib_test1 contains the tried calibration, including the calib-file
./data/rosbags contains the bag-file with the pattern you already saw
./data/data-sets/calib_test1_out contains the (empty) output from the collect data step

Again, thank you very much for your patience.

miguelriemoliveira, 2022-07-16:
Hi @sensorCalib ,

ok, thanks for the response. I will try to figure out what's going wrong and get back to you.

miguelriemoliveira, 2022-07-16:
Hi @sensorCalib ,

so I was working on this today and was able to advance a bit.

I created a new private repo to which I invited you. 

https://github.com/miguelriemoliveira/agri-gaia-test

Then fixed a few issues while trying to calibrate.
The crash you were reporting was fixed and does not occur anymore.

However, to test, now I am gonna need a bigger bag file (30 secs at least), because only 4 secs is too few.

Some tips on doing this bagfile:

[image]

1. You placed the chessboard on the ground. This way it will not be possible to detect the pattern in the lidar data. The pattern must be away from the ground, e.g. try to put it on top of a bench or a tripod.
2. Can you provide a dae file of the chessboard? Or at least its physical dimensions (width, height) so I can do a chessboard digital model? Right now we are using a model for the charuco while using the chessboard, which does not make sense.

https://github.com/miguelriemoliveira/agri-gaia-test/blob/5d16d57106456d22a2fd656fdbfea53a3c77fff8/agri-gaia_calibration/calibration/config.yml#L126

Thanks,
Miguel

sensorCalib, 2022-07-19:
Hi @miguelriemoliveira 

thank you very much for fixing the crashes. (And also fixing the the problem with the config file being None. I have noticed this bug occasionally, but I thought it was a mistake on my setup or something like that.)

1. A longer bag file has been uploaded here:
https://drive.google.com/drive/folders/1cufWx50qeXVnndhDZpJkdV8986wDW84x?usp=sharing
The pattern was standing slightly above the ground. However, since the sensor is now mounted on top of the machine, the total distance might have been to long. Also, the pattern is just laying; would it be favoruable to move the pattern around during the recording?
Or is that recording alright?

2. I have pushed a dae file of the chessboard to the test repo, is this sufficient? In case it isn't, the measurments are
Length: 8*10cm (square) + 2*5cm (border) = 90cm (total length)
Width: 11*10cm (square) + 2*5cm (border) = 120cm (total width)

I have retried the collect_data step with the pushed pattern and the longer recording: there are no crashes so far, but the process won't end and no ATOM dataset will be created.

miguelriemoliveira, 2022-07-20:
Hi @sensorCalib ,

thanks. I will try to pick this up on Friday and get back to you.

miguelriemoliveira, 2022-07-22:
Hi @sensorCalib ,

> The pattern isn't standing slightly above the ground now. However, since the sensor is now mounted on top of the machine, the total distance might have been to long. 

Right, this makes it impossible to segment the pattern automatically. But I will try to use the manual segmentation and see if it works.

[image]

> Also, the pattern is just laying; would it be favoruable to move the pattern around during the recording?

Yes, you have to move it around to get several collections. Not sure if you have done so, but read through the documenttion and check the ATOM youtube playlist for examples.

https://lardemua.github.io/atom_documentation/

https://www.youtube.com/watch?v=BYs1-H9vh0s&list=PLQN09mzV5mbI4h5IQt3Eu9kugSk-08mnY

If you have this questions having read the documentation then I should try to improve it.

> Or is that recording alright?

No, but don't despair ... I am always saying to my students that no bagfile is ok until you've recorded 10 or 15 others before :-)
Right now I think I can use it just to see if all the pipeline is working. Later you can try to calibrate with a better bagfile.

miguelriemoliveira, 2022-07-23:
So I traced down the problem (one of them) to the fact that this OS lidar, unlike the velodyne we use, outputs dense point clouds.

I will try to adjust the data labeller to tackle these as well.

miguelriemoliveira, 2022-07-24:
Hi @sensorCalib ,

There was a lot of work to do as you can see above, but now I think the collector is working fine also for ouster lidars.
You can see in #513 that the lidar auto detection is now working.

As we discussed, in your bag file the pattern is too close to the ground, so the detection will not be accurate.

Then I tried  to manually edit the lidar labels using the dataset playback functionality 

https://lardemua.github.io/atom_documentation/procedures/#dataset-playback

Right now I have another bug because also this script was not prepared to tackle dense point clouds (#520).
Will work on this.

miguelriemoliveira, 2022-07-24:
So I fixed #520 so now you can use dataset playback to manually label the chess.

You can try it out if you want. Don't forget to reconfigure your calibration package.

You should use the github I set up for agri gaia 

https://github.com/miguelriemoliveira/agri-gaia-test

If you give me a better bagfile I can try to calibrate. 

Some important tips for the bagfile:

1. The bag should contain the pattern away from the ground, we use the tripod for this.

2. Also, the pattern is too small / far away from the robot. It is not possible to detect the chess corners in the camera this way.
You have to put the pattern closer or get a bigger one. For large robots like cars we normally use a 800x600 mm pattern like this: 

 https://calib.io/collections/products/products/charuco-targets?variant=9400455004207

You can try the test detector script at

https://github.com/lardemua/atom/blob/noetic-devel/atom_calibration/scripts/deprecated/view_pattern.py

BTW, did you put the correct number of corners in the calibration.yaml?

3. Finally, you should move the pattern to be able to save collections with the pattern in different places.

miguelriemoliveira, 2022-07-26:
Hi @sensorCalib ,

> Could you explain, why you changed the child and parent links name for the zed camera?

As you had before ATOM should  also be able to calibrate.

In any case, I changed just because of following the ROS convention.
Note that ROS uses a x forward, y left, z up convention for the coordinate frames. The exception is the frames of camera sensors, that typically have a reference frame as per the ROS convention, and then have an optical frame that follows camera conventions, z forward, x right, y down.

So the transformation between the camera frame and the camera optical frame is just a matter of rotating to have the:

new z along the previous x
new x as the previous -y
new y as the previous -z

This is a standard transformation, and it does not make sense to change it. That's why I prefer to configure ATOM to change the 
upper level transformation instead.

[image]

Think of it as telling ATOM to calibrate the left_camera_frame w.r.t the camera_center, instead of telling ATOM to calibrate the left_camera_optical_frame w.r.t the left_camera_frame, which would mess up a standard transformation.

sensorCalib, 2022-07-27:
Hi @miguelriemoliveira 
thanks for that explanation.

> You can try the test detector script at
>
> https://github.com/lardemua/atom/blob/noetic-devel/atom_calibration/scripts/deprecated/view_pattern.py

Could you explain how to use this tool? This script accepts the following arguments:
 [-t {charuco,chessboard}] [-d DICT] -x NUM_X -y NUM_Y -L LENGTH [-l INNER_LENGTH] topic
In my case, I would call
-t chessboard,
-x 12 (since there are 12 edges in x direction)
 -y 9 (since there are 9 edges in y direction)
-L 100 (since each edge has a length of 100mm ?)
-topic (don't know which topic to use)

> As for https://github.com/lardemua/atom/issues/514
> Apparently, I have written the wrong frame name for the sensor data. I will have a look at this.
Identified the problem, but not fixed yet.

> Some important tips for the bagfile:
>
>    The bag should contain the pattern away from the ground, we use the tripod for this.
>
>    Also, the pattern is too small / far away from the robot. It is not possible to detect the chess corners in the camera this way.
>    You have to put the pattern closer or get a bigger one.

A new bagfile in which the pattern is away from the ground, closer to the sensor and moving has been made. But as it seems, the data collect step still won't collect data. Neither for the zed nor for the ouster. 
As mentioned in the description
https://lardemua.github.io/atom_documentation/procedures/#rgb-camera-labeling
"You can check if the detection is working by observing the overlays of top of the images."
the rgb-camera-labeling has automatic patttern detection, so I expected some blinking on the squares of the pattern from the zed camera or something like that, but I didn't notice any of that.

I then had a look again at the getting started and tried to setup everything again in a clean workspace. So after cloning the atom repo, I also tried to install the requirements.txt (something I didn't do before). This failed since ros_numpy apparently doesn't exist: 
`ERROR: Could not find a version that satisfies the requirement ros_numpy (from -r requirements.txt (line 11)) (from versions: none)
ERROR: No matching distribution found for ros_numpy (from -r requirements.txt (line 11))`
So, I did not install ros_numpy and went on, but somehow, this was enough to mess up my setup and now the data collect step crashes again:

`roslaunch agri_gaia_calibration collect_data.launch output_folder:=~/catkin_ws/data/agri_gaia_calibration_out
... logging to /home/martelv/.ros/log/ca43b19a-0da9-11ed-90f0-9f9d37bb287b/roslaunch-dev3vm-970975.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dev3vm:44761/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /use_sim_time: True

NODES
  /
    collect_data (atom_calibration/collect_data)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rosbag_play (rosbag/play)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [971007]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ca43b19a-0da9-11ed-90f0-9f9d37bb287b
process[rosout-1]: started with pid [971034]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [971037]
[... 76 lines cut ...]
  File "/home/martelv/catkin_ws/src/atom/atom_calibration/src/atom_calibration/collect/data_collector_and_labeler.py", line 382, in getAllAbstractTransforms
    all_frames = self.listener.getFrameStrings()
  File "/opt/ros/noetic/lib/python3/dist-packages/tf/listener.py", line 97, in getFrameStrings
    data = yaml.load(self._buffer.all_frames_as_yaml()) or {}
TypeError: load() missing 1 required positional argument: 'Loader'
================================================================================REQUIRED process [collect_data-5] has died!
process has died [pid 971046, exit code 1, cmd /home/martelv/catkin_ws/devel/lib/atom_calibration/collect_data -s 0.5 -o ~/catkin_ws/data/agri_gaia_calibration_out -c /home/martelv/catkin_ws/src/agri-gaia-test/agri_gaia_calibration/calibration/config.yml -ssl lambda sensor_name: False __name:=collect_data __log:=/home/martelv/.ros/log/ca43b19a-0da9-11ed-90f0-9f9d37bb287b/collect_data-5.log].
log file: /home/martelv/.ros/log/ca43b19a-0da9-11ed-90f0-9f9d37bb287b/collect_data-5*.log
Initiating shutdown!
================================================================================
[collect_data-5] killing on exit.467626   Duration: 2.689768 / 37.371216               
[rviz-4] killing on exit
[rosbag_play-3] killing on exit9.468447   Duration: 2.690589 / 37.371216               
[robot_state_publisher-2] killing on exit Duration: 2.691500 / 37.371216               
 [RUNNING]  Bag Time: 1658840029.716241   Duration: 2.938383 / 37.371216               
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

I am still trying to figure out what went wrong. Could you try to calibrate again with the new bag file in the mean time?
https://kronegroup.sharepoint.com:443/:f:/s/KRONECloud/Eu4ooUr6AtFDlK6X-V65sQEBChlCXgY9Qk9HZTHIgYP-6A?e=5%3aWyo1lC&at=9

Thank you very much.

miguelriemoliveira, 2022-07-27:
Hi @sensorCalib ,

> Could you explain how to use this tool? This script accepts the following arguments:
> [-t {charuco,chessboard}] [-d DICT] -x NUM_X -y NUM_Y -L LENGTH [-l INNER_LENGTH] topic
> In my case, I would call [...]

About the question you did before and I forgot to answer, how many corners should we say, 11 or 12, I am never completely sure about this one. It says in the config.yaml  file:

```yaml
# The number of corners the pattern has in the X and Y dimensions.
  # Note: The charuco detector uses the number of squares per dimension in its detector.
  # Internally we add a +1 to Y and X dimensions to account for that.
  # Therefore, the number of corners should be used even for the charuco pattern.
  dimension: { "x": 9, "y": 6 }
```

So I would try first the number of corners (not the number of squares) in each direction.

> -L 100 (since each edge has a length of 100mm ?)

Yes, the size of each square. 

> -topic (don't know which topic to use)

It is the topic in which your camera is being published. You can run a 

rostopic list 

to check all topics in the system. Also,  you have defined it in the config.yaml:

https://github.com/miguelriemoliveira/agri-gaia-test/blob/8793e2d5a48b0c15e53b6a62cd3af2d408487d3d/agri_gaia_calibration/calibration/config.yml#L84

miguelriemoliveira, 2022-07-27:
> I then had a look again at the getting started and tried to setup everything again in a clean workspace. So after cloning the atom repo, I also tried to install the requirements.txt (something I didn't do before). This failed since ros_numpy apparently doesn't exist:
> ERROR: Could not find a version that satisfies the requirement ros_numpy (from -r requirements.txt (line 11)) (from versions: none) ERROR: No matching distribution found for ros_numpy (from -r requirements.txt (line 11))
> So, I did not install ros_numpy and went on, but somehow, this was enough to mess up my setup and now the data collect step crashes again:
> 
> [calib_crash.txt](https://github.com/lardemua/atom/files/9199899/calib_crash.txt)

Ok, please open a specific issue for this one and I will try to take a look.

miguelriemoliveira, 2022-07-27:
> A new bagfile in which the pattern is away from the ground, closer to the sensor and moving has been made. But as it seems, the data collect step still won't collect data. Neither for the zed nor for the ouster.

Data collection is a manually triggered step. You have to press a green sphere which is somewhere in your 3d space in rviz and then select the "save collection" to save a collection.

> As mentioned in the description
> https://lardemua.github.io/atom_documentation/procedures/#rgb-camera-labeling
> "You can check if the detection is working by observing the overlays of top of the [...]

For this lets first try to run the view_frames script to make sure the configuration of the pattern is ok. I will try it and get back to you.

miguelriemoliveira, 2022-07-27:
> I am still trying to figure out what went wrong. Could you try to calibrate again with the new bag file in the mean time? [...]

Sure, I will try it tomorrow and get back to you.

And thank you for being so patient with ATOM. Its really nice to have an outsider finding all sorts of bugs :-)

Perhaps, I get this to calibrate in your system and , we can arranje a zoom meeting so I can show you how it works.

miguelriemoliveira, 2022-07-27:
oops, can't download the new bagfile. Not sure why, just get this message:

[image]

I went to the folder from a link I received from a Roman Weisgerber and I find there bag 

recording_2022-07-26-14-53-46_0.bag

is this the one?

sensorCalib, 2022-07-28:
> Hi @sensorCalib ,
> 
> > [...]
> 
> About the question you did before and I forgot to answer, how many corners should we say, 11 or 12, I am never completely sure about this one. It says in the config.yaml file:
> 
> ```yaml
> # The number of corners the pattern has in the X and Y dimensions.
>   # Note: The charuco detector uses the number of squares per dimension in its detector.
>   # Internally we add a +1 to Y and X dimensions to account for that.
>   # Therefore, the number of corners should be used even for the charuco pattern.
>   dimension: { "x": 9, "y": 6 }
> ```
> 
> So I would try first the number of corners (not the number of squares) in each direction.
> 
> > -L 100 (since each edge has a length of 100mm ?)
> 
> Yes, the size of each square.
> 
> > -topic (don't know which topic to use)
> 
> It is the topic in which your camera is being published. You can run a
> 
> rostopic list
> 
> to check all topics in the system. Also, you have defined it in the config.yaml:
> 
> https://github.com/miguelriemoliveira/agri-gaia-test/blob/8793e2d5a48b0c15e53b6a62cd3af2d408487d3d/agri_gaia_calibration/calibration/config.yml#L84

Thanks. What should happen, if the configuration of the pattern is correct? And what if it isn't? I called the script, but didn't see any success message or something like that. As it seems, the configuration is incorrect(?)
[image]

The screenshot shows the output for "-x 12 -y 9", however it is the same for "-x 11 -y 8". (And it is also the same for "-x 9 -y 12" and "-x 8 -y 11", in case the axes were swapped somehow.)

miguelriemoliveira, 2022-07-28:
It should draw something in the image I think ... but I will try it later today or tomorrow and get back to you ...

sensorCalib, 2022-07-28:
> oops, can't download the new bagfile. Not sure why, just get this message:
> 
> [image]
> 
> I went to the folder from a link I received from a Roman Weisgerber and I find there bag
> 
> recording_2022-07-26-14-53-46_0.bag
> 
> is this the one?

Yes, that is the latest bag file.

> Sure, I will try it tomorrow and get back to you.

Thank you for having a look at it.
 
> And thank you for being so patient with ATOM. Its really nice to have an outsider finding all sorts of bugs :-)
> 
> Perhaps, I get this to calibrate in your system and , we can arranje a zoom meeting so I can show you how it works.

No problem. Although, I can not deny that I would have preferred it to work immediately without having to find these bugs. :-)

miguelriemoliveira, 2022-07-28:
> Although, I can not deny that I would have preferred it to work immediately without having to find these bugs. :-)

Me too :- )

miguelriemoliveira, 2022-07-28:
Hi @sensorCalib ,

I just tried the view pattern script and the result was this:

[image]

The figure on the right is created by the script and shows the circles on top of the corners, signaling that the chessboard is being detected.

I ran the script like this:

    rosrun atom_calibration view_pattern.py -t chessboard -x 10 -y 7 -L 0.1 /zed2i_0/rgb_raw/image_raw_color

So it turns out you only need to count the number of corners you see in the horizontal (10) and the vertical (7).

I will see if I can collect data now ...

miguelriemoliveira, 2022-07-28:
I just successfully collected a dataset

https://youtu.be/7EEnSdLBdyk

The dataset is here:

[test1.zip](https://github.com/lardemua/atom/files/9212726/test1.zip)

Tomorrow I can try the calibration to see how it goes.

Also, I will try to help you with your bugs.

miguelriemoliveira, 2022-07-28:
Just pushed to agri_gaia_test.

After git pull you should check the calibration.yaml, uncomment your paths and comment mine and reconfigure your calibration package.

miguelriemoliveira, 2022-07-28:
I just tested the dataset playback and corrected the lidar automatic labels for two collections.

https://youtu.be/Xk56xqCWAj0

Here's the dataset also with the **dataset_corrected.json**

[Uploading test1.zip…]()

sensorCalib, 2022-08-03:
> I ran the script like this:
> 
> ```
> rosrun atom_calibration view_pattern.py -t chessboard -x 10 -y 7 -L 0.1 /zed2i_0/rgb_raw/image_raw_color
> ```
> 
> So it turns out you only need to count the number of corners you see in the horizontal (10) and the vertical (7).
> 
> I will see if I can collect data now ...

This worked for me too, thank you. Ok, understood which parameters to choose for x and y. (Also, the parameter for -L has to be passed in meters, not in milimeters as I previously did.)

> I just successfully collected a dataset
> 
> https://youtu.be/7EEnSdLBdyk
> 
> The dataset is here:
> 
> [test1.zip](https://github.com/lardemua/atom/files/9212726/test1.zip)
> 
> Tomorrow I can try the calibration to see how it goes.
> 
> Also, I will try to help you with your bugs.

Great to see, that you were able to collect data. (It's still crashing for me as mentioned in
https://github.com/miguelriemoliveira/agri-gaia-test/issues/1 )

miguelriemoliveira, 2022-08-06:
> https://github.com/miguelriemoliveira/agri-gaia-test/issues/1

Should be solved now. I will try to run a calibration ...

miguelriemoliveira, 2022-08-06:
So to calibrate I need all collections correctly annotated, so I went back to the dataset playback and manually corrected the annotations, so we now have:

**Collection 000**
[image]

**Collection 001**
[image]

**Collection 002**
[image]

**Collection 003**
[image]

**Collection 004**
[image]

Here's the dataset. with the last **dataset_corrected.json** file

[test1.zip](https://github.com/lardemua/atom/files/9274868/test1.zip)
.

miguelriemoliveira, 2022-08-06:
Hi @sensorCalib ,

so here's the first calibration of agri-gaia.

[image]

Results were the following:

```bash
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+---------+
| Collection | ouster_os1 | zed2i_0 |
+------------+------------+---------+
|    000     |   0.0215   |  0.4476 |
|    001     |   0.0247   | 19.2592 |
|    002     |   0.0418   | 17.3578 |
|    003     |   0.0562   | 28.4024 |
|    004     |   0.0396   | 85.9234 |
|  Averages  |   0.0368   | 30.2781 |
+------------+------------+---------+
```

miguelriemoliveira, 2022-08-06:
The visualization is not working well. I will check the pattern. I think its because of the position of the origin ...
Created #526

The pixel errors of the zed are too high. I will investigate.

sensorCalib, 2022-08-09:
Hi @miguelriemoliveira 
with the crash in
https://github.com/miguelriemoliveira/agri-gaia-test/issues/1
being fixed, I was able to collect data as well. Thank you very much once again. 

I then tried to calibrate as you did above, but ran into another error. This might have happened because I am not directly working on the machine and instead using a remote desktop software (x2go), but I havn't figured it out yet Or do you think this is something related to to ATOM?

> rosrun atom_calibration calibrate -json /home/martelv/catkin_ws/data/ag_out/dataset.json -v -rv -si
Traceback (most recent call last):
  File "/home/martelv/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/martelv/catkin_ws/src/atom/atom_calibration/scripts/calibrate", line 23, in <module>
    from atom_calibration.calibration.objective_function import errorReport, objectiveFunction
  File "/home/martelv/catkin_ws/src/atom/atom_calibration/src/atom_calibration/calibration/objective_function.py", line 27, in <module>
    from atom_core.vision import projectToCamera
  File "/home/martelv/catkin_ws/src/atom/atom_core/src/atom_core/vision.py", line 14, in <module>
    import open3d as o3d
  File "/home/martelv/.local/lib/python3.8/site-packages/open3d/__init__.py", line 56, in <module>
    _CDLL(str(next((_Path(__file__).parent / 'cpu').glob('pybind*'))))
  File "/usr/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/martelv/.local/lib/python3.8/site-packages/open3d/cpu/pybind.cpython-38-x86_64-linux-gnu.so: undefined symbol: glGetQueryObjectui64v

miguelriemoliveira, 2022-08-09:
I never had this before. What is your open3d version? Mine is 0.15.2 ...

[image]

miguelriemoliveira, 2022-08-09:
Hi again @sensorCalib ,

here they say they had to compile from source ... What machine do you have? A standard ubuntu?

The remote desktop may have problems, I remember using anyviewer crashed sometimes with rviz. 
You could just use ssh -X to the machine, but visualization is slower ...

sensorCalib, 2022-08-11:
Hi @miguelriemoliveira ,

"pip show open3d" told me, that open3d was installed. However, calling python3 and trying to "import open3d" as you did, resulted in an error.

So, I tried to install open3d from source as you suggested following this guide
http://www.open3d.org/docs/release/compilation.html
and created the python package with
make python-package
But I still got the same error when trying to "import open3d".

I then found out, that there is a difference between "pip" and "python3 -m pip"
https://stackoverflow.com/questions/25749621/whats-the-difference-between-pip-install-and-python-m-pip-install
so I retried installing the python-package using "python3 -m pip". This was successful and "import open3d" was not crashing anymore:

> python3
Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import open3d
>>> open3d.__version__
'0.15.2+b40282c9'
`

After that, I got another crash: 

> [ERROR] [1660134840.981366787]: PluginlibFactory: The plugin for class 'rviz_visual_tools/KeyTool' failed to load.  Error: 
According to the loaded plugin descriptions the class rviz_visual_tools/KeyTool with base class type rviz::Tool does not exist. Declared types are  rviz/FocusCamera rviz/Interact rviz/Measure rviz/MoveCamera rviz/PublishPoint rviz/Select rviz/SetGoal rviz/SetInitialPose rviz_plugin_selected_points_publisher/SelectedPointsPublisher rviz_plugin_tutorials/PlantFlag

Since I wanted to install open3d from source, I had to update cmake which in turn probably messed up my ROS-setup. Luckily, I was able to resolve that error through installing rviz visual tools again:
`sudo apt-get install ros-noetic-rviz-visual-tools`

No further crashes since then :)

With that being out of the way, I retried the calibration process. So, I collected data again and went to the "Dataset playback" step, just as you did in the videos.
`roslaunch agri_gaia_calibration dataset_playback.launch`
`clear && rosrun atom_calibration dataset_playback -json data/ag_out3/dataset.json -ow`

However, I don't receive any images from rviz:
[image]

Do you have an idea, why no frames were received?

dataset_playback is reacting to button presses and created a dataset_corrected.json, so it seems only the visualisation is buggy for me:
[image]

miguelriemoliveira, 2022-08-11:
Hi,

Glad to hear you advanced.

To visualize in rviz you should activate the visualization flags . Try with

-v -rv -si

https://lardemua.github.io/atom_documentation/procedures/#calibrate

sensorCalib, 2022-08-11:
Hi,

I currently am at the "Dataset playback" step to correct the labels. The flags you mentioned arn't available here (?).

But since that step is optional, I can skip it and directly go to the "Calibrate sensors" step. So, now I am calling

`roslaunch agri_gaia_calibration calibrate.launch`
`rosrun atom_calibration calibrate -json /home/martelv/catkin_ws/data/ag_out3/dataset.json -v -rv -si`

But there still is no visualisation:
[image]
[image]

miguelriemoliveira, 2022-08-11:
Hi @sensorCalib ,

sorry, that's what happens when you try to answer during lunch ...

Let's try again:

> Do you have an idea, why no frames were received?
> dataset_playback is reacting to button presses and created a dataset_corrected.json, so it seems only the visualisation is buggy for me:

Not sure why this happens. Can you confirm the messages are being sent?
Just do a 

    rostopic echo /zed2i_0/rgb_raw/image_raw_color/labeled2d

Do you see any data being sent?

To me it works fine with my dataset. Can you try with this dataset to make sure its not the dataset?

[test1.zip](https://github.com/lardemua/atom/files/9274868/test1.zip)

Also, can you share the details of your dataset, i.e. 

    rosrun atom_calibration inspect_atom_dataset -json <your dataset.json>

About the problem with calibrate, it sounds like the same so lets first figure this one out.

sensorCalib, 2022-08-11:
Hi,

> sorry, that's what happens when you try to answer during lunch ...

no problem, thanks for helping even during lunch break.

> Let's try again:
>
>
> Not sure why this happens. Can you confirm the messages are being sent? Just do a
>
> ```
> rostopic echo /zed2i_0/rgb_raw/image_raw_color/labeled2d
> ```
>
> Do you see any data being [...]

There were no messages being sent:

> rostopic echo /zed2i_0/rgb_raw/image_raw_color/labeled2d
WARNING: no messages received and simulated time is active.
Is /clock being published?

However, that warning lead me to
https://answers.ros.org/question/12083/messages-being-blocked-from-publishing/
which told me how to fix the problem.

I don't know how and when I set "use_sim_time" to true and why this was blocking my messages from being published, but it's working now. 

Going to correct the labels now.

miguelriemoliveira, 2022-08-11:
> There were no messages being sent:
>
>
> However, that warning lead me to https://answers.ros.org/question/12083/messages-being-blocked-from-publishing/ which told me how to fix the problem.
>
> I don't [...]

That's strange. 

When you run a bag file, its normal to set the use_sim_time parameter to true.  Its in the automatically generated launch files:  

https://github.com/miguelriemoliveira/agri-gaia-test/blob/4de14bfd4054c1d6c302dc97fc00e67f72500f23/agri_gaia_calibration/launch/playbag.launch#L67

In order for this to work some node should publish the clock (not the real clock, but the bag file clock). This is done with the --clock flag as in:

https://github.com/miguelriemoliveira/agri-gaia-test/blob/4de14bfd4054c1d6c302dc97fc00e67f72500f23/agri_gaia_calibration/launch/playbag.launch#L69-L76

So I am really lost at why you had that problem. The  agri-gaia/launch/playbag.launch

should set the parameter to true ...

miguelriemoliveira, 2022-08-11:
> Going to correct the labels now.

Don't forget that for that you need to:

1. Compile the ATOM repo. There's a small c++ code that runs as an rviz plugin. Actually, this is the only ATOM c++ code.

   cd catkin_ws
   catkin_make

2. Install a special fork of rviz as explained here

https://lardemua.github.io/atom_documentation/getting_started/#clone-rviz-fork

miguelriemoliveira, 2022-08-11:
> 2\. Install a special fork of rviz as explained here

Just thought about it ... I don't think you'll need this since you don't have a depth camera.

sensorCalib, 2022-08-12:
> So I am really lost at why you had that problem. The **agri-gaia/launch/playbag.launch** should set the parameter to true ... Are you using this launch file (using the dataset_playback.launch calls this one)?

I called
> roslaunch agri_gaia_calibration dataset_playback.launch

so I guess this I am using "agri-gaia/launch/playbag.launch".

As for the calibration: I corrected the labels and started a calibration, these were the results:

> Initializing optimization ...
One optimizer iteration has 13 function calls.
Starting optimization ...
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+----------+
| Collection | ouster_os1 | zed2i_0  |
+------------+------------+----------+
|    000     |   0.0557   |  0.3102  |
|    001     |   0.0409   | 157.5191 |
|    002     |   0.0408   |  8.8233  |
|    003     |   0.0394   | 99.9691  |
|    004     |   0.0346   | 133.7405 |
|    005     |   0.0616   | 69.8063  |
|  Averages  |   0.0455   | 78.3614  |
+------------+------------+----------+
   Iteration     Total nfev        Cost      Cost reduction    Step norm     Optimality   
       0              1         1.0620e+05                                    1.29e+05    
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+----------+
| Collection | ouster_os1 | zed2i_0  |
+------------+------------+----------+
|    000     |   0.0303   | 26.5169  |
|    001     |   0.0303   | 133.0785 |
|    002     |   0.0203   | 27.6154  |
|    003     |   0.0256   | 124.7527 |
|    004     |   0.0246   | 108.7841 |
|    005     |   0.0323   | 95.5582  |
|  Averages  |   0.0272   | 86.0510  |
+------------+------------+----------+
       1              2         1.0273e+05      3.47e+03       3.45e-01       3.74e+04    
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+----------+
| Collection | ouster_os1 | zed2i_0  |
+------------+------------+----------+
|    000     |   0.0272   | 23.6644  |
|    001     |   0.0341   | 135.4535 |
|    002     |   0.0191   | 29.1886  |
|    003     |   0.0230   | 122.7202 |
|    004     |   0.0284   | 112.1459 |
|    005     |   0.0313   | 91.2909  |
|  Averages  |   0.0272   | 85.7439  |
+------------+------------+----------+
       2              5         1.0270e+05      3.28e+01       5.55e-02       5.32e+04    
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+----------+
| Collection | ouster_os1 | zed2i_0  |
+------------+------------+----------+
|    000     |   0.0297   | 22.6901  |
|    001     |   0.0312   | 135.3852 |
|    002     |   0.0190   | 27.2675  |
|    003     |   0.0244   | 122.3284 |
|    004     |   0.0254   | 111.7949 |
|    005     |   0.0328   | 91.4350  |
|  Averages  |   0.0271   | 85.1502  |
+------------+------------+----------+
       3              6         1.0211e+05      5.91e+02       1.26e-02       2.94e+04    
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+----------+
| Collection | ouster_os1 | zed2i_0  |
+------------+------------+----------+
|    000     |   0.0347   | 22.4792  |
[... 71 lines cut ...]
+------------+------------+----------+
       9             12         1.0176e+05      9.96e+00       2.14e-03       3.92e+03    
`ftol` termination condition is satisfied.
Function evaluations 12, initial cost 1.0620e+05, final cost 1.0176e+05, first-order optimality 3.92e+03.

-----------------------------
Optimization finished in 31.70428 secs: `ftol` termination condition is satisfied.
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+----------+
| Collection | ouster_os1 | zed2i_0  |
+------------+------------+----------+
|    000     |   0.0348   | 20.4498  |
|    001     |   0.0267   | 137.2471 |
|    002     |   0.0203   | 23.9775  |
|    003     |   0.0272   | 120.2209 |
|    004     |   0.0216   | 113.4144 |
|    005     |   0.0356   | 89.9654  |
|  Averages  |   0.0277   | 84.2125  |
+------------+------------+----------+
output_folder is: /home/martelv/catkin_ws/data/ag_out2
Saved json output file to /home/martelv/catkin_ws/data/ag_out2/atom_calibration.json.

[ag_out2.zip](https://github.com/lardemua/atom/files/9316879/ag_out2.zip)

The average error is higher than in your case, but I don't know how to interpret these numbers. Could you explain what they mean and when they are small enough?

miguelriemoliveira, 2022-08-12:
Hi @sensorCalib ,

About the use_sim_time, I figured it out. See #527. 

About the calibration, camera error is in pixels, lidar in meters (opened #528 to improve that).

I will try to pick it up tonight or tomorrow and get back to you.

Have a nice weekend.

miguelriemoliveira, 2022-08-14:
Hi again @sensorCalib,

so I looked around for a reason why the errors were so high and there is a problem in the calibration.yaml.

We are saying that the calibration pattern is fixed (line 114). In your case the pattern is not fixed, since it moves from collection to collection.

I will fix this and results should be better.

miguelriemoliveira, 2022-08-14:
Hi again @sensorCalib ,

```bash
Optimization finished in 4.20546 secs: `xtol` termination condition is satisfied.
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+------------+---------+
| Collection | ouster_os1 | zed2i_0 |
+------------+------------+---------+
|    000     |   0.0073   |  0.4478 |
|    001     |   0.0071   |  0.3347 |
|    002     |   0.0078   |  0.3116 |
|    003     |   0.0064   |  0.4439 |
|    004     |   0.0075   |  0.4209 |
|  Averages  |   0.0072   |  0.3918 |
+------------+------------+---------+
```

So now its much better.  Camera errors under 1 pixel.

Can you try to achieve these values as well?

Note: I had to manually change the parameter fixed from true to false. For consistency I did it in both the **dataset.json** and the **dataset_corrected.json** files. But this was a hack to avoid having to create a new dataset.

If you change the calibration.yaml and collect a new dataset it should be fine.

miguelriemoliveira, 2022-08-14:
I ran the point_cloud_to_image evaluation to see if the calibration makes sense:

    clear && rosrun atom_evaluation point_cloud_to_image -json atom_calibration.json -ls ouster_os1 -cs zed2i_0

[image]

Looks good. Color transitions are changes in lidar range measurements, and are aligned with the depth changes in the image.
So the calibration looks good, even though it should contain around 20 or 30 collections instead of just 5.

sensorCalib, 2022-08-16:
> Hi again @sensorCalib ,
> 
> ```shell
> Optimization finished in 4.20546 secs: `xtol` termination condition is satisfied.
> Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
> +------------+------------+---------+
> | Collection | ouster_os1 | zed2i_0 |
> +------------+------------+---------+
> |    000     |   0.0073   |  0.4478 |
> |    001     |   0.0071   |  0.3347 |
> |    002     |   0.0078   |  0.3116 |
> |    003     |   0.0064   |  0.4439 |
> |    004     |   0.0075   |  0.4209 |
> |  Averages  |   0.0072   |  0.3918 |
> +------------+------------+---------+
> ```
> 
> So now its much better. Camera errors under 1 pixel.
> 
> Can you try to achieve these values as well?
> 
> Note: I had to manually change the parameter fixed from true to false. For consistency I did it in both the **dataset.json** and the **dataset_corrected.json** files. But this was a hack to avoid having to create a new dataset.
> 
> If you change the calibration.yaml and collect a new dataset it should be fine.

I used your quick hack too and modified the parameter "fixed" from true to false, this was the result:

```shell
Optimization finished in 29.78579 secs: `xtol` termination condition is satisfied.
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+----------------+--------------+
| Collection | ouster_os1 (m) | zed2i_0 (px) |
+------------+----------------+--------------+
|    000     |     0.0241     |    0.3103    |
|    001     |     0.0120     |    0.3321    |
|    002     |     0.0261     |    0.3117    |
|    003     |     0.0235     |    0.4718    |
|    004     |     0.0415     |    0.3959    |
|    005     |     0.0324     |    0.3241    |
|  Averages  |     0.0266     |    0.3576    |
+------------+----------------+--------------+
```

After evaluating the calibration with
```shell
clear && rosrun atom_evaluation point_cloud_to_image -json atom_calibration.json -ls ouster_os1 -cs zed2i_0
```

I got this result:
[image]

This looks off.

miguelriemoliveira, 2022-08-16:
Hi ,

Can you share your dataset? I think your labelling may have something wrong
...

sensorCalib, 2022-08-16:
Here is the dataset I used:
[ag_out2.zip](https://github.com/lardemua/atom/files/9349826/ag_out2.zip)

miguelriemoliveira, 2022-08-16:
Thanks @sensorCalib ,

I have done several experiments and I get the same results. The alignment is not the best,

I did not understand whats the problem yet. Strange because in my dataset it calibrates fine, Can you try with my dataset?

sensorCalib, 2022-08-17:
I think I found the problem: The path to the mesh file is wrong in both my dataset.json and my dataset_corrected.json. Could that be the reason? If yes, I will adjust the path, collect a new data set and give the calibration another try.

By the way, where did you put the chess_pattern.dae? It was removed in
https://github.com/miguelriemoliveira/agri-gaia-test/commit/4de14bfd4054c1d6c302dc97fc00e67f72500f23
but the path in test1/dataset.json still refers to
"package://agri_gaia_description/meshes/pattern_chess.dae"
Where should I put the pattern instead?

sensorCalib, 2022-08-17:
The calibration looks fine with your dataset:
[image]

(In a  previous and now deleted message, I assumed that I used the wrong mesh-file for the chessboard pattern, but I am not sure. Is 
`"package://atom_calibration/meshes/chessboard_7x11_corners/pattern.dae"`
equivalent to
`"package://agri_gaia_description/meshes/pattern_chess.dae"`
? The latter has been removed by you some commits ago, but can be found here
https://github.com/miguelriemoliveira/agri-gaia-test/blob/f5bb191a8abd84c2f1ff7cf71fc6dea37e2214a9/agri_gaia_description/meshes/pattern_chess.dae )

[12 more replies]

closed 2022-10-22 by miguelriemoliveira