lardemua/atom #500: Create a system evaluation script

manuelgitgomes, 2022-06-27:
Create a script that evaluates every sensor pair possible inside a robotic system. Exports the results in .csv of .pdf.

miguelriemoliveira, 2023-03-31:
Not a priority for now.

manuelgitgomes, 2023-09-25:
Hello @miguelriemoliveira!

As I had to take results, I decided to move forward in this issue.
The updates are on the `dev/full-evaluation` branch.

I added the feature that during configuration, a new launch file is created that launches all required evaluation nodes.
This nodes have `save_file_results` enabled by default, so it automatically records the results.
To achieve this, I had to change how `save_file_results` and `save_file_results_name` are used.
The first is now a `store_true` argument, that, when enabled, saves the evaluation results.
The last is now used to define the path in which the results will be saved.
The default path is inside the `results` subdirectory inside the test dataset directory.

It also has the possibility to use a `csf`, so the same collection can be eliminated from all evaluation scripts.
An example:

https://github.com/miguelriemoliveira/softbot2/blob/main/softbot2_calibration/launch/full_evaluation.launch

To run this file, I simply used:

```bash
roslaunch softbot2_calibration full_evaluation.launch train_json:=$ATOM_DATASETS/softbot2/train_20230913/atom_calibration.json test_json:=$ATOM_DATASETS/softbot2/test_20230920/dataset_corrected.json csf:="lambda x: int(x) in [1,2,3]"
```

I still need to adjust the depth evaluation scripts so they can better fit here.
I plan to do it tommorow.

miguelriemoliveira, 2023-09-25:
Hi @manuelgitgomes ,

Sounds great.

Are you creating this launch file during configuration?

manuelgitgomes, 2023-09-26:
Hello @manuelgitgomes!

Yes, this is done during configuration!

miguelriemoliveira, 2023-09-26:
Nice!

manuelgitgomes, 2023-09-27:
Hello @miguelriemoliveira.

This is now fully functional I believe.
To achieve this:

- Cross collection was changed to inter collection
- Depth evaluation was fixed and placed up to date with other scripts
- Inter collection depth evaluation scripts were created.

Now with SoftBot2, a complex system, this full evaluation is fully functional.

miguelriemoliveira, 2023-09-27:
Thanks  @manuelgitgomes ,

Can you put here the output of a full evaluation, so we get an idea of what we get?

Also, you should update the documentation accordingly. Do you know how to?

manuelgitgomes, 2023-09-27:
Hello @miguelriemoliveira.

Honestly, the output is a mix of all the other outputs, so lots of tables that do not appear all on the terminal.
This is not very useful and only can be useful if extra scripts are written (for showing only the averages, p.e.)

What is very useful, however, is that it generates all csv files automatically.
From these files we can gather the results:
[image]

miguelriemoliveira, 2023-09-27:
Ah, ok. I think that's ok as well. But it should be written in the documentation.

manuelgitgomes, 2023-10-02:
This is now merged!

closed 2023-10-02 by manuelgitgomes