lardemua/atom #501: Possible depth labelling problem with robot

danifpdra, 2022-07-04:
What do you think @miguelriemoliveira ?

We should really implement the delete points feature in the dataset reviewer

[image]

miguelriemoliveira, 2022-07-04:
Yes, perhaps we should. Can you post a small bagfile please?

miguelriemoliveira, 2022-07-04:
> I don't think we have the sync bagfiles folder in the larcc server so I used wetransfer

What do you mean?

If its a new folder it copies ...

I will download and add.

danifpdra, 2022-07-05:
The sync is only for the datasets folder, no?

miguelriemoliveira, 2022-07-05:
Right, forgot about that. Sorry. I am doing a lot at the same time ... I will try to look at this later in the week.

miguelriemoliveira, 2022-07-12:
@danifpdra ,

as I told you I cannot work on this until probably next september. Do you know how you could do it? Or we can wait til then ...

miguelriemoliveira, 2023-11-15:
Fixed. Labeling for depth works fine now.

closed 2023-11-15 by miguelriemoliveira