lardemua/atom #503: Images don't appear while calibration due to wrong configuration of rviz template

danifpdra, 2022-07-05:
Hi @miguelriemoliveira ,

You have changed the format of the number of the collections but you have not changed the configurations generated by the configure_calibration_package for example while calibrating...

[image]

Now the images don't appear and we have to change to the correct topic if we want to visualize

miguelriemoliveira, 2022-07-05:
Hi @danifpdra ,

right, If you know how to please do it ..., otherwise I will try to find  some time later this week.

danifpdra, 2022-07-06:
Yes, I did it.. The question is... That value is static, always collection 000... For example, in my case I'm not using collection 000, I'm using only from 001 because the 000 was not good. So I can't see the images anyway... Is there any way that this value could be read in real time?

miguelriemoliveira, 2022-07-06:
Hum, I don't think so. 

This Image displays only show one image of a sensor, so a collection must be selected.

I don't think there is a way, at configuration time, to know which is the best collection, so 000 is selected.

You may change it in RVIZ later on.

danifpdra, 2022-07-06:
ok, it is solved then. closing