lardemua/atom #505: Evaluate if Realsense images are not fit for depth labelling

danifpdra, 2022-07-12:
Several examples... Some are ok and some are not 

[image]
[image]
[image]
[image]
[image]

miguelriemoliveira, 2022-07-12:
Asus vs real sense

[image]

danifpdra, 2022-07-21:
[image]

miguelriemoliveira, 2022-07-21:
I don't like the depth very much, but we can try to calibrate and see how it goes...

danifpdra, 2022-07-22:
[image]

danifpdra, 2022-07-22:
[image]

danifpdra, 2022-07-22:
I feel like a doctor analysing an x ray

[image]

miguelriemoliveira, 2022-07-22:
The data looks awful ...I would not use this camera.

danifpdra, 2022-07-22:
so should I remove this from the paper?

closed 2022-08-15 by miguelriemoliveira

miguelriemoliveira, 2022-10-11:
No. Lets see the results ... whats the accuracy?