lardemua/atom #506: Access to multiple ORBEC's rgb image with a simplified ROS node

eupedrosa, 2022-07-13:
Hi @miguelriemoliveira and @danifpdra 

How do you access to multiple ORBEC's rgb image? Do you have a simplified ROS node for this?

miguelriemoliveira, 2022-07-13:
Hi @eupedrosa ,

we set up this launch file to call the usb_cam_node script, and it works fine.

```xml
<launch>
    <arg name="video_device" default="/dev/video2"/>
    <arg name="camera" default="camera_0"/>
    <arg name="viz" default="false"/>

    <!--    <group ns="$(arg camera)">-->
    <node name="$(arg camera)" pkg="usb_cam" type="usb_cam_node" output="screen">
        <param name="video_device" value="$(arg video_device)"/>
        <param name="camera_name" value="$(arg camera)"/>
        <param name="image_width" value="1280"/>
        <param name="image_height" value="800"/>
        <param name="pixel_format" value="yuyv"/>
        <param name="camera_frame_id" value="$(arg camera)_rgb_optical_frame"/>
        <param name="io_method" value="mmap"/>
        <remap from="/$(arg camera)/image_raw" to="/$(arg camera)/rgb/image_raw"/>
    </node>
    <!--    </group>-->
    <group if="$(arg viz)">
        <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
            <remap from="image" to="/$(arg camera)/rgb/image_raw"/>
            <param name="autosize" value="true"/>
        </node>
    </group>
</launch>
```

eupedrosa, 2022-07-14:
It looks like you solution only works with the Pro version. The older Orbbecs do not create a /dev/video* device.
Anyway, I was able to have two (2) old Orbbecs running in the same computer with the `astra_camera` package. 

@miguelriemoliveira, do you think that the 640x480 resolution of the old orbbecs is too low? Maybe we can try this in the simulation?

danifpdra, 2022-07-14:
> > It looks like you solution only works with the Pro version. The older Orbbecs do not create a /dev/video* device.
> > Anyway, I was able to have two (2) old Orbbecs running in the same computer with the astra_camera package.
> 
> ok, great.
> 
> > @miguelriemoliveira, do you think that the 640x480 resolution of the old orbbecs is too low? Maybe we can try this in the simulation?
> 
> I think it will be. @danifpdra do you have any input on this?
> 
> @eupedrosa you can also just run the detector with that script you created to see whether or not it can detect. @danifpdra now has the calibration pattern, but she told me that when you need it she can bring it to the UA.

640x480 works depending on the distance from the pattern to the sensor... I could not make it work at the position of the sensors we have right now. A higher resolution would be better/safer. 

Yes, I have the pattern at home and I'm currently running some tests. When do you need it @eupedrosa ?

eupedrosa, 2022-07-14:
Would the smallest pattern that I have (30x14 cm) be detected by the velodyne? It would allow us to have the pattern closer to the cameras.

> Yes, I have the pattern at home and I'm currently running some tests. When do you need it @eupedrosa ?

Not 100% sure. Because of @miguelriemoliveira, the cameras are not yest installed on the robot, he wants them to overlap. But I already found a way to installed them the way he wants. I will dot it next monday. After that is just a matter of choosing a day for the data capture.

danifpdra, 2022-07-14:
> Would the smallest pattern that I have (30x14 cm) be detected by the velodyne? It would allow us to have the pattern closer to the cameras.

I have never tried to use a smaller pattern on LiDARs. You could maybe try on simulation. But LiDARs have a low resolution so I guess that using a smaller pattern would increase the error.

> Not 100% sure. Because of @miguelriemoliveira, the cameras are not yest installed on the robot, he wants them to overlap. But I already found a way to installed them the way he wants. I will dot it [...]

Ok, next week I will leave the pattern at the university

danifpdra, 2022-07-14:
FYI @eupedrosa, I ended up having to go to UA today so I brought the calibration pattern back to LAR

danifpdra, 2022-08-30:
Hi @eupedrosa,

Are you still needing the ChArUcO?

eupedrosa, 2022-08-30:
When do you need the ChArUcO?

danifpdra, 2022-08-30:
I need to record some new data because I will be going to Barcelona for 3 months and I can't do it there. Do you need it next week ?

eupedrosa, 2022-08-30:
No, I do not need it next week. When do you want to get it at the lab?

danifpdra, 2022-08-30:
I am waiting for the engineer to produce a support piece to install a different camera on the robotic arm, and he only comes back from the holidays on Monday, so can I let you know when that's ready because I only need the calibration pattern after that?

eupedrosa, 2022-08-30:
Ok. Just let me know when you need it.

closed 2023-03-31 by miguelriemoliveira