lardemua/atom #509: ERROR: Could not find a version that satisfies the requirement KeyPressManager (from versions: none)

robofoundry, 2022-07-16:
Tried to follow getting started instructions here - https://lardemua.github.io/atom_documentation/getting_started/

First error I received was relating to ros_numpy not found and had to install it from source, I logged and closed a separate issue #508 for that. After that I got the next error below.

trying to install requirement.txt got the error:

ERROR: Could not find a version that satisfies the requirement KeyPressManager (from versions: none)
ERROR: No matching distribution found for KeyPressManager

Tried to do some google searches to find a fix/workaround and found this repo [https://github.com/miguelriemoliveira/KeyPressManager] but it has some more dependencies like cv2 which could not be installed on Ubuntu 20.04 + ROS2 foxy.

Also, somewhere in issue #185 it said, some of these unnecessary dependencies were removed but the requirement.txt in root folder still has these dependencies which fails the very first step of installation. I would love to try and use this tool which looks very promising but struggling to get it installed.

Not sure if the Atom tool is supported on ROS2 at this point after having all these errors, I'm probably going to give up installing. Hope you can fix the requirements.txt and instructions to install on the wiki page so people can install it and use it.
[image]

miguelriemoliveira, 2022-07-16:
Hi @robofoundry ,

thanks for giving it a shot.

I removed the KeyPressManager dependency from the requirements.txt. This functionality was directly integrated into ATOM in an effort to reduce the number of dependencies.

ATOM is not supported for ROS2, just ROS 1.  I will try to make it clear in the instructions. 

I did not work with ROS2 yet, so I am not sure what will be the effort to change to ROS2. We are considering to work on this in the next few months.

robofoundry, 2022-07-16:
Thanks for being so fast in response and updating the requirements.txt. Appreciate the clarification ROS2. Would be glad to help if any help is needed to port it to ROS2. I learned ROS on ROS2 so always looking for all these great tools to become available on foxy/humble.

closed 2022-07-16 by robofoundry