lardemua/atom #511: How should I annotate a calibration pattern that is not entirely visable?

danifpdra, 2022-07-20:
Hi @manuelgitgomes and @miguelriemoliveira ,

Take this example...

[image]

This new annotation tool doesn't allow me to ignore one side of the chessboard... or maybe it does, and I don't know how. can you explain me?

manuelgitgomes, 2022-07-20:
Hello! I had this problem as well. What I did was randomly click the side not used, and then clean it. It's not the best form, but it works.

danifpdra, 2022-07-20:
How do I clean just one side @manuelgitgomes ?

manuelgitgomes, 2022-07-20:
The letter "x" cleans all points from the current side!

danifpdra, 2022-07-20:
oh ok, thanks

miguelriemoliveira, 2022-07-20:
Thanks for helping  Manuel

closed 2022-07-24 by miguelriemoliveira