lardemua/atom #523: Requirements can not be installed

sensorCalib, 2022-07-28:
The requirements from the tutorial
https://lardemua.github.io/atom_documentation/getting_started/
couldn't be installed. Apparently, ros_numpy doesn't exist anymore.

`sudo pip3 install -r src/atom/requirements.txt 
[sudo] password for martelv: 
Requirement already satisfied: rospy_message_converter in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 1)) (0.5.2)
Requirement already satisfied: opencv-contrib-python in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 2)) (4.6.0.66)
Requirement already satisfied: JSON-minify in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 3)) (0.3.0)
Requirement already satisfied: jinja2 in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 4)) (3.1.2)
Requirement already satisfied: alphashape in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 5)) (1.3.1)
Requirement already satisfied: descartes in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 6)) (1.1.0)
Requirement already satisfied: colorama in /usr/lib/python3/dist-packages (from -r src/atom/requirements.txt (line 7)) (0.4.3)
Requirement already satisfied: graphviz in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 8)) (0.20.1)
Requirement already satisfied: pandas in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 9)) (1.4.3)
Requirement already satisfied: pynput in /usr/local/lib/python3.8/dist-packages (from -r src/atom/requirements.txt (line 10)) (1.7.6)
ERROR: Could not find a version that satisfies the requirement ros_numpy (from -r src/atom/requirements.txt (line 11)) (from versions: none)
ERROR: No matching distribution found for ros_numpy (from -r src/atom/requirements.txt (line 11))`

This was first noticed in
https://github.com/lardemua/atom/issues/498#issuecomment-1196821098

miguelriemoliveira, 2022-07-28:
Hi @sensorCalib ,

    sudo apt-get install ros-$release-ros-numpy

from https://answers.ros.org/question/270439/ros_numpy-package/

Does it work?

sensorCalib, 2022-07-28:
Hi @miguelriemoliveira 

`sudo apt-get install ros-noetic-ros-numpy`

did work for me.

miguelriemoliveira, 2022-07-28:
OK, thanks. I will change the documentation to add this information.

closed 2022-07-28 by miguelriemoliveira