lardemua/atom #524: Hand-eye calibration not going as expected

danifpdra, 2022-07-28:
[image]

miguelriemoliveira, 2022-07-28:
Hi @danifpdra ,

I need more details as well as the conditions to replicate.

In any case I am not I will be able to work on this in july ...

danifpdra, 2022-07-29:
This was just for documentation... I will work on this

miguelriemoliveira, 2022-07-29:
Ah, ok. You assigned me so I panicked :- )

danifpdra, 2022-07-29:
> Ah, ok. You assigned me so I panicked :- )

Just FYI

miguelriemoliveira, 2022-09-13:
Hi @danifpdra ,

any news on this front? Should we close the issue?

danifpdra, 2022-09-13:
Hi @miguelriemoliveira,

No news, I am waiting on the support for the new camera to be installed.

Also, the RGB-D calibration is not working as expected so that should be solved first. I made lots of experiments but could not understand the difference between real data (not well calibrated) and simulated data (well calibrated)

miguelriemoliveira, 2022-09-13:
strange ... we can discuss it when you want, but perhaps its better we do it with the new data.

danifpdra, 2022-09-13:
yeah, sure. I can give you the asus rosbag (only RGB and depth) if you want to try it on your side on the camerabot

danifpdra, 2022-09-15:
Hi @miguelriemoliveira,

as we talked before, this serves to pass you information for you to replicate the problems on RGB-D calibration on your side.

**Bagfile**

https://uapt33090-my.sharepoint.com/:u:/g/personal/danielarato_ua_pt/EfOzjgb-Rg1LrsYEf73EBWkBOX67YofuvjGNGzfhRy_NcQ?e=OSw6UY

**System**

https://github.com/danifpdra/camerabot

**Configuration**

`rosrun camerabot_calibration configure -c config_asus.yml`

**Run all collections without inducing error**

`clear && rosrun atom_calibration calibrate -json $ATOM_DATASETS/camerabot/train_asus/dataset_corrected.json -ipg -si -rv -nig 0.0 0.00`

**Run a set of collections**

`clear && rosrun atom_calibration calibrate -json $ATOM_DATASETS/camerabot/train_asus/dataset_corrected.json -ipg -si -rv -nig 0.0 0.00 -csf "lambda x: int(x)<2"`

**Visualization**

roslaunch camerabot_calibration calibrate.launch dataset_file:=$ATOM_DATASETS/camerabot/train_asus/dataset_corrected.json run_calibration:=false

**Run bagfile with optimized parameters**

roslaunch camerabot_calibration playbag.launch optimized:=true

The dataset in question will be inside the camerabot folder if you sync your dataset folder with the script

miguelriemoliveira, 2022-09-15:
Thanks @danifpdra ,

I will try to look into this sometime next week, hopefully. Cannot make any promises right now.

danifpdra, 2022-09-15:
Not a problem. If you use the already recorded dataset you can run it easily. It calibrates fast.

miguelriemoliveira, 2023-03-31:
This is a strange problem. @JorgeFernandes-Git  was calibrating a hand eye case and it worked fine, right @JorgeFernandes-Git ?

JorgeFernandes-Git, 2023-03-31:
Hi @miguelriemoliveira, @danifpdra.

The hand eye calibration went smoothly, although I didn't use the depth data for it.

Let me know if there's anything else I can do to help.

miguelriemoliveira, 2023-06-06:
Closing for now. @JorgeFernandes-Git used a hand eye calibration and it worked.

closed 2023-06-06 by miguelriemoliveira