lardemua/atom #533: ZED camera to INS Calibration

alexlin2, 2022-08-30:
Problem Statement:

I have an INS and a ZED camera that are rigidly attached to each other with a fixed transform. I want to get the transform between INS and ZED. Below is the transform summary pdf. The INS gives 2cm accuracy RTK corrected Odometry. The camera is approximately 20 cm from the pose of the center of the INS. Attached below is also a photo of the INS and the camera. 
[summary.pdf](https://github.com/lardemua/atom/files/9456543/summary.pdf)
[image]
[image]

Issues that I've run into:

- There is no example of calibrating a ZED camera relative to a fixed joint, there is no direct link between `zed2i_left_camera` and `ins_link`. There is a link between `zed_camera_center` and `ins_link`, but `zed_camera_center` does not have an optical frame that produces a camera topic. How should we configure `config.yaml` so that we can calibrate a transform between camera_center and `base_link` (`ins_link` in my case)

- What should I set the `world_link` and `parent_link` of the charuco tag? The predicted pose of the charuco tag is in the ZED camera frame, but setting `parent_link` of the charuco tag to `zed2i_camera_center` would give just the transform set in initial calibration.

- I am moving the INS and camera around the charuco tag. I found that keeping the INS stationary while moving the charuco tag does not produce a feasible transform. I have not seen other people do that, in all of the examples, the charuco tag is moved relatively to a stationary robot. 

This is my current calibration file 
```
description_file: "package://vision_processing_module_calibration/urdf/vision_processing_module.urdf.xacro"

# The calibration framework requires a bagfile to extract the necessary data for the calibration.
bag_file: "/home/alexlin/2022-08-30-17-53-48.bag"

# You must define a frame of reference for the optimization process.
# It must exist in the transformation chains of all the sensors which are being calibrated.
world_link: "odom"

# ATOM will calibrate the extrinsic parameters of your sensors.
# In this section you should discriminate the sensors that will be part of the calibrations.
sensors:
  # Each key will define a sensor and its name, which will be use throughout the calibration.
  # Each sensor definition must have the following properties:
  #       link:
  #           The frame of the sensor's data (i.e. the header.frame_id).
  #
  #       parent_link:
  #           The parent link of the transformation (i.e. link) to be calibrated.
  #
  #       child_link:
  #           This is the transformation (i.e. link) that we be optimized.
  #
  #       topic_name:
  #           Name of the ROS topic that contains the data produced by this sensor.
  #           If you are calibrating an camera, you should use the raw image produced by the
  #           sensors. Additionally, it the topic is an image it will automatically use the
  #           respective `camera_info` topic.
  # OPTIONAL
  #       throttle:
  #           Set throttle: desired_frame_rate. If you don't use throttle, collapse the throttle option
  #       modality:
  #           Set this to identify the sensor modality. If this flag is not set, the sensor will be
  #           identified by the message type.
  #           Current options: lidar3d, rgb, depth, lidar2d
  #
  # If you are using an image compressed topic such as:
  #   /world_camera/rgb/image_raw/compressed
  # you should not add the "compressed" part, use only:
  #   /world_camera/rgb/image_raw
  #

  left_camera:
          link: "zed2i_left_camera_optical_frame"
          parent_link: "ins_link"
          child_link : "zed2i_camera_center"
          topic_name : "/sensors/stereo/left/image_rect_color"
          modality: rgb

# The calibration requires a detectable pattern.
# This section describes the properties of the calibration pattern used in th calibration.
calibration_pattern:

  # The frame id (or link) of the pattern.
  # This link/transformation will be optimized.
  link: "pattern_link"

  # The parent frame id (or link) of the pattern.
  # For example, in hand-eye calibration the parent link
  # of the pattern can be the end-effector or the base of the arm
  parent_link: "odom"

  # Defines if the pattern link is the same in all collections (i.e. fixed=true),
  # or each collection will have its own estimate of the link transformation.
  fixed: true

  # The type of pattern used for the calibration.
  # Supported pattern are:
  # - chessboard
  # - charuco
  pattern_type: "charuco"

  # If the pattern type is "charuco" you need to define
  # the aruco dictionary used by the pattern.
  # See https://docs.opencv.org/trunk/dc/df7/dictionary_8hpp.html
  dictionary: "DICT_4X4_50"

  # Mesh file (collada.dae or stl) for showing pattern on rviz. URI or regular path.
  # See: description_file
  mesh_file: "package://atom_calibration/meshes/charuco_5x5.dae"

  # The border width from the edge corner to the pattern physical edge.
  # Used for 3D sensors and lidars.
  # It can be a scalar (same border in x and y directions), or it can be {'x': ..., 'y': ,,,}
  border_size: { 'x': 0.025, 'y': 0.025 }

  # The number of corners the pattern has in the X and Y dimensions.
  # Note: The charuco detector uses the number of squares per dimension in its detector.
  # Internally we add a +1 to Y and X dimensions to account for that.
  # Therefore, the number of corners should be used even for the charuco pattern.
  dimension: { "x": 4, "y": 4 }

  # The length of the square edge.
  size: 0.11

  # The length of the charuco inner marker.
  inner_size: 0.087

# Miscellaneous configuration

# If your calibration problem is not fully constrained you should anchored one of the sensors.
# This makes it immovable during the optimization.
# This is typically referred to as gauge freedom.
anchored_sensor: ""

# Max time delta (in milliseconds) between sensor data messages when creating a collection.
max_duration_between_msgs: 1000
```

miguelriemoliveira, 2022-08-30:
Hi @alexlin2 ,

thanks for posting here. We never had a calibration case with and INS system implemented in ATOM.
Let me think about it for a couple of days and get back to you until the end of this week.

alexlin2, 2022-08-30:
> Hi @alexlin2 ,
> 
> thanks for posting here. We never had a calibration case with and INS system implemented in ATOM. Let me think about it for a couple of days and get back to you until the end of this week.

I am trying a lot of different parameters and I've got some reasonable calibrations. But I'm not sure if this is just because I have a really good initial estimate. I'd love to hear back from you. Thank you so much.

miguelriemoliveira, 2022-08-30:
> I am trying a lot of different parameters and I've got some reasonable calibrations. But I'm not sure if this is just because I have a really good initial estimate. 

you can get an answer to that question by testing with some noise on top of the initial estimate.
Use the --noisy-initial-guess (-nig) option in the calibrate script, and see if ATOM still finds a good solution.

https://lardemua.github.io/atom_documentation/procedures/#calibrate

miguelriemoliveira, 2022-09-02:
Hi @alexlin2 ,

So finally I had a bit of time to look into this.

First thing is your summary.pdf is strange.

   1. The odometry information published by the INS is providing transformations from **odom
(world link)** to **ins_link**? If so that sounds of, provided that the **map** to **odom(world link)** remains static.

   2. Second strange thing is that you are trying to calibrate **ins_link** to **zed2i_camera_center** and in it says this transformation is dynamic. It should be static! It could be a bug from the software we have producing the pdf, but we should check it our, because only static transformations are possible to calibrate. If you can share your code I can look into it... if not I can try to give a few tips to search yourself.

We never did this kind of calibration, but after thinking a bit about it I think atom should be able to solve the problem. You must have a static calibration pattern (you are correctly marking it as fixed in the config.yaml), and then move the system around the pattern in order to collect different camera images of the pattern. The INS will provide the pose of the **ins_link** for each of these images (we call them collections in atom) and the calibration will estimate **the ins_link**  to **zed2i_camera_center** transformation.

This sound right. I would expect atom to be able to calibrate.

Note that ATOM would only provide the above mentioned transformation, and that its accuracy depends on the accuracy of the data provided by the INS.

About the problems you identify:

> There is no example of calibrating a ZED camera relative to a fixed joint, there is no direct link between zed2i_left_camera and ins_link. There is a link between zed_camera_center and ins_link, but [...]

ATOM can hadle this. Actually, the name ATOM comes from Atomic Transformations Optimization Method. That means you can select one transformation to be optimized without requiring a direct connection between the camera and the ins. 
The only constrains is that the transformation you define to be optimized must be somewhere in the chain from the sensor to the ins link. The way you configure seems correct.

> What should I set the world_link and parent_link of the charuco tag? The predicted pose of the charuco tag is in the ZED camera frame, but setting parent_link of the charuco tag to zed2i_camera_center would give just the transform set in initial calibration.

To answer that I would first need to understand why you have an odom and a map frame. I mean, whos publishing these two different dynamic transformations map->odom and odom->ins ?

Assuming the map->odom is static, you can set the world link to the odom and set the charuco parent link also to the odom.

> I am moving the INS and camera around the charuco tag. I found that keeping the INS stationary while moving the charuco tag does not produce a feasible transform. I have not seen other people do that, in all of the examples, the charuco tag is moved relatively to a stationary robot.

ATOM is very general and we have several cases in which ATOM can tackle. Most often we move the calibration pattern, but in this case it should be fixed. We did a case of [eye-on-hand calibration where the problem was similar](https://www.youtube.com/watch?v=62lKtvloU70&list=PLQN09mzV5mbI4h5IQt3Eu9kugSk-08mnY&index=21)

the difference being that we moved the camera with a robotic manipulator, and we would get the pose estimate using the encoders, while you are moving the camera by hand, and getting the pose estimate using the INS.

You can read more of this type of ATOM application [here](https://doi.org/10.1109/TRO.2021.3062306).

Hope it helps,
Miguel

.

alexlin2, 2022-09-02:
> Hi @alexlin2 ,
> 
> So finally I had a bit of time to look into this.
> 
> First thing is your summary.pdf is strange.
> 
> 1. The odometry information published by the INS is providing transformations from **odom
>    (world link)** to **ins_link**? If so that sounds ok, provided that the **map** to **odom(world link)** remains static.
> 2. Second strange thing is that you are trying to calibrate **ins_link** to **zed2i_camera_center** and in it says this transformation is dynamic. It should be static! It could be a bug from the software we have producing the pdf, but we should check it our, because only static transformations are possible to calibrate. If you can share your code I can look into it... if not I can try to give a few tips to search yourself.
> 
> We never did this kind of calibration, but after thinking a bit about it I think atom should be able to solve the problem. You must have a static calibration pattern (you are correctly marking it as fixed in the config.yaml), and then move the system around the pattern in order to collect different camera images of the pattern. The INS will provide the pose of the **ins_link** for each of these images (we call them collections in atom) and the calibration will estimate **the ins_link** to **zed2i_camera_center** transformation.
> 
> This sound right. I would expect atom to be able to calibrate.
> 
> Note that ATOM would only provide the above mentioned transformation, and that its accuracy depends on the accuracy of the data provided by the INS.
> 
> About the problems you identify:
> 
> > There is no example of calibrating a ZED camera relative to a fixed joint, there is no direct link between zed2i_left_camera and ins_link. There is a link between zed_camera_center and ins_link, but zed_camera_center does not have an optical frame that produces a camera topic. How should we configure config.yaml so that we can calibrate a transform between camera_center and base_link (ins_link in my case)
> 
> ATOM can hadle this. Actually, the name ATOM comes from Atomic Transformations Optimization Method. That means you can select one transformation to be optimized without requiring a direct connection between the camera and the ins. The only constrains is that the transformation you define to be optimized must be somewhere in the chain from the sensor to the ins link. The way you configure seems correct.
> 
> > What should I set the world_link and parent_link of the charuco tag? The predicted pose of the charuco tag is in the ZED camera frame, but setting parent_link of the charuco tag to zed2i_camera_center would give just the transform set in initial calibration.
> 
> To answer that I would first need to understand why you have an odom and a map frame. I mean, whos publishing these two different dynamic transformations map->odom and odom->ins ?
> 
> Assuming the map->odom is static, you can set the world link to the odom and set the charuco parent link also to the odom.
> 
> > I am moving the INS and camera around the charuco tag. I found that keeping the INS stationary while moving the charuco tag does not produce a feasible transform. I have not seen other people do that, in all of the examples, the charuco tag is moved relatively to a stationary robot.
> 
> ATOM is very general and we have several different cases which ATOM can tackle. Most often we move the calibration pattern, but in this case it should be fixed. We did a case of [eye-on-hand calibration where the problem was similar](https://www.youtube.com/watch?v=62lKtvloU70&list=PLQN09mzV5mbI4h5IQt3Eu9kugSk-08mnY&index=21) the difference being that we moved the camera with a robotic manipulator, and we would get the pose estimate using the encoders, while you are moving the camera by hand, and getting the pose estimate using the INS.
> 
> You can read more of this type of ATOM application [here](https://doi.org/10.1109/TRO.2021.3062306).
> 
> Hope it helps, Miguel
> 
> .

Hi Miguel,

Thank you so much for looking into this. This is of huge importance to my project. First, I'm going to answer some of your questions. 

1. The INS is providing transforms from odom to ins_link. There is only a static transform between map and odom, and this transform is not necessary at all so I will remove it. Also the transform between `ins_link` and `zed2i_camera_center` should be 
static. It is defined in the launch file that publishes a static transform between them.

2. All of the static transforms are defined in this launch file below, this is pretty much all of the code to launch everything. 
```
<?xml version="1.0"?>
<launch>

  <arg name="odom_child" default="ins_link"/>
 
  <include file="$(find zed_svo_recorder)/launch/zed2i.launch">
    <arg name="base_frame" value="$(arg odom_child)" />
  </include>
  <include file="$(find tc_localizer)/launch/microstrain.launch">
    <arg name="filter_child_frame_id" value="$(arg odom_child)" />
  </include>

  <node name="relay_gps_relative_odom" pkg="topic_tools" type="relay"
    output="screen"
    args="/sensors/gps/nav/relative_pos/odom odom">
  </node>

  <node name="map_odom_broadcaster" pkg="tf" type="static_transform_publisher"
      args="0 0 0 0 0 0 map odom 15"
      output="screen" />

  <node name="odom_to_base_link" pkg="tf" type="static_transform_publisher"
          args="0 -0.05 -0.13 0 1.0472 0  ins_link zed2i_camera_center 10"
          output="screen" />

</launch>

```

[... cut ...] rguments I used to configure atom: `rosrun atom_calibration configure_calibration_pkg -utf -n vision_processing_module_calibration`

It is also very strange to me why atom is thinking that the transform between `ins_link` and `zed2i_camera_center` is dynamic. Because in the launch file it is clearly publishing a static transform. It might be a bug somewhere in the code.
 
Thank you again for looking into this. I really appreciate your help. If you can look into that bug it would be very helpful.

miguelriemoliveira, 2022-09-13:
Hi @alexlin2 ,

Can you let me know if ATOM was successful in calibrating the setup you described?

Thanks,
Miguel

closed 2022-09-13 by miguelriemoliveira