lardemua/atom #534: Configure calibration pkg detecting static transform as dynamic

miguelriemoliveira, 2022-09-02:
Appeared from #533 ,

> It is also very strange to me why atom is thinking that the transform between ins_link and zed2i_camera_center is dynamic. Because in the launch file it is clearly publishing a static transform. It might be a bug somewhere in the code.

miguelriemoliveira, 2022-09-02:
The problem should be here ...

https://github.com/lardemua/atom/blob/8e51fa5cf5ad5b13b268c7ccafa8c604c02a324d/atom_calibration/scripts/configure_calibration_pkg#L146-L179

miguelriemoliveira, 2022-09-02:
Hi @alexlin2,

The way the configure calibration pkg script (the code above) detects if a transform is static or not is by checking if transformation is published on the  **/tf** or  **/tf_static** topic.

If you take a look at the [tf2 migration guide](http://wiki.ros.org/tf2/Migration) they suggest the usage of **tf_static** for static for static transforms.

So I would say that  ATOM's assumption for detecting static transforms makes sense, and I suggest that you publish static transforms in the topic **/tf_static**.

For this you can either remap the tf to tf_static, if you are using the old tf static_transform_publisher, e.g.

    rosrun tf static_transform_publisher 0 0 0 0 0 0 frame1 frame2 10 tf:=tf_static

or you could just use the newer tf2 static_transform publisher, e.g.

    rosrun tf2_ros static_transform_publisher 0 0 0 0 0 0 frame1 frame2 10

alexlin2, 2022-09-06:
> Hi @alexlin2,
>
> The way the configure calibration pkg script (the code above) detects if a transform is static or not is by checking if transformation is published on the **/tf** or **/tf_static** [...]

Sorry I missed that. I will try the calibration with the static transform published in /tf_static. And I will keep you updated on the results.