lardemua/atom #535: Problem configuring mysoftbot

miguelriemoliveira, 2022-09-18:
Hi   @eupedrosa ,

I was trying to test your bag, but when tring to configure the mysoftbot_calibration package I get this error:

```terminal
Traceback (most recent call last):
  File "/home/mike/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/mike/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 366, in <module>
    with open(launch_file, 'wb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/mike/catkin_ws_mysoftbot/src/mysoftbot/mysoftbot_calibration/launch/playbag.launch'
```

I will look into it later.

eupedrosa, 2022-09-19:
Should't the launch file be auto-generated?

miguelriemoliveira, 2022-09-19:
Yes, its a strange problem, but I've had it before in #498 .

I will check it ...

miguelriemoliveira, 2022-09-19:
Got it. The problem is when the directories launch , urdf, etc, that were supposed to be there do not exist. Now we create them if they are not there.

miguelriemoliveira, 2022-09-19:
Another issue ... cannot fin the urdf 

```
No such file or directory: /home/mike/catkin_ws_mysoftbot/src/mysoftbot/mysoftbot_calibration/urdf/initial_estimate.urdf.xacro [Errno 2] No such file or directory: '/home/mike/catkin_ws_mysoftbot/src/mysoftbot/mysoftbot_calibration/urdf/initial_estimate.urdf.xacro'
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '/home/mike/catkin_ws_mysoftbot/src/mysoftbot/mysoftbot_calibration/urdf/initial_estimate.urdf.xacro']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro $(arg description_file)"/>
The traceback for the exception was written to the log file
```

miguelriemoliveira, 2022-09-19:
Solved. Had to create the nonexistent folders at the beginning of the script.

miguelriemoliveira, 2022-09-19:
Hi @eupedrosa ,

so finally I could test your bagfile.  Comments:

- In general it looks good.
- I would make sure the height of the pattern is enough for the automatic labeler to work.
- Your camera topics are at 30 hz, no need for so much. I normally put 10Hz for recording the calibration bagfile.
- for consistency, I would create a namespace for the velodyne and put the topics under it, something like **lidar3d/points**
- Sensor names in the calibration.yaml and the topic namespaces should coincide (cameras are ok).
- I would make the bag file with pauses of 3 secs every time you are aiming to save a collection.  Check this description https://lardemua.github.io/atom_documentation/procedures/#collect-data

Thanks for the work. Let me know if I can help more.

miguelriemoliveira, 2022-10-05:
Hi @eupedrosa ,

any news?

eupedrosa, 2022-10-05:
Still on hold, but not forgotten!!

miguelriemoliveira, 2022-10-05:
OK, great.

closed 2023-03-31 by miguelriemoliveira