lardemua/atom #536: RViz fork integration failure

kovlo, 2022-09-20:
collectdata.launch and set_initial_estimate.launch gives the following error:

`[ERROR] [1663681699.871330719]: PluginlibFactory: The plugin for class 'rviz_visual_tools/KeyTool' failed to load.  Error: According to the loaded plugin descriptions the class rviz_visual_tools/KeyTool with base class type rviz::Tool does not exist. Declared types are  rviz/FocusCamera rviz/Interact rviz/Measure rviz/MoveCamera rviz/PublishPoint rviz/Select rviz/SetGoal rviz/SetInitialPose rviz_plugin_selected_points_publisher/SelectedPointsPublisher rviz_plugin_tutorials/PlantFlag`

The RViz window opens, but the ATOM calibration window (visible in the lower left corner of your calibration videos on youtube https://www.youtube.com/watch?v=eII_ptyMq5E.) does not seem to be loaded. How can I activate that menu/tab in the GUI?

Screenshot of the GUI visible for me:
[image]

During installation, I cloned the mentioned rviz fork to /src under my ros workspace, followed by calling catkin_make.

Please let me know if I made a mistake at installation or elsewhere.

Thanks in advance!

miguelriemoliveira, 2022-09-20:
Hi @kovlo ,

Two separate problems:

1. The ATOM calibration RVIZ plugin that appears on the video is not available yet, as we do not have it robust enough for release.

2. The crash you are having is another problem. Some ideas:

Key tool is not an ATOM plugin, so I am not really sure what's wrong.

a) Did you install rviz visual tools?

   sudo apt-get install ros-noetic-rviz-visual-tools

b) Another thing, the rviz fork you downloaded must be compiled. Did you compile the catkin workspace where the rviz fork is?

c) Also, in order to use this functionalities ATOM must be compiled, in particular the [selected_points_publisher](https://github.com/lardemua/atom/tree/noetic-devel/atom_rviz_plugins/selected_points_publisher) rviz plugin.

kovlo, 2022-09-21:
Hi @miguelriemoliveira !

Thanks for your quick replies!
a) was missing for me, thanks for the help!

This caused the error! This does not seem to be a real bug but only a missing package.

#######################

I collected the required additional packages, they might be shown in the installation notes and/or requirements.txt:
```
sudo apt-get install ros-noetic-rviz-visual-tools python3-tk
pip3 install -U setuptools
pip3 install readchar urdf-parser-py SQLAlchemy open3d pytictoc
```

b) My installation steps were:
- I cloned the atom repo to _project/src/_
- installed the python3 requirements.txt
- I cloned the rviz fork near the atom folder at _project/src/_
- I called catkin_make on the project folder
c) do I have to compile separately the selected_points_publisher rviz plugin.

Current status:
- The playback of the pointcloud and the camera-Labels-Image show the data from the recorded bag.
- I can add a new panel RvizVisualToolsGui. The four buttons are shown, but their function do not work.
- I can run the initial estimation
- the collect_data runs, I can se all required panels.

But unfortunately now I'm struggling with the detection of my charuco board. (8x6 board with 2cm edges, 12cm size, 8cm inner size, 4x4 dictionary.) It currently detects only the half of it:
[image]

If you have any suggestions for me, please let me know!

Thank you very much!

closed 2022-09-21 by kovlo

miguelriemoliveira, 2022-09-21:
> c) do I have to compile separately the selected_points_publisher rviz plugin.

if both ATOM and the rviz fork are on the same catkin workspace, then you only have to run catkin_make once.

I have them in separate workspaces...

kovlo, 2022-09-21:
Great, thanks!
I have both the ATOM and the rviz fork in the same workspace, they seem to work fine.