lardemua/atom #537: Charuco pattern detection not working

miguelriemoliveira, 2022-09-21:
Spawned from #536 . 

Detection of pattern not working for @kovlo

miguelriemoliveira, 2022-09-21:
The usual problem is a problem with the configuration of the pattern in the calibration/config.yaml of your calibration package.

Can you post your config.yaml so I can take a look?

Also, you can make sure the configuration of the pattern is correct by using the test script view pattern, e.g.

    rosrun atom_calibration view_pattern.py

which has to be properly configured in order to work. This way you can be sure your parameters are correct.

kovlo, 2022-09-23:
I checked my config line by line, and I configured my pattern with wrong row and column sizes.
Now the patterns seem to be detected, the rainbow colored markings are visible on the GUI as well, and I can run the calibration. But the results are not satisfying, I'm working on it to figure them out.

miguelriemoliveira, 2022-09-23:
OK, let me know if you think I can help.

kovlo, 2022-09-23:
> rosrun atom_calibration view_pattern.py

This is what is shows for this call: 
`rosrun atom_calibration view_pattern.py /camera/image_color -t charuco -d DICT_4X4_100 -x 7 -y 5 -L 120 -l 80`
Is this what I should expect?
[image]

When I run the calibration
(`rosrun atom_calibration calibrate -json /data/ATOM_Result/dataset.json -v -rv -si -oi`),
I observe the following:
- the board shown in the 3D view lies in an inaccurate orientation and position
- this is visible in the displayed image as well.
- the "texture" of the board is shown in small size between the colored marks.
- only one table is shown in the 3D window, while in your video all collection has its own 3d board overlaid
[image]

I use calib.io_charuco_760x1000_8x6_120_93_DICT_4X4 board. The pdf was converted to svg and to dae using online tools, and it was specified in the calibration.yaml's mesh_file field. But this dae is not exactly the same as the calibration board (80mm vs 92mm inner figure size). When I used the original dae path, the behavior was the same.

Also, I have the intrinsic parameters of the camera, should I add them to the pointgrey camera driver when I record the calibration bag file?

If you have any suggestions, please share them with me.
Thanks again!

miguelriemoliveira, 2022-09-23:
> This is what is shows for this call:
> rosrun atom_calibration view_pattern.py /camera/image_color -t charuco -d DICT_4X4_100 -x 7 -y 5 -L 120 -l 80
> Is this what I should expect?

No, I don't think its right, the circles should all be on the corners of the pattern. Is your dictionary correct?

eupedrosa, 2022-09-23:
Hi there. I belive the size must be in SI, i.e in meters not in millimiters.
Also, make sure the X and Y are not switched.

kovlo, 2022-09-26:
Thank you @miguelriemoliveira  and @eupedrosa for your comments!
I set the W,H in wrong order as you pointed out, and even the sizes were in mm in the script I used (the config.yaml was good.)

This is what I see now, I hope it is closer to the expected:
[image]
[image]

miguelriemoliveira, 2022-09-26:
That's seems right. 

Let us know if you have other problems ahead.

closed 2022-09-26 by miguelriemoliveira