lardemua/atom #541: Problem with opencv

manuelgitgomes, 2023-01-26:
Hello @miguelriemoliveira!

When installing ATOM in a new PC, a problem arose with the `requirements.txt`. Because no versions were specified, it installed the latest version. In the latest version of opencv (4.7.0), the Aruco code (such as  `cv2.aruco.CharucoBoard_create()`) was moved from opencv-contrib to the main repository.
This causes ATOM to break.
Two possible solutions for this are:

- Define an old version where this error does not occur
- Use the current opencv version and change the code in ATOM

What do you think the better route is?

miguelriemoliveira, 2023-01-26:
Is the new computer using 20.04?

If so then i recommend the second option.

manuelgitgomes, 2023-01-26:
Yes it is!
Ok, I can make those changes ASAP.

miguelriemoliveira, 2023-01-26:
Great. Thanks.

manuelgitgomes, 2023-01-27:
Hello @miguelriemoliveira!

After changing the functions to their new names and configured some parameters, the script did not function correctly.
The changed code in a new branch and can be seen here:

https://github.com/lardemua/atom/blob/fa117ec275083a7ee555eeeb5ff6a1d570fcad91/atom_calibration/src/atom_calibration/collect/patterns.py#L112-L118

The error occurs inside line 118. I tried looking at opencv source code, but failed to extract any conclusion.

The only documentation on this is [here](https://docs.opencv.org/4.7.0/d9/df5/classcv_1_1aruco_1_1CharucoDetector.html), as this is a very recent version of opencv (it came out in the 29th of December).

To make things worse, the code outputs without an error message, leaving me completely clueless on why this is happening.

Can you help me?

Thank you!

miguelriemoliveira, 2023-01-28:
Hi @manuelgitgomes ,

I will try to help. Which branch are you working on?

manuelgitgomes, 2023-01-28:
Thanks! It is called `bug/opencv`!

miguelriemoliveira, 2023-02-06:
Sorry @manuelgitgomes ,

this is not forgotten, I simply have not had any time at all.
Hopefully next week.

manuelgitgomes, 2023-02-07:
No problem!

kovlo, 2023-02-07:
For me the following package specification solved the issue:
pip install opencv-contrib-python==4.6.0.66

manuelgitgomes, 2023-02-07:
Thank you for the support @kovlo!

Nevertheless, I believe we are trying to make ATOM up to date with the newest version of OpenCV.

If this is not possible/too difficult, we will accept your PR.

Best regards!

kovlo, 2023-02-07:
Great! I agree that using the newest openCV is a good idea.
But today I installed the framework from scratch, and it failed to run due to this issue. The proposed solution maintains the usability of the package for new/reinstalls just by specifying the previous opencv version, while the proper modifications for the new opencv versions are implemented.

miguelriemoliveira, 2023-02-09:
Thanks @manuelgitgomes and @kovlo for the help.

I agree that we could accept @kovlo 's solution until we find a way to make it work with the latest opencv.

@manuelgitgomes can you take a look at the PR please?

manuelgitgomes, 2023-02-09:
Hello @miguelriemoliveira and @kovlo 

I have taken a look at the PR, and tested in my machine and it works.

It is now merged, thanks!

miguelriemoliveira, 2023-05-12:
Did not have time to work on this. For now, @kovlo 's solution is working well, so I am closing the issue.

miguelriemoliveira, 2023-06-06:
Seems to be working for now. Closing.

closed 2023-06-06 by miguelriemoliveira