lardemua/atom #547: Improve the function saveResultsXacro

JorgeFernandes-Git, 2023-02-26:
This function was adapted to accommodate the changes made to calibrate `additional_tfs`:

https://github.com/lardemua/atom/blob/2bbee3117b37e1053dea31a347c2e98d756b23f5/atom_core/src/atom_core/xacro_io.py#L31

Was added a third argument `transforms_list`. This argument contain a list composed of the parent-child keys to be calibrated.

Here, we iterate each key and separate the parent from the child:
https://github.com/lardemua/atom/blob/2bbee3117b37e1053dea31a347c2e98d756b23f5/atom_core/src/atom_core/xacro_io.py#L42-L45

To be used in this if statement:
https://github.com/lardemua/atom/blob/2bbee3117b37e1053dea31a347c2e98d756b23f5/atom_core/src/atom_core/xacro_io.py#L52

This approach works, but improvements can be made.

miguelriemoliveira, 2023-03-02:
@JorgeFernandes-Git all the code you are developing is still under your branch?

miguelriemoliveira, 2023-03-02:
Hi @JorgeFernandes-Git ,

I looked into this function and for now I think we can leave it as is. Closing the issue.

closed 2023-03-02 by miguelriemoliveira

JorgeFernandes-Git, 2023-03-04:
Hi @miguelriemoliveira.
Just seeing this now.

Yes all code I developed is under my branch.