lardemua/atom #549: Save results from calibration and evaluations as a `.csv` file

JorgeFernandes-Git, 2023-03-03:
When running the calibration, if the user wants to save the results in an CSV file.
Usage example:

>rosrun atom_calibration calibrate -json $ATOM_DATASETS/... -v -rv -si -vo **-sfr 'path to folder'**

-sfr stands for save file results.

I'll implement something similar to the evaluation scripts. 
Suggestions are welcome @miguelriemoliveira and @manuelgitgomes.

Committed changes:
https://github.com/lardemua/atom/commit/6d3361941f96f2730be3fb09d6e797d75cc1350b

miguelriemoliveira, 2023-03-03:
Sounds good.

JorgeFernandes-Git, 2023-03-03:
I add the same principle to the evaluation scripts that print tables.

Some of them **need testing**, at the moment I don't have datasets to test it. Those are:
* lidar_to_rgb_evaluation
* lidar_to_lidar_evaluation
* depth_to_rgb_evaluation
* cross_collection_lidar_to_rgb_evaluation
* cross_collection_lidar_to_lidar_evaluation

JorgeFernandes-Git, 2023-03-04:
@miguelriemoliveira I'm closing this issue, since the task is done.

Some scripts need testing, but if errors were found they will be minor.

closed 2023-03-04 by JorgeFernandes-Git