lardemua/atom #552: Calibration evaluation output file

Kazadhum, 2023-03-08:
Hi @JorgeFernandes-Git and @miguelriemoliveira! I'm working on making a plugin for Rigel that validates the results of the calibration evaluation. As I understand it, ATOM currently outputs a file with these results. Could you give me some info on what these results look like in terms of structure?

miguelriemoliveira, 2023-03-08:
Hi @JorgeFernandes-Git ,

can you tell @Kazadhum how to run the rgb to rgb evaluation in order to produce a file?

JorgeFernandes-Git, 2023-03-09:
Hi @miguelriemoliveira. 
Yesterday I couldn't do it, but I called @Kazadhum and talked to him about this.

I rerun the t2rgb with noise initial guess `nig` 0.1 0.1. However, based on what @Kazadhum told me, the tests probably want more reliable results, so no `nig`. Either way, here are the 3 datasets.

Data collection:
Dataset: https://drive.google.com/file/d/1Leb1uupE2FRVqHaeOY5wCuavORyhlhP4/view?usp=share_link

Calibration without `nig`
Atom_calibration: https://drive.google.com/file/d/1t6oTzi_Upb9Yj0CJaCyEj5ROW2-NQQBM/view?usp=share_link

Calibration with `nig` 0.1 0.1:
Atom_calibration_nig_0.1_0.1: https://drive.google.com/file/d/1XHioImSyYnQ0Rstd1ds3DzLY_5DaNT9o/view?usp=share_link

_____________________________________
To run the evaluation without `nig`:
```
rosrun atom_evaluation rgb_to_rgb_evaluation -train_json ~/atom_calibration.json -test_json ~/dataset.json -ss right_camera -st left_camera -sfr ~/Desktop/
```

The `-sfr` flag, is for saving the results in an .csv file, but it's only on my branch of atom a.t.m. 
Here is the script: 
https://github.com/lardemua/atom/blob/JorgeFernandes-Git/issue543/atom_evaluation/scripts/rgb_to_rgb_evaluation

Just replace it.
_____________________________________
Results:
https://drive.google.com/file/d/1H05Oog6skurVr-VUlljijpPs8WDzedHo/view?usp=sharing

| Collection # | RMS (pix) | X err (pix) | Y err (pix) | Trans (mm) | Rot (deg) |
|--------------|-----------|-------------|-------------|------------|-----------|
|     000      |   0.6728  |    0.2344   |    0.1950   |   1.1937   |   0.0943  |
|     001      |   0.9645  |    0.4505   |    0.1992   |   0.9350   |   0.3064  |
|     002      |   0.9748  |    0.4043   |    0.2292   |   0.9052   |   0.4921  |
|     003      |   0.6913  |    0.2870   |    0.1196   |   0.6552   |   0.1432  |
|     004      |   1.0504  |    0.4688   |    0.2139   |   1.2206   |   0.1286  |
|     005      |   1.0161  |    0.4543   |    0.2597   |   3.7918   |   0.6532  |
|     006      |   1.7904  |    0.6863   |    0.6689   |   3.5626   |   0.8756  |
|     007      |   1.1596  |    0.7077   |    0.2700   |   1.7165   |   0.1138  |
|     008      |   1.1571  |    0.4282   |    0.2591   |   0.7773   |   0.1470  |
|     009      |   0.7931  |    0.3688   |    0.2147   |   0.3806   |   0.2895  |
|     010      |   0.6359  |    0.2425   |    0.1413   |   2.6483   |   0.3653  |
|     011      |   0.6140  |    0.1915   |    0.1478   |   1.5099   |   0.2399  |
|     012      |   0.7708  |    0.2404   |    0.1998   |   0.7713   |   0.1853  |
|     013      |   0.4231  |    0.1673   |    0.1195   |   0.5489   |   0.2427  |
|     014      |   0.6227  |    0.2287   |    0.1581   |   0.3730   |   0.3643  |
|     015      |   0.6223  |    0.1832   |    0.1451   |   2.0446   |   0.1788  |
|     016      |   0.5866  |    0.2148   |    0.1218   |   2.8953   |   0.4397  |
|     017      |   0.6718  |    0.3038   |    0.2089   |   1.3041   |   0.3109  |
|     018      |   0.7126  |    0.2355   |    0.2051   |   1.0623   |   0.1510  |
|     019      |   0.8364  |    0.3005   |    0.2224   |   1.5281   |   0.1040  |
|     020      |   0.7772  |    0.2767   |    0.1774   |   0.3027   |   0.2399  |
|     021      |   0.7669  |    0.2766   |    0.1896   |   1.4254   |   0.2247  |
|     022      |   0.4784  |    0.1991   |    0.1214   |   1.3561   |   0.2064  |
|     023      |   0.4625  |    0.1966   |    0.1373   |   0.3238   |   0.2167  |
|     024      |   0.6344  |    0.2682   |    0.1935   |   0.2392   |   0.3336  |
|     025      |   0.7248  |    0.3582   |    0.1618   |   4.4076   |   0.5480  |
|     026      |   0.8901  |    0.3155   |    0.2905   |   2.1256   |   0.3069  |
|     027      |   0.6728  |    0.2344   |    0.1950   |   1.1937   |   0.0943  |
|   Averages   |   0.7919  |    0.3187   |    0.2059   |   1.4714   |   0.2856  |

To run with `nig` just change the `train_json`.
Anything else, just tell me.

Kazadhum, 2023-03-10:
Hi @JorgeFernandes-Git and @miguelriemoliveira! Thank you for the quick help! If I recall correctly, professor, you mentioned there's a value you use to evaluate the quality of the calibration, that I could use for introspection. What value is that?

miguelriemoliveira, 2023-03-10:
Hi @Kazadhum  ,

you can use the average of the RMS (pix) . A good calibration has <1, a great calibration <0.5. I would say you can see if its under 2 or 3 pixels to check if all went well.

Kazadhum, 2023-03-11:
Hey @miguelriemoliveira!

Thanks for the help! It's now implemented.

closed 2023-03-31 by miguelriemoliveira