lardemua/atom #553: Ground truth frame evaluation script

JorgeFernandes-Git, 2023-03-10:
As we had been discussing @miguelriemoliveira and @manuelgitgomes, here are the developments on this script:

https://github.com/lardemua/atom/blob/JorgeFernandes-Git/issue543/atom_evaluation/scripts/ground_truth_frame_evaluation

```
usage: ground_truth_frame_evaluation [-h] -train_json TRAIN_JSON_FILE -test_json TEST_JSON_FILE [-tf TARGET_FRAME] [-sf SOURCE_FRAME] [-sfr SAVE_FILE_RESULTS]

optional arguments:
  -h, --help            show this help message and exit
  -train_json TRAIN_JSON_FILE, --train_json_file TRAIN_JSON_FILE
                        Json file containing train input dataset.
  -test_json TEST_JSON_FILE, --test_json_file TEST_JSON_FILE
                        Json file containing test input dataset.
  -tf TARGET_FRAME, --target_frame TARGET_FRAME
                        Target transformation frame.
  -sf SOURCE_FRAME, --source_frame SOURCE_FRAME
                        Source transformation frame. If no frame is provided, computes all estimated frames.
  -sfr SAVE_FILE_RESULTS, --save_file_results SAVE_FILE_RESULTS
                        Output folder to where the results will be stored.
```
_________________________
General usage:
```
rosrun atom_evaluation ground_truth_frame_evaluation -train_json ~/atom_calibration.json -test_json ~/dataset.json
```

| Transformation # | X (mm) | Y (mm) | Z (mm) | Roll (deg) | Pitch (deg) | Yaw (deg) | Trans (mm) | Rot (deg) |
| ------------------------------------ | -------- | --------- | --------- | ------------ | ------------- | ----------- | ------------ | ----------- |
| base_link_mb-base_link | 4.3836 | 0.4544 | 12.7781 | 0.0079 | 0.1234 | 0.2010 | 13.5168 | 0.2360 |
| camera_link-camera_rgb_frame | 1.5529 | 23.6514 | 11.9330 | 0.0442 | 0.0711 | 0.0488 | 26.4956 | 0.2463 |
| camera_mb_link-camera_mb_rgb_frame | 2.5032 | 23.1558 | 9.6644 | 0.0068 | 0.0164 | 0.1365 | 25.2162 | 0.1376 |

_________________________
Source sensor per collection:

```
rosrun atom_evaluation ground_truth_frame_evaluation -train_json ~/atom_calibration.json -test_json ~/dataset.json -sf camera
```

| Collection # | X (mm) | Y (mm) | Z (mm) | Roll (deg) | Pitch (deg) | Yaw (deg) | Trans (mm) | Rot (deg) |
| -------------- | -------- | --------- | --------- | ------------ | ------------- | ----------- | ------------ | ----------- |
| 000 | 1.4474 | 25.9952 | 16.2564 | 0.0573 | 0.2468 | 0.1017 | 30.6939 | 0.2730 |
| 001 | 1.4474 | 25.9952 | 16.2564 | 0.0573 | 0.2468 | 0.1017 | 30.6939 | 0.2730 |
| 002 | 2.6778 | 26.5256 | 15.2453 | 0.0033 | 0.2468 | 0.1112 | 30.7115 | 0.2707 |
| 003 | 0.7811 | 27.2805 | 15.0023 | 0.0300 | 0.2468 | 0.1086 | 31.1433 | 0.2713 |
.
.
.
| 066 | 2.6294 | 20.5422 | 14.1002 | 0.0861 | 0.1284 | 0.1508 | 25.0542 | 0.2160 |
| Averages | 1.5529 | 23.6514 | 11.9330 | 0.0442 | 0.0711 | 0.0488 | 26.4956 | 0.2463 |

miguelriemoliveira, 2023-03-10:
Hi @JorgeFernandes-Git ,

looks great. One comment though: in the case of the evaluation for a single transformation, the last case where you show the output for all collections, the idea is to provide a transformation by giving these two:

> -tf TARGET_FRAME, --target_frame TARGET_FRAME
>   -sf SOURCE_FRAME, --source_frame SOURCE_FRAME

In your example you only have -sf, shouldn't there also be a -tf?

Also, the table should have a header which says explicitly which transformation is being evaluated.

Keep up the good work.

JorgeFernandes-Git, 2023-03-10:
Hi @miguelriemoliveira.

You can provide the -tf, the target frame, and the program will verify if it exists and if it is on the calibration chain of the source frame. If both comes true, it provides the results.

If you omit the target frame, it will assign the first link in the calibration chain after the reference frame (world), e.g. the base_footprint. 

>Also, the table should have a header which says explicitly which transformation is being evaluated.

I'll work on that. Thanks.

miguelriemoliveira, 2023-03-12:
Hi @JorgeFernandes-Git ,

> If you omit the target frame, it will assign the first link in the calibration chain after the reference frame (world), e.g. the base_footprint.

In that case I would return an error saying that tf needs to be defined. But if you write a header on the table is alright because we'll know the target frame and the source frame.

JorgeFernandes-Git, 2023-03-12:
Hi @miguelriemoliveira.
The title for the table is like this a.t.m. for this command:
```
... -train_json ~/atom_calibration.json -test_json ~/dataset.json -sf camera_mb -tf base_link
```

```
+----------------------------------------------------------------------------------------------------------+
|                                     base_link to camera_mb_rgb_frame                                     |
+--------------+--------+---------+--------+------------+-------------+-----------+------------+-----------+
| Collection # | X (mm) |  Y (mm) | Z (mm) | Roll (deg) | Pitch (deg) | Yaw (deg) | Trans (mm) | Rot (deg) |
+--------------+--------+---------+--------+------------+-------------+-----------+------------+-----------+
|     000      | 0.1880 | 25.2856 | 0.5038 |   0.0198   |    0.0513   |   0.0591  |  25.2913   |   0.0807  |
|     001      | 0.1880 | 25.2856 | 0.5038 |   0.0198   |    0.0513   |   0.0591  |  25.2913   |   0.0807  |
|     002      | 0.1880 | 25.2856 | 0.5038 |   0.0198   |    0.0513   |   0.0591  |  25.2913   |   0.0807  |
|     003      | 0.1880 | 25.2856 | 0.5038 |   0.0198   |    0.0513   |   0.0591  |  25.2913   |   0.0807  |

...
...
...
```

miguelriemoliveira, 2023-03-12:
Nice!

manuelgitgomes, 2023-03-13:
Hello @JorgeFernandes-Git!

Looks great, congrats!

I do have some suggestions:

> -tf TARGET_FRAME, --target_frame TARGET_FRAME
>                         Target transformation frame.

Maybe write a better description for this argparse, similar to the one in the `-sf`.

And maybe create another argparse option to transform the table from all collections to the average, something like:

```
  -sac SHOW_ALL_COLLECTIONS, --show_all_collections SHOW_ALL_COLLECTIONS
                        In the outputted table, it displays the result for every collections, as opposed to the averages.
```

Does this make sense to you?

manuelgitgomes, 2023-03-13:
> You are suggesting to output only that last line by default and have a flag to output for all collections, right?

@JorgeFernandes-Git  that's basically it, yes!

JorgeFernandes-Git, 2023-03-13:
Done: 

```
+----------------------------------------------------------------------------------------------------------+
|                                     base_link to camera_mb_rgb_frame                                     |
+--------------+--------+---------+--------+------------+-------------+-----------+------------+-----------+
| Collection # | X (mm) |  Y (mm) | Z (mm) | Roll (deg) | Pitch (deg) | Yaw (deg) | Trans (mm) | Rot (deg) |
+--------------+--------+---------+--------+------------+-------------+-----------+------------+-----------+
|   Averages   | 0.1880 | 25.2856 | 0.5038 |   0.0198   |    0.0513   |   0.0591  |  25.2913   |   0.0807  |
+--------------+--------+---------+--------+------------+-------------+-----------+------------+-----------+
```

miguelriemoliveira, 2023-05-12:
This is working. Closing the issue.

closed 2023-05-12 by miguelriemoliveira