lardemua/atom #564: pattern in mmtbot is inverted in rviz

miguelriemoliveira, 2023-04-22:
Doing some experiments for comparing with #555 I noticed the pattern is inverted for mmtbot.

[image]

I think the problem is where the axis of the pattern are:

[image]

miguelriemoliveira, 2023-04-22:
The calibration works perfect though..., unlike #555, when I use a single camera sensor the starting point already has below 1 pixel error...

miguelriemoliveira, 2023-04-22:
The same is true for the nrp ... provided we use the same sensor in the optimization and for computing the first guess

[image]

miguelriemoliveira, 2023-04-22:
@danifpdra , do you also have the pattern inverted in mmtbot?

danifpdra, 2023-04-24:
No

[image]

miguelriemoliveira, 2023-04-24:
That's strange. Do you have mmtbot up to date?

danifpdra, 2023-04-24:
> That's strange. Do you have mmtbot up to date?

Yes, I've checkeed

miguelriemoliveira, 2023-04-24:
> > That's strange. Do you have mmtbot up to date?
> 
> Yes, I've checkeed

Strange. Let me check again.

miguelriemoliveira, 2023-04-24:
Well I cannot use your dataset because I have a crash

```
rosrun atom_calibration calibrate -json /home/mike/datasets/mmtbot/dataset_31_03_2021/data_collected.json -v -rv -si -uic -csf "lambda x: x not in ['001','002','003','004','005','006']" --phased -ssf "lambda x: x not in ['hand_camera']" 

Traceback (most recent call last):
  File "/home/mike/workspaces/catkin_ws/devel/lib/atom_calibration/calibrate", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 486, in <module>
    main()
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/calibrate", line 106, in main
    dataset, json_file = loadResultsJSON(args["json_file"], args["collection_selection_function"])
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_core/src/atom_core/dataset_io.py", line 64, in loadResultsJSON
    if not (sensor['modality'] == 'rgb' or sensor['modality'] == 'lidar3d' or sensor['modality'] == 'lidar2d' or
KeyError: 'modality'
INFO - 2023-04-24 11:05:41,034 - core - signal_shutdown [atexit]
```

I remember we added this modality key. How are you running this and on which dataset?

danifpdra, 2023-04-24:
> Well I cannot use your dataset because I have a crash
>
> ```
> rosrun atom_calibration calibrate -json /home/mike/datasets/mmtbot/dataset_31_03_2021/data_collected.json -v -rv -si -uic -csf "lambda x: x [...]

You asked me for the dataset we used in the paper... it's possible it's deprecated

danifpdra, 2023-04-24:
[small_test.zip](https://github.com/lardemua/atom/files/11309219/small_test.zip)

Try this one

danifpdra, 2023-04-24:
Anyway, you should have access to all datasets... I don't know if you remember, but you made a script to sync the dataset folder of everyone and made us all put our datasets there

miguelriemoliveira, 2023-04-24:
> Anyway, you should have access to all datasets... I don't know if you remember, but you made a script to sync the dataset folder of everyone and made us all put our datasets there

Right. I forgot : - )

miguelriemoliveira, 2023-04-24:
For this dataset its fine:

[image]

How / when did you create this dataset?

Can you test with [my dataset simple](https://drive.google.com/file/d/1CIGPd3QSNpdHRIGtwsCtYKOKInRVg0aS/view?usp=sharing)?

miguelriemoliveira, 2023-04-24:
Thanks. And your opencv version is?

mine is 4.6.0

danifpdra, 2023-04-24:
Mine is the same

miguelriemoliveira, 2023-04-24:
> Mine is the same

I am starting to get upset with this #"#$"#ng thing...

danifpdra, 2023-04-24:
> > Mine is the same
> 
> I am starting to get upset with this #"#$"#ng thing...

I don't know how to help you :(

miguelriemoliveira, 2023-04-24:
Hi @danifpdra ,

I developed a new view_pattern3d.py script which not only detects the pattern but also computes its pose in 3D and publishes a tf. I want to make sure you are detecting the pattern the same way I am.

Can you please run an experiment to compare with me?

First, download the simple bagfile from mmtbot.

Then, reconfigure your calibration package

    rosrun mmtbot_calibration configure

Then launch the bagfile playback:

    roslaunch mmtbot_calibration playbag.launch bag_rate:=0.4

and finally, launch the pattern detector scritpt:

    clear && rosrun atom_calibration view_pattern3d.py -t charuco -d DICT_5xX_100 -x 11 -y 8 -L 0.06 -tn /world_camera/rgb/image_raw -cit /world_camera/rgb/camera_info

The questions is how do you see the frame of the pattern. Mine in the image below is like this (the one more to the right):

[image]

So in my case z points away from the cameras ... is yours like this as well?

danifpdra, 2023-04-24:
Hi @miguelriemoliveira ,

I had a lot of errors running this in the file collect/patterns.py because of deprecated function of the cv2.aruco package...

Finally I manage to run it ,  results here

[image]

danifpdra, 2023-04-24:
Can you pull atom and check if it still works for you?

miguelriemoliveira, 2023-04-24:
Now I get this error:

```
[ERROR] [1682352449.499708, 839.695690]: bad callback: <bound method SimplePatternDetector.onImageReceived of <__main__.SimplePatternDetector object at 0x7f77dadf4ac0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/deprecated/view_pattern3d.py", line 82, in onImageReceived
    ret, rvecs, tvecs = cv2.solvePnP(objp[ids], np.array(corners, dtype=np.float32), K, D)
cv2.error: OpenCV(4.6.0) /io/opencv/modules/calib3d/src/solvepnp.cpp:831: error: (-215:Assertion failed) ( (npoints >= 4) || (npoints == 3 && flags == SOLVEPNP_ITERATIVE && useExtrinsicGuess) || (npoints >= 3 && flags == SOLVEPNP_SQPNP) ) && npoints == std::max(ipoints.checkVector(2, CV_32F), ipoints.checkVector(2, CV_64F)) in function 'solvePnPGeneric
```

miguelriemoliveira, 2023-04-24:
Hi @danifpdra ,

I had to roll back your changes because I could not make it work. We are back using the CharucoBoard_create... We can try to make it work for you in zoom.

miguelriemoliveira, 2023-04-24:
Upgraded the code to also show the coordinate system on the image. Z is pointing outwards

[image]

danifpdra, 2023-04-26:
Did you push the code to show the coordinated system? When I run that script no pattern appears

miguelriemoliveira, 2023-04-27:
> Did you push the code to show the coordinate system? When I run that script no coordinate system appears

Hi @danifpdra ,

sorry, only now I could return to this. I think I forgot. Can you try again?

miguelriemoliveira, 2023-04-27:
> anyway, my coordinate system is different than yours

Actually, when I compare this image of yours [with mine](https://github.com/lardemua/atom/issues/564#issuecomment-1520048499) I think they are the same no?

danifpdra, 2023-04-27:
No, the red axis is pointing to the screen in mine

miguelriemoliveira, 2023-04-28:
Right, I was looking at the z axis only, because the x axis can change when you move the pattern.

In any case, can we talk. I am not sure today is a good day for you ... if not, no problem.

danifpdra, 2023-05-03:
> Right, I was looking at the z axis only, because the x axis can change when you move the pattern.
> 
> In any case, can we talk. I am not sure today is a good day for you ... if not, no problem.

Sorry, I forgot to answer you. we can talk, yes

danifpdra, 2023-05-03:
Anyway, in your picture, if you compare the pattern's axis with the other axis in the picture, the red and green are switched and they don't match the other axis.

miguelriemoliveira, 2023-10-09:
Duplicate of #630. Closing.

closed 2023-10-09 by miguelriemoliveira