lardemua/atom #569: Installing Zaubot breaks other robots

miguelriemoliveira, 2023-04-23:
Hi @JorgeFernandes-Git ,

the problem is that your zaubot repo has the following packages:

```
➜  zau_bot git:(Zau_INESCTEC_eye_on_hand_calibration) ls UR5 
universal_robot      ur5_moveit_config  ur_e_description  ur_msgs
universal_robots     ur_bringup         ur_e_gazebo
ur5_e_moveit_config  ur_driver          ur_kinematics
```

and this will brake any other packages that will also need the universal robots, such as the mmtbot or the larcc.

The idea is not to include these packages in our repo, but rather to add to the readme what needs to be installed.

Take a look at the [mmtbot example](https://github.com/miguelriemoliveira/mmtbot#installation) or the [larcc example](https://github.com/miguelriemoliveira/mmtbot#installation).

Would it be possible to do it like I suggest?

JorgeFernandes-Git, 2023-05-19:
Hi @miguelriemoliveira.

Yes, I'll correct this. When I started the Zau project, I didn't know about this problem.

I've also an issue on Zau repo for this same porpose.

miguelriemoliveira, 2023-05-19:
OK, so I will close this one as a duplicate.

miguelriemoliveira, 2023-05-19:
Can you point here to the zaubot issue?

closed 2023-05-19 by miguelriemoliveira

JorgeFernandes-Git, 2023-05-19:
Here: https://github.com/JorgeFernandes-Git/zau_bot/issues/20