lardemua/atom #572: Problems with naming tfs with "/" symbols?

miguelriemoliveira, 2023-04-30:
Following JorgeFernandes-Git/zau_bot#19 I was trying to configure the calibration package:

    rosrun atom_calibration configure_calibration_pkg -n zau_eye_on_hand_calibration -utf

but I get this error:

>   File "/home/mike/workspaces/catkin_ws/devel/lib/atom_calibration/configure_calibration_pkg", line 15, in <module>
>     exec(compile(fh.read(), python_script, 'exec'), context)
>   File "/home/mike/workspaces/catkin_ws/src/calibration/atom/atom_calibration/scripts/configure_calibration_pkg", line 345, in <module>
>     paths = nx.shortest_path(gx, config['world_link'], sensor['link'])  # path between root and sensor data frame
>   File "/usr/local/lib/python3.8/dist-packages/networkx/classes/backends.py", line 145, in wrapper
>     return func(*args, **kwds)
>   File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/generic.py", line 167, in shortest_path
>     paths = nx.bidirectional_shortest_path(G, source, target)
>   File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/unweighted.py", line 224, in bidirectional_shortest_path
>     results = _bidirectional_pred_succ(G, source, target)
>   File "/usr/local/lib/python3.8/dist-packages/networkx/algorithms/shortest_paths/unweighted.py", line 292, in _bidirectional_pred_succ
>     raise nx.NetworkXNoPath(f"No path between {source} and {target}.")
> networkx.exception.NetworkXNoPath: No path between base_link and cam_2_color_optical_frame.

I suspect that this is because the zaubot coordinate frames have "/" symbols ...

JorgeFernandes-Git, 2023-05-01:
Hi @miguelriemoliveira.

Try with this config file:

https://github.com/JorgeFernandes-Git/zau_bot/blob/Zau_INESCTEC/Zau_INESCTEC/zau_full_calibration/calibration/config.yml

```
rosrun atom_calibration configure_calibration_pkg -n zau_full_calibration -utf
```

miguelriemoliveira, 2023-05-01:
Right, thanks. 

I will try later today.

miguelriemoliveira, 2023-05-14:
The problem should be a bad config file. Closing.

closed 2023-05-14 by miguelriemoliveira