lardemua/atom #575: Interactive markers not showing on set initial estimate and collect data

Joao-G-Martins, 2023-05-08:
Hello @miguelriemoliveira,
I am trying to run the set initial estimate and collect data launch files but interactive markers are not showing on rviz.
I tried to install interactive markers from apt but it was already installed. 

In this zip file you have my robot configuration and the recorded bag. The bag was stored under /home/Bags.

https://drive.google.com/drive/folders/12p0VkURPndXgEUNo8EyFFOZs_bIs25oj?usp=sharing

Because RVIZ was taking too long to load due to the heavy mesh file, I commented on drone.urdf the visual of the robot model.

Can you please run the collect data launch file to help understand the issue?

Thank you!

miguelriemoliveira, 2023-05-08:
Hi @Joao-G-Martins ,

I will try to help. Can you share a smaller bagfile? 10GB is a lot of stuff. It is enough to have a 30 secs or 1 min bag to test.

Also, don't you have a github repo for your code? It would be better for me to share the repository with you. This way if I change something you will be able to see what changed ...

Once I have more news I will let you know.

miguelriemoliveira, 2023-05-08:
Hi again @Joao-G-Martins ,

> Because RVIZ was taking too long to load due to the heavy mesh file, I commented on drone.urdf the visual of the robot model.

Why not create a simplified version of the mesh?

miguelriemoliveira, 2023-05-08:
Hi @Joao-G-Martins ,

I tried downloading the bag twice but end up with a network error. Tomorrow I will try from the University, but if you can provide a smaller bag that would be helpful.

Joao-G-Martins, 2023-05-08:
Hi @miguelriemoliveira,

Actually, the bag that I provided correspond to just 1 minute of data. The main reason is the point cloud from an ouster 64 and the camera.
As for the code, I will create a repository with it.

Joao-G-Martins, 2023-05-09:
Hi @miguelriemoliveira,

I created a simpler representation of the drone and put the code in the following repository: https://github.com/Joao-G-Martins/drone_calibration_atom

As for the bag, let me know if you still have problems with the download.

miguelriemoliveira, 2023-05-09:
> I created a simpler representation of the drone and put the code in the following repository: https://github.com/Joao-G-Martins/drone_calibration_atom

Thanks.

miguelriemoliveira, 2023-05-09:
Right, then we should use the throttle feature of ATOM to reduce cpu load. I will try to change the calibration.yaml accordingly.

I am still trying to download the bag :- (

miguelriemoliveira, 2023-05-10:
Hi @Joao-G-Martins ,

I got the bag and the repo, thanks. 

Will try to check what's happening and get back to you.

miguelriemoliveira, 2023-05-10:
Hi @Joao-G-Martins ,

first problem I noticed. You do not have camera_info messages for your camera. These should be in the bag ... Not sure if that's the problem, but it is certainly a problem to fix.

miguelriemoliveira, 2023-05-10:
I can confirm the same problem occurs in my computer. Rviz interactive markers do not appear.

miguelriemoliveira, 2023-05-10:
Hi @Joao-G-Martins ,

I also think that your configuration file does not make sense. Look at the calibration summary.pdf

[image]

Is this what you wanted to do?

miguelriemoliveira, 2023-05-10:
Hi @Joao-G-Martins ,

I think I found the problem. The code waits for a camera_info message which never comes and it stays waiting for the message forever. The interactive markers never get to be created, which is why you do not see them.

The set_initial_estimate code prints the following:

> Creating a new sensor named fpv_camera
> Waiting for camera_info message on topic /dji_osdk_ros/camera_info ...2

I think ATOM should've let you know your bag file was invalid when you configured the calibration package. Created #576 for this.

Joao-G-Martins, 2023-05-10:
Hi @miguelriemoliveira,

On the configuration file, I am interested in the transform between the LiDAR and the camera. Maybe I can introduce a world link to make it more clear. 

As for the camera images, the topic from the SDK only publishes sensor_msgs/Image and not the camera_info. I will double check tomorrow if I can get also the camera_info. If not, I need to publish one camera_info topic with the camera calibration and it should resolve the problem?

miguelriemoliveira, 2023-05-10:
Hi @Joao-G-Martins ,

I implemented #576. Now with the bagfile you gave me the command:

    rosrun drone_lidar_front configure

outputs the following error:

> ValueError:  Topic /dji_osdk_ros/camera_info (from sensor fpv_camera) does not exist in the bag file. This topic must exist. Aborting configuration!

miguelriemoliveira, 2023-05-11:
> On the configuration file, I am interested in the transform between the LiDAR and the camera. Maybe I can introduce a world link to make it more clear.

The classic tf tree topology would be:

world->lidar
world->camera

you have world->camera->lidar, which does not make sense, I think.

> As for the camera images, the topic from the SDK only publishes sensor_msgs/Image and not the camera_info. I will double check tomorrow if I can get also the camera_info. If not, I need to publish one camera_info topic with the camera calibration and it should resolve the problem?

I think you have to do a [monocular camera intrinsic calibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) first, and then the topic should be published.

Which camera drivers are you using?

Joao-G-Martins, 2023-05-11:
> I think you have to do a [monocular camera intrinsic calibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) first, and then the topic should be published.
> 
> Which camera drivers are you using?

I am using [DJI OSDK](https://github.com/dji-sdk/Onboard-SDK-ROS) that already publishes the sensor_msgs/Image messages from the camera driver. However, I have not seen any camera_info topic (I have to double check) being published. 

To solve the original problem, I will publish the camera_info from the intrinsic calibration .yaml file. 
I let you know if the problem persists!

miguelriemoliveira, 2023-05-12:
Hi @Joao-G-Martins ,

do you have any news about this? Can I close the issue?

Joao-G-Martins, 2023-05-12:
Hi @miguelriemoliveira,

Sorry for the delay. Yes, I can confirm that now the interactive markers are showing up, it was the camera_info problem. You can close the issue.

Thank you so much for your time and help!

miguelriemoliveira, 2023-05-12:
Great. No problem.

closed 2023-05-12 by miguelriemoliveira