lardemua/atom #584: Create a script to change a transformation from dynamic to static in a bagfile

miguelriemoliveira, 2023-05-21:
@JorgeFernandes-Git your should be aware of this.

Because of #578  we decided to add a verification on the calibration package configuration to check if the transformations defined to be optimized are static (#581).

If now, the configuration fails. we cannot calibrate the system.

One solution would be to "convert"  a dynamic transformation in a bagfile into a static one.

miguelriemoliveira, 2023-05-21:
Implemented and tested. Seems to be working fine. @JorgeFernandes-Git , can you test to see if it solves the [problem of the zau_full_calibration bagfile](#578)?

JorgeFernandes-Git, 2023-05-22:
Hi @miguelriemoliveira.

I'm testing this:

I run a test for 10 seconds:
```
rosrun atom_calibration dynamic_to_static_tf_rosbag -bfi zau_full_calibration_train_bag.bag -bfo test.bag -p zau/base_link -c base_link
```

Got the message:
```
Converting bagfile. Please wait.
Processing bag messages:   1%|██                                                                                                                                          | 5997/411400 [00:00<00:34, 11746.35it/s]
Finished in 2.61 seconds.
Created a /tf_static of transform zau/base_link to base_link
Removed 99 transformations from bag.
```

But the test bag come like this:
[image]

The tree is disconnected.

miguelriemoliveira, 2023-05-22:
Hi @JorgeFernandes-Git ,

Hum, I have to look into it...

miguelriemoliveira, 2023-05-22:
Hi @JorgeFernandes-Git ,

I just run for the entire bag and the results are ok:

[image]

Are you sure you updated before running? You seem to have a different version~.
What do you mean run a test for 10 secs? That was a debug version. Now it should process the entire bagfile.

miguelriemoliveira, 2023-05-22:
> Can you send me?

Run it again in your bag file. Its much easier no?

miguelriemoliveira, 2023-05-22:
This is not working. @JorgeFernandes-Git , let me know when you have a simpler bag file for testing.

JorgeFernandes-Git, 2023-05-22:
I create a simple bag to test this script where:
* static_child is published 1 time on /tf_static;
* dynamic_child is published on /tf at a rate of 10 Hz.

[image]

Bag info:

```
path:        fake_bag.bag
version:     2.0
duration:    27.4s
start:       May 22 2023 19:20:18.93 (1684779618.93)
end:         May 22 2023 19:20:46.37 (1684779646.37)
size:        58.3 KB
messages:    275
compression: none [1/1 chunks]
types:       rosgraph_msgs/Log  [acffd30cd6b6de30f120938c17c593fb]
             tf2_msgs/TFMessage [94810edda583a504dfda3829e70d7eec]
topics:      /rosout        2 msgs    : rosgraph_msgs/Log 
             /tf          272 msgs    : tf2_msgs/TFMessage
             /tf_static     1 msg     : tf2_msgs/TFMessage
```

After run the script like this:

```
rosrun atom_calibration dynamic_to_static_tf_rosbag -bfi fake_bag.bag -bfo mod.bag -p parent -c dynamic_child
```

I got this result: 

```
path:        mod.bag
version:     2.0
duration:    0.3s
start:       May 22 2023 19:20:18.93 (1684779618.93)
end:         May 22 2023 19:20:19.27 (1684779619.27)
size:        12.0 KB
messages:    4
compression: none [1/1 chunks]
types:       rosgraph_msgs/Log  [acffd30cd6b6de30f120938c17c593fb]
             tf2_msgs/TFMessage [94810edda583a504dfda3829e70d7eec]
topics:      /rosout      2 msgs    : rosgraph_msgs/Log 
             /tf_static   2 msgs    : tf2_msgs/TFMessage
```

With this tf tree:

[image]

**Conclusion: Seems the result is what we expected.**

JorgeFernandes-Git, 2023-05-22:
The bags:

[Original](https://drive.google.com/file/d/1Atz4DxusPhx6qNVRIToOjfpRGerYJYl-/view?usp=sharing)
And:
[Modified](https://drive.google.com/file/d/13VniP05sW3z0xiO3URwHd6h8VQjeeH7q/view?usp=share_link)

JorgeFernandes-Git, 2023-05-23:
@miguelriemoliveira.

I also made this script to try to remove tfs from the bag:

https://github.com/JorgeFernandes-Git/zau_bot/blob/Zau_INESCTEC/zau_bot_bringup/scripts/delete_tfs_in_bagfile.py

I'm trying other ideas.

JorgeFernandes-Git, 2023-05-23:
After reading more about this, I realize that the /tf_static should have one message, with all the static tfs.

"tf2_ros provides the [StaticTransformBroadcaster](http://docs.ros.org/jade/api/tf2_ros/html/c++/classtf2__ros_1_1StaticTransformBroadcaster.html) class that manages a vector of transforms, (...) calling sendTransform automatically adds the transforms you pass and publishes everything in a single message."

After inspect with `rosbag info` other good bags I have, I notice that all of them have one message.

JorgeFernandes-Git, 2023-05-23:
Hi @miguelriemoliveira.

After running some tests I discovered that the real problem is on the script.

Here is an example:
Before running the scipt:
[image]

After running:
```
rosrun atom_calibration dynamic_to_static_tf_rosbag -bfi e3_rgb2agv.bag -bfo test.bag -p base_link_mb -c wheel_left_link
```
Here is the tf_tree:
[image]

All the static tfs are erased... I try some ideas, but no futher sucess.

JorgeFernandes-Git, 2023-05-23:
I repeat the previous experiment but with 2 dynamic frames:

[image]

Run the same command:
```
rosrun atom_calibration dynamic_to_static_tf_rosbag -bfi new_fake_bag.bag -bfo new_test.bag -p parent -c dynamic_child
```

Result:
[image]

The tf_static are erased again.

JorgeFernandes-Git, 2023-05-23:
Running on the test.bag:

```
rqt_bag new_test.bag
```

[image]

The tf_static are there ...... But don't appear on the tf_tree.

[New_fake_bag.bag](https://drive.google.com/file/d/1lCWJSYExbEXB4DghQn1tGHaXDB7MHY-3/view?usp=sharing) - Original
[New_test.bag](https://drive.google.com/file/d/1kDvQV71_IEqbgQijyaRw9qhuFrUa8I8k/view?usp=sharing) - Created by the script

miguelriemoliveira, 2023-05-24:
We now have lots of experiments. I have created a repository for rosbag tools [here](https://github.com/miguelriemoliveira/rosbag_tools).

miguelriemoliveira, 2023-05-25:
Hi @JorgeFernandes-Git ,

finally, this is solved. Check https://github.com/miguelriemoliveira/rosbag_tools/commit/7dfc7dd95faa92a116963624526f6654a081da6c

As I said, I created a new repository named [rosbag_tools](https://github.com/miguelriemoliveira/rosbag_tools) for collecting several rosbag processing scripts.

In the particular case of **zau_full_calibration_train_bag.bag** #578, to repair you should run:

1. Aggregate all msgs on tf_static topic into a single one:

    ./aggregate_tf_static_msgs -bfi ~/bagfiles/zaubot/zau_full_calibration_train_bag.bag

2. Move the desired transform from the tf to the tf_static topic:

    ./dynamic_to_static_tf -bfi ~/bagfiles/zaubot/aggregated.bag -p zau/base_link -c base_link

After this the bag runs smooth. You can play the bag and only later run rqt and you'll see the complete tree without disconnections.

PR #585 (2023-05-21): Fixes #584 Implemented and functional.
Fixes #584
commit 471278ba (2023-05-21): Merge pull request #585 from lardemua:miguelriemoliveira/issue584

Fixes #584 Implemented and functional.
commit 71d8b9c2 (2023-05-21): Fixes #584 Implemented and functional.
closed 2023-05-25 by miguelriemoliveira