lardemua/atom #589: Handling additional_tfs better

miguelriemoliveira, 2023-05-28:
https://github.com/lardemua/atom/blob/3d9f04dcb8d208519768c5648e40db4ba46aa592/atom_core/src/atom_core/config_io.py#L136

manuelgitgomes, 2023-11-23:
This might be also used for the `package_name`.

Do you  have any idea on how this should be used?

manuelgitgomes, 2023-11-27:
@miguelriemoliveira, please answer this so I can close #682

miguelriemoliveira, 2023-11-27:
Hi @manuelgitgomes ,

basically, that's for defining which fields in the config.yml should be mandatory or optional.

If the package_name must be mandatory (which I think it should), then you should also include it in the check.

Is this what you needed? I am a bit lost ...

manuelgitgomes, 2023-11-27:
> Is this what you needed? I am a bit lost ...

The thing is that this would require every one to reconfigure ATOM.
But, as stated in #710, for a lot of reasons, they would need reconfiguring nevertheless, so this is a non-issue, right?