lardemua/atom #593: Calibration of real softbot

miguelriemoliveira, 2023-07-16:
This issue is to track the work of calibrating the real softbot. @manuelgitgomes is also working on this.

@manuelgitgomes do you have Bruno's git account? We could add him also here to keep him posted.

@manuelgitgomes described running through the procedure of calibrating the softbot using the bagfile taken by Eurico.

https://drive.google.com/file/d/1YlU2PxqEYwShidhZ0wja6Jp_b2rA-opW/view?usp=drive_web

Then we created a dataset (let's have better names for the datasets, e.g. softbot_real_train1, etc).

[softbot_20230712.tar.gz](https://github.com/lardemua/atom/files/12063539/softbot_20230712.tar.gz)

and the calibration does not work well for the lidar. I am investigating ...

miguelriemoliveira, 2023-07-16:
First test I did was to run the dataset on the dataset_playback toll to confirm that all lidar labels are correct. I confirm, could not see any incorrect annotations.

miguelriemoliveira, 2023-07-16:
Calibration with a single sensor, I get this:

```
+------------+--------------+--------------+-----------+
| Collection | camera0 (px) | camera1 (px) | lidar (m) |
+------------+--------------+--------------+-----------+
|    000     |    0.4606    |    0.4985    |   0.0314  |
|  Averages  |    0.4606    |    0.4985    |   0.0314  |
+------------+--------------+--------------+-----------+
```

It is not bad, but lidar errors are 3 centimeters. And we can see them in the image:

[image]

miguelriemoliveira, 2023-07-16:
So I changed the [normalizer](https://github.com/lardemua/atom/blob/885d8c9f704e7885f2e8ee47d87105b2bf18ee0d/atom_calibration/scripts/calibrate#L425C5-L441C63) for the lidar3d , by adding the line:

    normalizer['lidar3d'] = normalizer['lidar3d']*0.01

and got the following results:

```
+------------+--------------+--------------+-----------+
| Collection | camera0 (px) | camera1 (px) | lidar (m) |
+------------+--------------+--------------+-----------+
|    000     |    0.3550    |    0.3766    |   0.0094  |
|  Averages  |    0.3550    |    0.3766    |   0.0094  |
+------------+--------------+--------------+-----------+
```

now we get under one centimeter. Curiously, also the rgb errors went down, even though their weights are smaller.
Visualizy, it also looks better.

[image]

miguelriemoliveira, 2023-07-16:
I noticed there are some problems with collections 4 and 5. Optimizing without the, gives:

```
+------------+--------------+--------------+-----------+
| Collection | camera0 (px) | camera1 (px) | lidar (m) |
+------------+--------------+--------------+-----------+
|    000     |    4.0438    |    2.0003    |   0.0196  |
|    001     |    2.8612    |    1.9661    |   0.0137  |
|    002     |    1.7982    |    1.8422    |   0.0106  |
|    003     |    3.2156    |    2.8287    |   0.0123  |
|    006     |    2.0828    |    2.4366    |   0.0130  |
|    007     |    3.1220    |    3.7569    |   0.0145  |
|    008     |    4.3712    |    2.2515    |   0.0141  |
|  Averages  |    3.0707    |    2.4403    |   0.0140  |
+------------+--------------+--------------+-----------+
```

Which is not bad...

miguelriemoliveira, 2023-07-16:
**Got it**!!!

Look at the labels in collection 4:

[image]

From above:

[image]

My suggestion is that you follow the same procedure to figure out which collections still have incorrect labels and fix them.

Send me the corrected dataset back please. Perhaps we can name it **mysoftbot_real_train1** ...

manuelgitgomes, 2023-07-17:
Hello! After correcting the dataset and removing the outliers, the results is as follows:
```
+------------+--------------+--------------+-----------+
| Collection | camera0 (px) | camera1 (px) | lidar (m) |
+------------+--------------+--------------+-----------+
|    000     |    2.4997    |    1.4875    |   0.0188  |
|    001     |    3.3260    |    1.4954    |   0.0166  |
|    002     |    2.2202    |    1.7408    |   0.0160  |
|    003     |    1.5199    |    2.9828    |   0.0178  |
|    004     |    5.7187    |    5.0652    |   0.0182  |
|    005     |    4.3242    |    3.0555    |   0.0166  |
|    006     |    3.1556    |    2.5532    |   0.0145  |
|    007     |    4.7572    |    3.1732    |   0.0154  |
|    008     |    4.0306    |    2.4069    |   0.0154  |
|    009     |    3.7246    |    2.3266    |   0.0128  |
|    010     |    1.9670    |    2.1183    |   0.0151  |
|    011     |    2.5314    |    1.7015    |   0.0127  |
|    012     |    2.0042    |    1.4680    |   0.0125  |
|    013     |    6.3597    |    1.9768    |   0.0130  |
|    015     |    3.5422    |    2.3883    |   0.0188  |
|    016     |    2.7318    |    1.9155    |   0.0123  |
|    017     |    2.0943    |    1.2860    |   0.0125  |
|    018     |    2.1871    |    2.1843    |   0.0126  |
|    019     |    2.9071    |    1.5126    |   0.0141  |
|    020     |    3.4967    |    1.8192    |   0.0156  |
|    021     |    2.4659    |    3.3485    |   0.0148  |
|  Averages  |    3.2173    |    2.2860    |   0.0151  |
+------------+--------------+--------------+-----------+
```

manuelgitgomes, 2023-07-17:
Hello!

All the results are here, in this compressed file!

[real_softbot_results.tar.gz](https://github.com/lardemua/atom/files/12070314/real_softbot_results.tar.gz)

eupedrosa, 2023-07-22:
Hi there. This is the calibration with an anchored sensor, not relative to the robot's frame, correct?

miguelriemoliveira, 2023-12-03:
Closing. Quite old.

closed 2023-12-03 by miguelriemoliveira