lardemua/atom #595: Generalization of evaluation scripts

manuelgitgomes, 2023-07-17:
Evaluation scripts and not generalized, especially the save file results flag.

As an example, imagine you have 3 cameras in a robotic system, camera0, camera1, camera2.

To evaluate the transformation between camera0 and camera1, the results will be saved as `rgb_to_rgb_evaluation.csv`.
To evaluate the transformation between camera0 and camera2, the results will also be saved as `rgb_to_rgb_evaluation.csv`.

I purpose to save it as `camera0_to_camera1_evaluation.csv` and  `camera0_to_camera2_evaluation.csv`.

manuelgitgomes, 2023-07-17:
I have changed this only for lidar and rgb evaluations, as depth evaluations are in a older format and I am currently unable to test it.

miguelriemoliveira, 2023-07-17:
Thanks. Makes sense.

manuelgitgomes, 2023-07-27:
I have an addition to this.

When carrying out calibration with other approaches, such as OpenCV, the results of calibration will be saved with the same name, overwriting the existing results.

I propose two solutions for this:

- The simpler one is to read the train dataset name, such as `atom_calibration.json` or `cv_calibration.json` and carrying out this name to the results. The results will therefore have names such as `atom_calibration_camera0_to_camera1_evaluation.csv`
- The more complex one, albeit more versatile, is to add a new key-value pair to the json file, either on `_metadata` or `calibration_config`. This pair would be of syntax: `calibration_approach: atom`. The results will therefore have names such as `atom_camera0_to_camera1_evaluation.csv`

I will start working on the more complex on a new branch, as I believe it to be more clean.
If you (@miguelriemoliveira) prefer the simpler one, it can be easily implemented.

miguelriemoliveira, 2023-07-27:
Hi Manuel,

The problem with the second is that it deprecates datasets, as the old ones
would not have this key.

Why not:

3. if the file exists , ask the user if it should be replaced
4. Force the user to give the name of the output results file?

manuelgitgomes, 2023-07-27:
We can make a fusion of 1, 3 and 4,

It defaults to copying the name of the calibrated dataset.
The user has the ability to rename the dataset.
If it already exists, it asks the user if it should be replaced.

What do you think?

miguelriemoliveira, 2023-07-27:
Great

manuelgitgomes, 2023-07-27:
Done! Merging in #596!

closed 2023-09-25 by manuelgitgomes