lardemua/atom #597: Lidar Labeler Interactive Marker frame_id

manuelgitgomes, 2023-09-07:
Hello @miguelriemoliveira!

I want to know if there is any reason why the interactive marker used for labeling lidar data has the frame_id of the sensor (i.e., the lidar)?

This question came to me because in mobile robots, when the robot moves, the marker moves as well, getting away from the pattern.
This is cumbersome, as the user needs to move the marker every time the robot moves.

If the frame_id was set to the world frame, the marker would remain static, easing the calibration process.
In cases where the robot is static, everything would stay the same, because both the sensor and the world link are static.

miguelriemoliveira, 2023-09-07:
Hi @manuelgitgomes ,

I think the reason is to facilitate the code. If we get the pose of the marker already in the sensor frame, we need not do any transformations on top of that.

I think its what makes more sense.

Not sure I understand you example. The interactive marker of the lidar should be used only to set the intitial guess, then it carries on with the tracking, so why does it bother? 

I am not sure about all the implications of this, so I would preffer not to change or to do it using an optional flag.

manuelgitgomes, 2023-11-29:
Closed because it is not needed.

closed 2023-11-29 by manuelgitgomes