lardemua/atom #600: Problem understanding the normalization function

manuelgitgomes, 2023-09-15:
Hello @miguelriemoliveira!

I am having some problems in calibration and I couldn't understand the normalization function:

https://github.com/lardemua/atom/blob/8a32d459675bc009ad665f8a7d52c0374c26d310/atom_calibration/scripts/calibrate#L441

The normalization function described in the papers is the mean error.
So why do we apply the absolute logarithm?
The logarithm might make some sense to attenuate differences between different ranges of values.
The absolute, however, does not make sense to me.
I understand that the normalization factor should not be negative, but applying an absolute operation breaks the monotony of the function. 
This defeats the purpose of the normalization.
If the log is to be kept, isn't it better to be:

```python3
 normalizer[modality] = np.log(1 + np.mean(values)) 
```
This achieves the elimination of negative values without breaking the monotony of the function.

miguelriemoliveira, 2023-09-15:
Hi @manuelgitgomes ,

I agree with you. Does not seem to make sense. Can you adapt it and see if it works?

miguelriemoliveira, 2023-10-09:
You fixed it, right? Can we close?

closed 2023-10-09 by manuelgitgomes