lardemua/atom #601: Calibration -Give a warning if the anchored sensor is not present in the dataset

manuelgitgomes, 2023-09-19:
As it is, if the user changes the dataset to have an invalid anchored sensor (as I did), the calibration carries on without a problem, removing the anchor and not giving any warning to the user.
The code is as follows:

https://github.com/lardemua/atom/blob/8a32d459675bc009ad665f8a7d52c0374c26d310/atom_calibration/scripts/calibrate#L179-L193

I suggest to change this, so that when the defined anchored sensor is not on the dataset, a warning must appear.
This warning must contain something like "Anchored sensor not found. Do you want to continue the calibration without one?"

What do you think @miguelriemoliveira?

miguelriemoliveira, 2023-09-20:
Hi @manuelgitgomes ,

thanks for finding out yet another bug in ATOM : - ).

My feeling is that if we have an incorrect configuration the calibration should not be carried out.
So I would preffer to throw an error that aborts the calibration, instead of a warning. Can you implement it please?

Actually, this is already done in here:

https://github.com/lardemua/atom/blob/8a32d459675bc009ad665f8a7d52c0374c26d310/atom_calibration/scripts/configure_calibration_pkg#L199-L203

but probably you want to add it in the calibrate script?

manuelgitgomes, 2023-09-20:
Hello @miguelriemoliveira!

I created the function `verifyAnchoredSensor`:
https://github.com/lardemua/atom/blob/8dad0c3f63c039245b4e8f036424fea6c289f7f3/atom_core/src/atom_core/utilities.py#L174-L179

It is now implemented, as you can see, both on calibrate and on configure:

[image]
[image]

closed 2023-09-20 by manuelgitgomes