lardemua/atom #603: Calibration of atom_test_rgb_camera also with depth

miguelriemoliveira, 2023-09-20:
Hi @manuelgitgomes ,

I was working with @danifpdra on the calibration of the  atom_test_rgb_camera also with depth.

We created a small bagfile , then recorded a single collection dataset, verified that the labels were correct, and finally calibrated.

Results during calibration were:

```
Optimization finished in 1.41617 secs: `ftol` termination condition is satisfied.
Errors per collection (anchored sensor,  max error per sensor, not detected as "---")
+------------+-------------------+-----------------+
| Collection | depth_camera (px) | rgb_camera (px) |
+------------+-------------------+-----------------+
|    000     |       0.0031      |      0.2054     |
|  Averages  |       0.0031      |      0.2054     |
+------------+-------------------+-----------------+
```

Then we went on to do an evaluation. Results from the evaluation were:

```
Starting evalutation...
Errors per collection
+--------------+-----------+-------------+-------------+
| Collection # | RMS (pix) | X err (pix) | Y err (pix) |
+--------------+-----------+-------------+-------------+
|     000      |   4.6799  |    2.0401   |    2.4686   |
|   Averages   |   4.6799  |    2.0401   |    2.4686   |
+--------------+-----------+-------------+-------------+
```

There is a large difference from the 0.2 in the calibrate script, with the 4.6 in the case of the evaluation.

@manuelgitgomes you told me you calibrated the depth as well. Did you run into this kind of disparity?

Perhaps we can talk tomorrow afternoon about this.

miguelriemoliveira, 2023-09-20:
We forgot to run the ground truth calibration. That's the one that will make things clearer. Tomorrow we can do it.

manuelgitgomes, 2023-09-21:
Hello @miguelriemoliveira.

The results I got were:

```bash
+--------------------------------+--------+--------+--------+------------+-------------+-----------+------------+-----------+
|        Transformation #        | X (mm) | Y (mm) | Z (mm) | Roll (deg) | Pitch (deg) | Yaw (deg) | Trans (mm) | Rot (deg) |
+--------------------------------+--------+--------+--------+------------+-------------+-----------+------------+-----------+
|  camera_link-camera_rgb_frame  | 3.4506 | 0.0137 | 2.6260 |   0.1125   |    0.0238   |   0.0062  |   4.3362   |   0.1152  |
| camera_link-camera_depth_frame | 0.0000 | 0.0000 | 0.0000 |   0.0000   |    0.0000   |   0.0000  |   0.0000   |   0.0000  |
+--------------------------------+--------+--------+--------+------------+-------------+-----------+------------+-----------+
```

With initial residuals:

```
+------------+-------------+-------------------+
| Collection | camera (px) | depth_camera (px) |
+------------+-------------+-------------------+
|    000     |    0.0204   |       0.0093      |
|  Averages  |    0.0204   |       0.0093      |
+------------+-------------+-------------------+

```

And end residuals:

```bash
+------------+-------------+-------------------+
| Collection | camera (px) | depth_camera (px) |
+------------+-------------+-------------------+
|    000     |    0.0195   |       0.0079      |
|  Averages  |    0.0195   |       0.0079      |
+------------+-------------+-------------------+

```

And even changing the normalizer (#600) to the mean, I achieved very similar results.

> Then we went on to do an evaluation. Results from the evaluation were:

Did you use another dataset to evaluate it?
And what evaluation did you ran?

miguelriemoliveira, 2023-09-21:
Hi @manuelgitgomes ,

we first did a depth to rgb evaluation, but then remembered that the ground truth evaluation makes much more sense to debug.

miguelriemoliveira, 2023-09-21:
@danifpdra I will continue from yesterday, if you want we can zoom...

danifpdra, 2023-09-21:
> @danifpdra I will continue from yesterday, if you want we can zoom...

Yes, sure. I will connect

miguelriemoliveira, 2023-09-21:
Hi @manuelgitgomes ,

I talked now with @danifpdra and she was able to calibrate the depth with very good accuracy. I think the depth is working fine. Perhaps we can abandon this?

manuelgitgomes, 2023-09-21:
Hello @miguelriemoliveira and @danifpdra,

Really? How were you able to do it?
How accurate?

Should I drop my problem with the residuals or should I continue?

miguelriemoliveira, 2023-10-09:
Deprecated. I will continue with rgb_depth_system in atom_examples.

closed 2023-10-09 by miguelriemoliveira