lardemua/atom #605: Adding normalizer to calibration results table

manuelgitgomes, 2023-09-21:
Hello @miguelriemoliveira!

I'm having some ideas to help us debug our problems.
I believe having the normalizer value on the calibration results table should be helpful.
I have already implemented this, if you agree I will commit this:

[image]

I also think having an option (maybe an argument) so that the tables show the normalized values might be helpful, to guarantee it is converging. 
What do you think?

manuelgitgomes, 2023-09-21:
Hello! Consider it done.
For the sake of conciseness, the normalizer values as presented at the beginning of the calibration, as:
```bash
Normalizer for lidar3d: 0.005443645289418344
Normalizer for depth: 0.08864145776057339
Normalizer for rgb: 125.75252938851234

```

The calibration now has a new argument: `-snv`/`-show_normalized_values`.
When place, this argument outputs a table with the normalized values inside parenthesis as:

```bash
+------------+-----------------------+----------------------+-----------------------+----------------------+-----------------+
| Collection | body_depth_camera [m] | body_rgb_camera [px] | hand_depth_camera [m] | hand_rgb_camera [px] |   lidar3d [m]   |
+------------+-----------------------+----------------------+-----------------------+----------------------+-----------------+
|    000     |    0.1634 (1.8428)    |  175.4700 (1.3954)   |    0.0382 (0.4304)    |   89.4515 (0.7113)   | 0.0059 (1.0797) |
|    001     |    0.1634 (1.8428)    |  175.4713 (1.3954)   |    0.0372 (0.4192)    |  100.4001 (0.7984)   | 0.0057 (1.0409) |
|    002     |    0.1003 (1.1313)    |  152.7633 (1.2148)   |    0.0446 (0.5026)    |   96.7739 (0.7696)   | 0.0058 (1.0723) |
|    003     |    0.1198 (1.3514)    |  200.7984 (1.5968)   |    0.0381 (0.4302)    |   98.9481 (0.7868)   | 0.0060 (1.0938) |
|    004     |    0.1192 (1.3444)    |  200.7691 (1.5965)   |    0.0254 (0.2863)    |   83.7184 (0.6657)   | 0.0059 (1.0845) |
|    005     |    0.0986 (1.1122)    |  153.7866 (1.2229)   |    0.0307 (0.3460)    |         ---          | 0.0050 (0.9159) |
|    006     |    0.0699 (0.7883)    |  155.7162 (1.2383)   |          ---          |         ---          | 0.0052 (0.9639) |
|    007     |    0.0685 (0.7726)    |  154.5595 (1.2291)   |    0.0739 (0.8343)    |  102.3256 (0.8137)   | 0.0052 (0.9627) |
|    009     |    0.1671 (1.8850)    |  145.5862 (1.1577)   |    0.0520 (0.5863)    |   89.8143 (0.7142)   | 0.0041 (0.7604) |
|    010     |    0.1674 (1.8880)    |  145.3446 (1.1558)   |    0.0482 (0.5438)    |   83.8461 (0.6668)   | 0.0041 (0.7521) |
|    011     |    0.1896 (2.1391)    |  132.3260 (1.0523)   |    0.0646 (0.7292)    |   67.5982 (0.5375)   | 0.0042 (0.7631) |
|    012     |    0.1896 (2.1395)    |  132.5620 (1.0541)   |    0.0884 (0.9970)    |   93.3939 (0.7427)   | 0.0042 (0.7667) |
|    013     |    0.1612 (1.8190)    |  126.7010 (1.0075)   |    0.0935 (1.0549)    |   99.4695 (0.7910)   | 0.0056 (1.0325) |
|    014     |    0.2198 (2.4793)    |  152.5769 (1.2133)   |    0.0775 (0.8740)    |   93.2230 (0.7413)   | 0.0051 (0.9382) |
|    015     |    0.2214 (2.4981)    |  154.1252 (1.2256)   |    0.0580 (0.6543)    |   67.9981 (0.5407)   | 0.0053 (0.9825) |
|    016     |    0.1727 (1.9482)    |  131.5035 (1.0457)   |    0.0634 (0.7151)    |   75.6876 (0.6019)   | 0.0060 (1.1039) |
|    017     |    0.2233 (2.5192)    |  176.5510 (1.4040)   |    0.0445 (0.5018)    |   70.9808 (0.5644)   | 0.0061 (1.1120) |
|    018     |    0.2232 (2.5183)    |  176.5477 (1.4039)   |    0.0340 (0.3834)    |   48.1485 (0.3829)   | 0.0060 (1.1045) |
|    019     |    0.1787 (2.0157)    |  144.6457 (1.1502)   |    0.0335 (0.3780)    |   44.2518 (0.3519)   | 0.0063 (1.1550) |
|    020     |    0.1929 (2.1766)    |  213.4122 (1.6971)   |    0.0321 (0.3622)    |   60.3669 (0.4800)   | 0.0060 (1.1007) |
|    021     |    0.1928 (2.1746)    |  213.4046 (1.6970)   |    0.0255 (0.2875)    |   52.4839 (0.4174)   | 0.0063 (1.1565) |
|    022     |    0.1542 (1.7398)    |  165.9539 (1.3197)   |    0.0177 (0.1992)    |   54.0995 (0.4302)   | 0.0059 (1.0837) |
|    023     |    0.1271 (1.4338)    |  138.3090 (1.0999)   |    0.0259 (0.2920)    |         ---          | 0.0061 (1.1157) |
|    024     |    0.1271 (1.4338)    |  138.3100 (1.0999)   |    0.0605 (0.6822)    |   74.2214 (0.5902)   | 0.0061 (1.1203) |
|    025     |    0.0815 (0.9192)    |  141.8142 (1.1277)   |    0.0635 (0.7166)    |   81.4853 (0.6480)   | 0.0051 (0.9334) |
|    026     |    0.1188 (1.3400)    |  144.0247 (1.1453)   |    0.0591 (0.6663)    |   74.6675 (0.5938)   | 0.0052 (0.9495) |
|    027     |    0.1188 (1.3401)    |  144.0305 (1.1453)   |    0.0672 (0.7578)    |   63.0668 (0.5015)   | 0.0054 (0.9936) |
|    028     |    0.0987 (1.1133)    |  139.8238 (1.1119)   |    0.0630 (0.7105)    |   65.2287 (0.5187)   | 0.0042 (0.7742) |
|    029     |    0.1202 (1.3563)    |  153.7668 (1.2228)   |    0.0672 (0.7576)    |   68.6959 (0.5463)   | 0.0040 (0.7334) |
|    030     |    0.1202 (1.3556)    |  153.4720 (1.2204)   |    0.0107 (0.1205)    |   28.2699 (0.2248)   | 0.0042 (0.7660) |
|    031     |    0.0930 (1.0492)    |  163.1447 (1.2973)   |    0.0136 (0.1531)    |   54.8790 (0.4364)   | 0.0041 (0.7528) |
|    032     |    0.0927 (1.0453)    |  162.9890 (1.2961)   |    0.0266 (0.2996)    |   79.7343 (0.6341)   | 0.0041 (0.7617) |
|    033     |    0.0706 (0.7961)    |  209.8323 (1.6686)   |    0.0348 (0.3924)    |  123.6921 (0.9836)   | 0.0056 (1.0248) |
|    034     |    0.0811 (0.9154)    |  189.6131 (1.5078)   |    0.0241 (0.2723)    |  113.5395 (0.9029)   | 0.0059 (1.0877) |
|    035     |    0.0811 (0.9151)    |  189.6453 (1.5081)   |    0.0344 (0.3880)    |  131.8905 (1.0488)   | 0.0053 (0.9764) |
|    036     |    0.1140 (1.2860)    |  161.5627 (1.2848)   |    0.0482 (0.5433)    |   90.5184 (0.7198)   | 0.0062 (1.1310) |
[... cut ...]
|    053     |    0.1403 (1.5827)    |  156.0247 (1.2407)   |    0.0436 (0.4916)    |  116.1903 (0.9240)   | 0.0048 (0.8744) |
|    054     |    0.1401 (1.5808)    |  155.8530 (1.2394)   |    0.0336 (0.3787)    |   80.4815 (0.6400)   | 0.0049 (0.9086) |
|  Averages  |    0.1375 (1.5507)    |  161.2431 (1.2822)   |    0.0478 (0.5391)    |   89.4421 (0.7113)   | 0.0054 (0.9869) |
+------------+-----------------------+----------------------+-----------------------+----------------------+-----------------+

```

closed 2023-09-21 by manuelgitgomes

miguelriemoliveira, 2023-09-23:
Thanks @manuelgitgomes . IT looks great!

miguelriemoliveira, 2023-09-23:
Hi @manuelgitgomes ,

I changed the documentation to include a screenshot of the new table:

https://lardemua.github.io/atom_documentation/procedures/#calibrate