lardemua/atom #611: Bad sensor assignement for normalizer

manuelgitgomes, 2023-09-21:
Hello @miguelriemoliveira!

I found the error, finally. (or at least one of them).

In the following line, the normalizer is calculated:

https://github.com/lardemua/atom/blob/6df1489bb7087e6057b3587a5a4740392830dce1/atom_calibration/scripts/calibrate#L441

Imagine I have two sensors. a depth and an rgb camera. Imagine they are named `depth_camera` and `camera`. 
When the normalizer of the rgb camera is being calculated, the residuals of the rgb and the depth camera are considered, as they both `camera` in their sensor key.

To solve this, a verification must be applied in the calibration configuration.

miguelriemoliveira, 2023-09-21:
Great work. Thanks

manuelgitgomes, 2023-10-09:
It is now @miguelriemoliveira

closed 2023-10-09 by manuelgitgomes