lardemua/atom #612: Improve manual labelling for depth sensors (simulation not working well)

danifpdra, 2023-09-21:


miguelriemoliveira, 2023-09-23:
Hi @danifpdra ,

can you describe the problem better?

miguelriemoliveira, 2023-10-12:
Steps to reproduce:

    roslaunch rgb_depth_system_calibration dataset_playback.launch

then 

   clear && rosrun atom_calibration dataset_playback -json $ATOM_DATASETS/rgb_depth_system/rgb_depth_system_example_train_dataset/dataset.json -ow

and we get this labeling of the first dataset:

[image]

miguelriemoliveira, 2023-10-12:
Tagging @brunofavs and @Kazadhum for visibility.

miguelriemoliveira, 2023-10-13:
So I went through the tutorial video for [correcting depth labels](https://lardemua.github.io/atom_documentation/multimedia/#correcting-depth-labels), and the manual annotation is supposed to create lines right at the border of the pattern.

Will try to anotate the polygon more carefuly ...

miguelriemoliveira, 2023-10-13:
So if I do a wrong annotation polygon, moving away from the pattern, See the red polygon:

[image]

with this polygon , I get this:

[image]

What would you want in this case @danifpdra ? Do you was that the polygon is well generated even with this bad annotarion of the polygon?

danifpdra, 2023-10-13:
> So if I do a wrong annotation polygon, moving away from the pattern, See the red polygon (its hard to see, zoom in):
> 
> [image]
> 
> with this polygon , I get this:
> 
> [image]
> 
> What would you want in this case @danifpdra ? Do you was that the polygon is well generated even with this bad annotarion of the polygon?
> 
> I think I could somehow filter out these very high jumps automatically. I am just not sure that would work in real data ...

No, that's not the problem. The problem is that the jump from the ground to the "sky" in the simulation is not well perceived by the labeling and many times it is very hard to label because you have to get the corners absolutely right or it will not be well labeled. the general problem is not the pattern itself, but the borders.

danifpdra, 2023-10-13:
Look at this example: 

[image]

miguelriemoliveira, 2023-10-13:
Thanks @danifpdra ,

I have an idea I will try to implement. But the part where you say this is a simulation only problem is what I do not understant. 

Can you give me steps to reproduce in real data?

danifpdra, 2023-10-13:
hi @miguelriemoliveira,

Just follow these steps: https://github.com/lardemua/atom/issues/645#issue-1941950885

miguelriemoliveira, 2023-10-15:
I am advancing (slowly). I have an idea and implemented it, to avoid getting those points very far away as boundaries.

I can create a mask which I give to function **labelDepthMsg**

https://github.com/lardemua/atom/blob/ab80d9aab2ddf5835e005f40df5c7db671c98623/atom_calibration/src/atom_calibration/dataset_playback/depth_manual_labeling.py#L76-L81

The problem now is that the function is returning boundary points outside that mask, which does not make sense ...

[image]

miguelriemoliveira, 2023-10-15:
Hi @danifpdra ,

I think I made some progress. The labeling is working much better now.

Here's an example. I defined this polygon, not really worrying about being inside the pattern. Look at the left side of the pattern / polygon.

[image]

And the result is:

[image]

Also for the boundaries I think it works. With this polygon:

[image]

we get this segmentation:

[image]

Notice how there are no boundary magenta points at the bottom. 
Another example, with this user defined polygon:

[image]

we get:

[image]

The downside is that this takes some 5 to 15 secs per segmentation. But in this case might as well wait and get a perfect result, that not wait and get unusable labels, you agree?

I only tested for simulation and using in the dataset playback. We need to test:

1. Simulation, during data collection, since there is a common function **labelDepthMsg** used in both
2. Test real data during data collection
3. Test real data during playback

Can you try these tests to see if / where it crashes? If I have time I will try them myself and let you know I did.

I am pushing the code to the main branch since you said the previous version was unusable.

miguelriemoliveira, 2023-10-15:
Hi @danifpdra ,

I already tried case 1. There were some bugs I corrected.

Please try the simulation as well, and then give me some feedback on how this behaves with real data.

miguelriemoliveira, 2023-10-15:
I am now trying with real data. We have some problems, namely we get this:

[image]

This is a really bad depth image ...

miguelriemoliveira, 2023-10-15:
Another example

[image]

[image]

miguelriemoliveira, 2023-10-15:
And another.

[image]

It seems the new algorithm does not like the holes in the pattern which occur for real data.

danifpdra, 2023-10-16:
Hummmm, that seems perfect for simulated data but real data is bound to have a lot of holes in it and that definitely doen't work, unless we create that manual eraser features for boundary points

miguelriemoliveira, 2023-10-22:
We did create a manual eraser useful for real data (#648), so I am closing this one.

closed 2023-10-22 by miguelriemoliveira