lardemua/atom #614: Depth label should be able to ignore outliters

danifpdra, 2023-09-22:
[image]

miguelriemoliveira, 2023-09-23:
Hi @danifpdra ,

what do you mean? You mean the automatic labeling should be able to do this?

miguelriemoliveira, 2023-09-30:
Hi @danifpdra ,

I will try to work on this, but first I will create a simple rgbd camera system to put in atom_examples.

miguelriemoliveira, 2023-10-02:
Hi @danifpdra ,

I created the rgb_depth_system. I think I solved the problem we had about the camera being a single chassis with two distinct sensors.

I cut the mesh in half and now we will be able to see a mesh for both  cameras. And if the calibration works fine we should have a nice visual feedback of both parts comming toghether.

Check https://github.com/lardemua/atom/tree/noetic-devel/atom_examples/rgb_depth_system

Will now try to calibrate it and solve this problem on the way.

miguelriemoliveira, 2023-11-14:
This is fixed since we implemented #648

closed 2023-11-14 by miguelriemoliveira