lardemua/atom #622: Why is the evaluation table style for the lidar_to_depth_evaluation is different from all of the other evaluation table in every other sensor pair?

danifpdra, 2023-09-29:
Normal tables:

[image]

Lidar to depth table:

[image]

I am asking because this table is much bigger and it's much harder to fit in the screen

miguelriemoliveira, 2023-09-29:
Hum, evaluations are now more on @manuelgitgomes 's side.

Can you help @manuelgitgomes ?

manuelgitgomes, 2023-09-29:
Hello all!

This will be fixed in my PR. I can do it now so it eases the process.
In 10/15 min it will be done.

danifpdra, 2023-09-29:
Don't worry @manuelgitgomes , it's not urgent. I was just evaluating a lot of stuff and I wondered why it was the only different one. But it's just a cosmetic question

manuelgitgomes, 2023-10-23:
Hello! This is already solved, sorry.

closed 2023-10-23 by manuelgitgomes