lardemua/atom #630: Meshes from pattern are on the opposite side

miguelriemoliveira, 2023-10-05:
I know @manuelgitgomes also has this problem, but for @danifpdra its ok.

Do you confirm?

Should investigate why this occurs ...

danifpdra, 2023-10-06:
I also have that problem during the calibration phase

[image]

miguelriemoliveira, 2023-10-07:
Hi @danifpdra , 

so you too? I was under the impression it was fine for you. That means me, you and @manuelgitgomes all have this problem. It must be fixed.

I think its because of this:

https://github.com/opencv/opencv-python/issues/755

miguelriemoliveira, 2023-10-07:
Solved. Had to create a new charuco pattern with a distinct coordinate system,

The new pattern is called **charuco_800_600_5x5_100**, where the last part is the reference of the aruco dictionary.

I will remove the old charuco_800x600 from the atom_worlds repo, but we have many robotic systems which still have it defined internally (see the example of [mmtbot](https://github.com/miguelriemoliveira/mmtbot/tree/main/mmtbot_gazebo/models/charuco_800x600) )

Also had to change several stuff in atom_worlds such as de world definition to include this new pattern.

I also want to rename and alter the structure of atom_worlds, but we'll do it later.

miguelriemoliveira, 2023-10-07:
Will push when branch called issue629 is merged.

closed 2023-10-07 by miguelriemoliveira