lardemua/atom #631: Use yaml file as input to replace all the parameters of atom scripts

miguelriemoliveira, 2023-10-06:
The idea is to have this working along with the classic argument arser so we can choose how to use.

We would need one yaml (at least) for each script  ...

Not sure if this should go ahead.

miguelriemoliveira, 2023-10-09:
@danifpdra gave the idea of having a script for each package ... something like

    rosrun rgb_rgb_system_calibration test_label_detector --sensor rgb_left

instead of using the script from the atom folder, like

    rosrun atom_calibration test_label_detector --sensor rgb_left --config <../rgb_rgb_system_calibration/calibration/config.yaml>

@manuelgitgomes , can you tell us what you think?

manuelgitgomes, 2023-10-09:
Hello @miguelriemoliveira, I think you have the wrong issue, it should be #633, right?

Nevertheless, it might be a good idea.
The user only runs thing inside their package, even if they only call something from the atom package.
It makes it easier for the user.

miguelriemoliveira, 2023-10-09:
> Hello @miguelriemoliveira, I think you have the wrong issue, it should be #633, right?

Hi @manuelgitgomes ,

right, its #633. I will copy the text there.